dresden-elektronik/deconz-rest-plugin #2742: 1000$ question - moving from raspbee to conbee

tosion, 2020-05-05:
Unfortunately this is only 1000$ worth for me but please help anyway ...

I am planning to move couple of my home automation systems from RPI3B+ to NUC (J4005).

I currently run homeassistant (HassIO) with Deconz addon with Raspbee. Now when I can not use Raspbee with NUC (can I - any adapter available?) and I need to move to Conbee, can I use Phoscon backup so that I don't need to pair all the devices again? (so can you jump with backup from another coordinator to another - is there an id problem)

On the other hand if I could use Phoscon backup between Conbee/Raspbee I could keep my RPI3B+ as a backup coordinator if NUC would chrash. -> remotely power of NUC, remotely power on RPI + HassIO snapshot and Phoscon backup in?

Comments?

Smanar, 2020-05-05:
Hello, have you take a look here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Backup-&-Restore

Or look at other issue ?
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1441
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1757

tosion, 2020-05-05:
Thank you very much for a very quick response!

Looks promising - however as far as I uderstand none of the links does not campare exactly my case:
New conbee / New computer

The first link "Backup & Restore" covers:
                    1. old conbee / new computer
                    2.  new conbee / old computer

#1441 "cloning" covers new conbee / old computer (it says: must be on the same machine)
#1757 "Migrate to Docker" I think is old conbee / new computer

It would be perfect if I could use new conbee in new computer with backup/restore feature. That would make it very easy to swap to new NUC and again back to old RPI if NUC crashes.

Smanar, 2020-05-05:
Yep, theses links are just example, you can find other.
To make it short, there is some data in the key and other on the machine, but all can be tranfered using this feature.
The only important point is don't run 2 machines with the same backup in same time.

tosion, 2020-05-05:
Thanks again!
Yes I understood it will probably work using backup restore, but it would be nice to know has someone already tested my case. I sure also understood it is not wise move to have them running same time, that's what I meant by:

"remotely power of NUC, remotely power on RPI"

Testing would be must - if it does not work, I think you could always go back to the old RPI/Raspbee system with hassio snapshot and phoscon backup restore (both work perfectly in "normal" situation)

Smanar, 2020-05-05:
It's a little more brutal but If  I have understand you are making
1. new conbee / new computer

So it's same than
1. old conbee / new computer
2. new conbee / old computer

But you are making the procedure in 1 step ^^

And yes, I think it will be easu to rollback for you if you have problem, with 2 coordinators and 2 machines.

tosion, 2020-05-05:
Step 1&2 would work for me at first installation. As also having a target to remotely swap the system between RPI/NUC it does not work anymore (need to touch hardware). I have systems far away so Phoscon backup/restore would work remotely when you need to start another unit as far as you have the latest hassio snapshot and phoscon backup in your hands ...

tosion, 2020-05-11:
The problem solved!

Basically using Phoscon backup and hidden configuration I was able to clone old Rasbee to new Conbee -> installed to new computer - all existing 66 nodes work in new machine ...

closed 2020-05-11 by tosion