dresden-elektronik/deconz-rest-plugin #2743: Aqara Relay, llkzmk11lm, issue with power report

epalzeolithe, 2020-05-05:
Hello, i managed to connect my aqara relay llkzmk11lm with ConBeeII

it appears in home assistant or Jeedom as 2 switch : good

But the power report is not working :
Power : -1W
Current : -
Voltage : -

I put an airbrush with 300W. > Still -1W

what's wrong ?

epalzeolithe, 2020-05-05:
Here's the output through REST command

{
    "config": {
        "battery": null,
        "on": true,
        "reachable": true,
        "temperature": 3800
    },
    "ep": 1,
    "etag": "793f12f9b797ebe86191b21b5947b17e",
    "manufacturername": "LUMI",
    "modelid": "lumi.relay.c2acn01",
    "name": "Power 28",
    "state": {
        "current": null,
        "lastupdated": "2020-05-05T11:25:58",
        "power": -1,
        "voltage": null
    },
    "swversion": "02-27-2019",
    "type": "ZHAPower",
    "uniqueid": "00:15:8d:00:04:0d:2f:f0-01-0b04"
}

Kristian8606, 2020-05-06:
I've been using deCONZ for a while I don't know what the problem may be but try to read the cluster attributes and turn on the search for sensors.
I was able to add my xiaomi smart plug EU.

SwoopX, 2020-05-06:
That's probably related to the special attribute reporting of Xiaomi devices. You could try running deconz with debugging enabled (`--dbg-info=2`). Then, make sure you have a load appplied and wait for a couple of minutes.

When done, check the debug output for something similar to this:

```
20:01:39:573 ZCL attribute report 0x04CF8CDF3C764934 for cluster: 0xFCC0, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
20:01:39:573 	payload: f700413d6410000328159839000000009539a8610b3f963900600b45973900000000052101009a20100821160107270000000000000000092100020b20009b1000
20:01:39:574 0x04CF8CDF3C764934 extract Xiaomi special attribute 0x00F7
20:01:39:574 	64 on/off 0
20:01:39:574 	03 temperature 21 °C
20:01:39:574 	98 power 0x00000000
20:01:39:574 	98 power 0
20:01:39:574 	95 consumption 0x3F0B61A8
20:01:39:574 	98 consumption 544
20:01:39:574 	96 voltage (?) 0x450B6000
20:01:39:574 	96 voltage (?) 223
20:01:39:574 	97 current 0x00000000
20:01:39:574 	97 current 0
20:01:39:574 	05 RSSI dB (?) 1 (0x0001)
20:01:39:574 	9a unknown 16 (0x10)
20:01:39:574 	08 unknown 278 (0x0116)
20:01:39:574 	07 unknown 0 (0x0000000000000000)
20:01:39:574 	09 unknown 512 (0x0200)
20:01:39:574 	0b unknown 0 (0x0000)
20:01:39:574 	9b unknown 0
```

That should enable us to investigate further.

xawme88, 2020-05-15:
Hi, I have the same issue with raspbee and llkzmk11lm. Power and Consumption are not reported through the API. I want to read from Openhab. #1491 reports the same issue.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix