dresden-elektronik/deconz-rest-plugin #2744: How many devices can I pair with Deconz

CarpeDiemRo, 2020-05-05:
Hi, I have a Conbee2 stick that I'm using with Hassio intalled on a RPI4. My question is how many zigbee devices can be paired with Deconz. I was trying to find this details in the product specifications but I didn't find it.
Thanks for your help.

Mimiix, 2020-05-05:
Was answered in : https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1379

CarpeDiemRo, 2020-05-05:
@Mimiix Thanks a lot for your quick answer. My bad, because I wasn't clear enough. 

My use case: 64 Gledopto lights spots. Can we all be managed by one Deconz stick? All are routers, no end point.

Mimiix, 2020-05-05:
Should work fine :)

CarpeDiemRo, 2020-05-05:
@Mimiix Thanks a lot. I dare to ask if you have tried or it is an assumption based on equipment specifications?

Mimiix, 2020-05-05:
The moderator on the deCONZ discord (Yabbah) has 80~ ish nodes connected :)

CarpeDiemRo, 2020-05-05:
Got it. Thanks a lot.

Mimiix, 2020-05-05:
> Got it. Thanks a lot.

You're welcome:)

closed 2020-05-05 by CarpeDiemRo