dresden-elektronik/deconz-rest-plugin #2745: [Coincidence] Damaged devices after few days after pairing to deCON

smesguich, 2020-05-05:
Hello,

Just to trace it. I'm not suspicious against the deCONZ Gateway. Legrand had also some mysterious way to use Zigbee but ....
Since I have start to move some devices from Legrand Gateway to deCONZ, I had :
- 1 wired-switch which didn't work any more (I mean physicaly and remotly)
- 1 roller-shuter which didn't work to roll up and only roll down (I mean Physicaly and remotly)

Both events was in the week after their first deCONZ pairing.

**It's for sure, just a coincidence and no correlation between thoses events** but if anybody had the same issue, that would be intresting to exchange about it ...

Smanar, 2020-05-05:
And reseting them, don't make them working again ?

What have you used as bulb on the wired switch ?

smesguich, 2020-05-05:
> 
> 
> Hello, and reseting them, don't make them working again ?
> 
> What have you used as bulb on the wired switch ? It's max 10 led and 60W (if you are using LED).
> I have already see some destoyed device issue for this one, but I dont remember if it was on deconz.

For roller shutter, I didn't tryed yet. I just remount it after changing a condo, because I was thinking the issue cames from the motor. I will tri tomorow with unplug motor.

For Wired switch, the module looks like unplug, nothing happends with reset or anything else. I was using it with 6x9W undimmable LED in // (54W totaly for 60W max so)

smesguich, 2020-06-03:
After few test, it didn't works anymore. I'm processing a RMA with my supplier.

Thanks all

[closed]

closed 2020-06-03 by smesguich