dresden-elektronik/deconz-rest-plugin #2746: Interlock

LukaszP2, 2020-05-06:
Hi,

How can I do interlock to 3 gang switch. Is there any option in conbee?

LukaszP2, 2020-05-06:
No... Lonsonho 
https://a.aliexpress.com/_BTuRoq

Smanar, 2020-05-06:
I can't be sure, but it seem it works https://easydomoticz.com/forum/viewtopic.php?p=81287#p81287

LukaszP2, 2020-05-06:
I dont see inter lock option (only one gang on at time) i want do use it for roller

Smanar, 2020-05-06:
Ha ok, sorry I haven't understand, so you want to make an lock feature to transform a double gang in a roller command ?

No, it's not possible in deconz, because on a normal double gang you can set both to "on", I think you need to program your "lock" using your home automation application.

LukaszP2, 2020-05-06:
i do this but it takes time (small but it is)... 
I see this future in shelly, than zigbee2mqtt and I hope I can do this in conbee ;/ 
maybe add this :)

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix