dresden-elektronik/deconz-rest-plugin #2749: Migrate from ConBee (Raspberry Pi 3 B) to RaspBee II (Raspberry Pi 4)

js-chris, 2020-05-08:
Hi,

as the title says, I am trying to migrate from ConBee (Raspberry Pi 3 B) to RaspBee II (Raspberry Pi 4).

Basically because my ConBee (Raspberry Pi 3 B) setup has started to crash from time to time and I am not able to find out why. The whole machine is not reachable anymore and I need to restart it. I searched in all logs to find the problem, but till now, I wasn't able to. 

So, after 3 years of successfully running this setup, I thought to move to a new setup.

My first attempt was, to setup the new Raspberry Pi 4 with the stable Raspbian Buster Desktop Image and exported the Phoscon configuration via "Backup" and restored it on the new machine. This actually resulted in all my devices being taken over, but unfortunately I was unable to switch them.  

On my old Pi 3 B, Phoscon runs on:
Produkt ConBee
Version 2.05.64 / 22.4.2019
Firmware 26330500

The new one is:
Produkt RaspBee
Version 2.05.67 / 26.8.2019
Firmware 26520700

I read a lot of articles, but I couldn't find something that would have helped me.

Is there any chance to migrate without resetting all my ZigBee Devices and adding them again to the new RaspBee?

Do you have any other ideas on how I might proceed or what I might try?

Thanks in advance
Chris

Smanar, 2020-05-08:
IDK for conbee 1 to raspbee 2 but you have some (bad) info here for conbee 2 to raspbee 2 Currently I wouldn't recommend the backup from ConBee II restore to RaspBee II approach since we still figuring out to get this work smoothly.

djashjones, 2020-05-08:
I went from conbee to raspb, fail!
Changed my channel number, fail!

Had to re-pair all my devices.

SwoopX, 2020-05-08:
@js-chris you don't run FHEM, do you? I also have a bricked system from time to time but in my case, it's FHEM which is responsible for that.

From memory, a switch from Conbee to Raspbee II brings some challenges. Smanar pointed you to the corresponding issue already. However, there might be improvements in this regard with the upcoming release, expected to be available next week.

js-chris, 2020-05-08:
Thanks for your answers. 

@Smanar I read the issue and the article and I will check the Zigbee configuration page. Hopefully tomorrow.

@SwoopX I do run FHEM, but not on the same Raspberry. FHEM is running on separate Ubuntu System. The Raspberry with ConBee does nothing more than Zigbee... What are your experiences with FHEM?

Thanks also for the hint to the upcoming release. If I can't make it working, I will definitely wait for that, before I start to re-pair all my Zigbee devices.

SwoopX, 2020-05-08:
@js-chris This is now all from top of my memory: I've deeply analyzed this with lots of command line Kung Fu that one of the perl processes related with FHEM was stuck waiting for I/O. It was totally bricked and definitively not recoverable any more. Also, it consumed a hughe amount of processing capacity due to whatever reason. As it was that fubar, a reboot was the only solution. I checked the file system after that and some errors got corrected...

js-chris, 2020-05-09:
Yeah, I got it working.

Just in case someone needs it.

- I flashed the "Raspbian Buster Desktop" from here https://phoscon.de/en/raspbee/sdcard to my new sdcard
- Back in the Raspi, I did all the config stuff through "sudo raspi-config".
- Now I copied the files session.default, zcldb.txt and zll.db from my old "ConBee Raspi" in
./home/pi/.local/share/dresden-elektronik/deCONZ/
to the same location on the new raspi
(But I think the Backup Function in Phoscon would have worked also...)
- The most important point was to load my last "old" zigbee configuration as described here:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-issues

Everything is working now. Thanks for your support.

closed 2020-05-09 by js-chris