dresden-elektronik/deconz-rest-plugin #2750: atsamr21r programming

ThomArmax, 2020-05-08:
Hello,

I have a [ATSAMR21B18-MZ210PA](https://www.microchip.com/wwwproducts/en/ATSAMR21B18) chip and I would like to flash your firmware on it
[image]

According to the datasheet, the module I delivered with a bootloader.
I have linked to ATSAMR21B18 to my RPi using FTDI 3.3V breakout board.

I have tried to flash the firmware using `GCFFlasher` with the following command : ` sudo GCFFlasher -f deCONZ_Rpi_0x26350500.bin.GCF`

When `GCFFlasher` prints `reset target` I put shortly `#RESETN` to GND to reset the chip, but it doesn't work.

Here is the `GCFFlasher` output:

```
pi@raspberrypi:~ $ sudo GCFFlasher -f deCONZ_Rpi_0x26350500.bin.GCF
GCFFlasher V2_11 (c) dresden elektronik ingenieurtechnik gmbh 2017/12/10
no device specified use RaspBee (/dev/ttyAMA0)
using firmware file: deCONZ_Rpi_0x26350500.bin.GCF
reset target

error: bootloader not responding
device used by other application?

```

1. Do you have any idea of what can be wrong ?
2. How can I ensure there is a bootloader ?

Thanks by advance

SwoopX, 2020-05-08:
I'm pretty sure that kind of use is not intended by DE. Better approach theit support directly on this.

ThomArmax, 2020-05-08:
> 
> 
> I'm pretty sure that kind of use is not intended by DE. Better approach theit support directly on this.

Yes I guess, since it's not the official module, but it's the same chip ... I would like to know if someone already did it or have any experience on this module

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix