dresden-elektronik/deconz-rest-plugin #2751: Smart plug with metering sensor creation

djashjones, 2020-05-08:
Why is it when I add a smart plug Deconz creates "Consumption_XX" & "Power_XX" rather than:

PlugName: ZHAConsumption
PlugName: ZHAPower

Smanar, 2020-05-08:
I don't realy understand, but what is "PlugName" for you ?
You are asking why the device is called Power_44 instead of Name_ZHAPower ?

djashjones, 2020-05-08:
Ok, sorry I was going by what node-red sees.

If I add a plug via Phoscon it will be called e.g. "Smart Plug XX: Smartplug" but the sensor creation will be "Consumption_XX: ZHA Consumption" & "Power_XX: ZHAPower". 

But if I add a sensor e.g. Samsung multi-purpose sensor, I get (node-red):

Multipurpose Sensor: ZHAOpenClose
Multipurpose Sensor: ZHATemperature
Multipurpose Sensor: ZHAVibration

SwoopX, 2020-05-08:
I never got anything comparable from the REST API, therefore I assume it's with node-red.

djashjones, 2020-05-08:
Another example. I added a Hue motion sensor and changed the name to something silly. All associated sensors updated accordingly.

    "34": {
        "config": {
            "alert": "none",
            "battery": 100,
            "delay": 0,
            "ledindication": false,
            "on": true,
            "pending": [],
            "reachable": true,
            "sensitivity": 2,
            "sensitivitymax": 2,
            "usertest": false
        },
        "ep": 2,
        "etag": "d9da6b1f7dd471a695d472c592f92c6f",
        "manufacturername": "Philips",
        "modelid": "SML001",
        "name": "Hue Mo Test",
        "state": {
            "lastupdated": "2020-05-08T19:27:00",
            "presence": false
        },
        "swversion": "6.1.1.27575",
        "type": "ZHAPresence",
        "uniqueid": "00:17:88:01:06:f5:69:07-02-0406"
    },
    "35": {
        "config": {
            "alert": "none",
            "battery": 100,
            "ledindication": false,
            "offset": 0,
            "on": true,
            "pending": [],
            "reachable": true,
            "usertest": false
        },
        "ep": 2,
        "etag": "2ecbb928afd76f4acecc7d9646eec35d",
        "manufacturername": "Philips",
        "modelid": "SML001",
        "name": "Hue Mo Test",
        "state": {
            "lastupdated": "2020-05-08T19:24:10",
            "temperature": 2063
        },
        "swversion": "6.1.1.27575",
        "type": "ZHATemperature",
        "uniqueid": "00:17:88:01:06:f5:69:07-02-0402"
    },
    "36": {
        "config": {
            "alert": "none",
            "battery": 100,
            "ledindication": false,
            "on": true,
            "pending": [],
            "reachable": true,
            "tholddark": 12000,
            "tholdoffset": 7000,
            "usertest": false
        },
        "ep": 2,
        "etag": "ade5c5397a91fd8dc18c7cb0cffe3f80",
        "manufacturername": "Philips",
        "modelid": "SML001",
        "name": "Hue Mo Test",
        "state": {
            "dark": true,
            "daylight": false,
            "lastupdated": "2020-05-08T19:26:02",
            "lightlevel": 5738,
            "lux": 4
        },
        "swversion": "6.1.1.27575",
        "type": "ZHALightLevel",
        "uniqueid": "00:17:88:01:06:f5:69:07-02-0400"
    },

djashjones, 2020-05-08:
My plugs how ever:

    "3": {
        "etag": "abc467502ca163f782bbbe82f0a33fe0",
        "hascolor": false,
        "manufacturername": "LDS",
        "modelid": "ZB-ONOFFPlug-D0005",
        "name": "Tweed PSU",
        "state": {
            "alert": "none",
            "on": true,
            "reachable": true
        },
        "swversion": "2.3",
        "type": "Smart plug",
        "uniqueid": "cc:cc:cc:ff:fe:bc:02:69-01"
    },

Consumption example:

    "8": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "0deafb274934cd5e95ad45e01633382e",
        "manufacturername": "LDS",
        "modelid": "ZB-ONOFFPlug-D0005",
        "name": "Consumption 8",
        "state": {
            "consumption": 253,
            "lastupdated": "2020-05-08T19:29:44"
        },
        "swversion": "2.2",
        "type": "ZHAConsumption",
        "uniqueid": "00:0d:6f:ff:fe:01:44:e8-01-0702"
    },

Power example : 

    "5": {
        "config": {
            "on": true,
            "reachable": true
        },
        "ep": 1,
        "etag": "ec1ad371a7a15416b407506d1d47bdd4",
        "manufacturername": "LDS",
        "modelid": "ZB-ONOFFPlug-D0005",
        "name": "Power 5",
        "state": {
            "lastupdated": "2020-05-08T19:29:32",
            "power": 4
        },
        "swversion": "2.3",
        "type": "ZHAPower",
        "uniqueid": "cc:cc:cc:ff:fe:bc:02:69-01-0b04"
    },

SwoopX, 2020-05-08:
In `de_web_plugin.cpp`, it's lines 17173-17187 I was referring to. But yeah, I know what you mean. Haven't paid too much attention to naming though.

djashjones, 2020-05-08:
oh I see. Would be nice, one day to line up the naming 👍

ebaauw, 2020-05-08:
> I added a Hue motion sensor and changed the name to something silly. All associated sensors updated accordingly.

How did you change the name?  I've never seen this behaviour in the REST API plugin.  This might actually be a courtesy of Phoscon.

Note that the Hue motion sensor is exposed as three `/sensors` resources.

> My plugs how ever:

Note that the plug is exposed as a `/lights` resource, whereas the ZHAConsumption and ZHAPower are exposed as `/sensors` resources.

> I'm guessing the "etag" links them

No, it doesn't.  The `uniqueid` does.  It composed of the Zigbee mac address, endpoint and cluster.  The ZHAConsumption you've listed does not belong to the plug you've listed.  The ZHAPower does.

djashjones, 2020-05-29:
Just deleted (Phoscon) a plug as I'm doing a firmware update via SmartThings and the sensors are still there. So it looks like I have to delete them via rest api?

closed 2020-07-11 by stale[bot]