dresden-elektronik/deconz-rest-plugin #2752: new philips device

maxnumberone, 2020-05-08:
Hello,
I hope it's the tight place to request some support for a new device.
Is it possible to add  the Philips HRC 0303 remote to the device list?
If you need details or any information I will try to provide what I can.
Thank you,
Z.

ebaauw, 2020-05-08:
Not sure if that's technically possible.  This remote is from the Living Colors era, before (at that time still) Philips introduced Hue.  I think they used touchlink to pair the remotes at that time, which is not supported by deCONZ.  I'm not sure if the remote supports regular pairing as well.

Did you already try and pair it to deCONZ?  Does it show in the GUI?  If so, please provide the screenshots as per the [Wiki](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support).

maxnumberone, 2020-05-10:
Thank you for your feedback,
I tried several things but no, I was not able to pair it to deconz.
I was not even able to pair it with the hue bridge.
I suppose I can recycle my both remotes :(

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix