dresden-elektronik/deconz-rest-plugin #2753: Capteur PST03-A

rudak, 2020-05-08:
salut
est il prévu d'intégrer le capteur Capteur PST03-A ?
merci

https://community.jeedom.com/t/capteur-pst03-a-zigbee-sous-jeedom/27441/9

Smanar, 2020-05-09:
Hello, try to use english please, at least with google translate, lot of people here don't understand french
I can perhaps, have you try to include it using deconz (and not phoscon). I need theses screenshoot to include it https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

Hello, essayes d'utiliser l'anglais, au pire avec google translate, beacoup de monde ici ne comprend pas le francais.
Sinon je peux faire un essais mais j'aurais besoin de capture d’écran de l'appareil inclus sous deconz et pas phoscon, les memes que sur le lien precedent.
C'est un appareil a pile donc tu risques d'avoir a le reveiller en meme temps que tu presses le bouton "read" pour remplir les champs.

rudak, 2020-05-09:
ok I can try but I only have Jeedom at disposal with the deconz plugin and obviously the captures come from somewhere else, so if you have indications to make these captures I will follow them.

Smanar, 2020-05-10:
Ok, so there is another problem, if I make code modification, you need to compile it to test it (I can't on my side because I haven't the device) and I already have some problem with users on jeedom. We can make without test, but the result will not be garanted)

BTW. to get theses screenshoot if you haven't acess to the GUI (From my memory jeedom are mostly headless machine), the easy way is installing deconz a second time on a PC machine.
Swap the key, make the inclusion on the PC machine, take screenshoot, and swap again the key.

- The configuration is on the machine so on the pc you will have an empty network (it's normal)
- Make fast because some devices like xiaomi will leave the network without the conbee on jeedom after some hours.
- The device will be invisible in Phoscon but visible on deconz (with GUI), after inclusion. If you have problem, you can make inclusion directly in deconz, but better try with phoscon first.

rudak, 2020-05-10:
Ok I'll try that, but first I have to make sure that when I put the flash drive back on the jeedom side, my devices will come back just like before. (I'm a noob)

So:
- I install the .exe on a pc
- I note the key
- I'm changing it to run the tests
- I'll retrieve the information mentioned and send it to you here.
- I put the conbee back on my rpi and I put the saved key back on.
- everything's back to the way it was before the operation

If you confirm me, I'll do it during the day.

Smanar, 2020-05-10:
1- install deconz on PC https://phoscon.de/en/conbee2/install
2 - Just unplug the conbee and plug it on the PC
3 - Start deconz
4 - pair the device, using Phoscon first, else deconz itself
5 - Use deconz to find information (french tuto > https://presentationdeconz.wordpress.com/linterface-graphique/)
6 - make capture
7 - unplug the conbee, and replug it on jeedom
8 - you will need to restart the deconz plugin on jeedom (not sure, but I think)

rudak, 2020-05-10:
Are those values enough for you?

[image]
[image]
![deCONZ_2020-05-10_16-15-20](https://user-images.githubusercontent.com/604478/81501686-
fab5d300-92d9-11ea-83c3-7241b1d45ea9.jpg)
[image]
[image]
[image]
[image]
[image]

Smanar, 2020-05-10:
Yes ^^. Perfect.

I just miss the manufacturer but I will try to find it somewhere else or the device can miss it.
And the "node info" but I think I can find the information too.

Smanar, 2020-05-10:
Ok si finnaly, I m interested by the panel "node info"
It's this one
[image]

You probably still have them on your PC.

I m asking because I can't say if your device in from "enki" or "philio" https://www.zwavetaiwan.com.tw/pst03

rudak, 2020-05-10:
> Yes ^^. Perfect.
> 
> I just miss the manufacturer but I will try to find it somewhere else or the device can miss it.
> And the "node info" but I think I can find the information too.

The manufacturer is [Philio ](https://www.zwavetaiwan.com.tw/pst03) and it works with Leroy Merlin Enki domotic box.

(I also found a doc that mentions the brand [Evology](https://www.leroymerlin.fr/v3/p/produits/detecteur-connecte-multi-fonctions-ouverture-et-mouvement-evology-e1501334365), which is also a brand distributed by Leroy Merlin).

Smanar, 2020-05-10:
But the code works with a number 0xXXXX, deconz use this value to reconize the manufacture, and I don't find it on the net.

rudak, 2020-05-10:
Here's the part I forgot.

[image]

Smanar, 2020-05-10:
Nice It's that 0xffa0.

Ok so now how to try the code ?
What is your jeedom machine ? ARm64 or a raspberry ?

rudak, 2020-05-10:
_Sorry, I answer point by point but at least the information is clearer._

I'm personally with Jeedom v4 on rpi4.

rudak, 2020-05-10:
I don't know how to test a plugin modification with jeedom without going through the original market circuit.
but if I have to I can do some research.
Otherwise, if you're sure of yourself, just merge the commit to the plugin and ...banzaï !

Smanar, 2020-05-10:
NP , I have time ^^.

So now 2 solutions.
1 - Compile the code yourself
2 - I can compile it for Raspberry 3, hopping for compatibility for you (I think pi3 > Pi4 will be ok, and both are 32bit )

If you choose the method 1, you have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin at "Install deCONZ development package (optional, Linux only)" just replace the step 1 by

`git clone https://github.com/Smanar/deconz-rest-plugin.git`

And if you want to make a backup, the only modified file will be "libde_rest_plugin.so" in "/usr/share/deCONZ/plugins"

It will take 10mn to compile, nothing changed in your configuration.

rudak, 2020-05-10:
I'm having a little trouble compiling
pi@raspberrypi:~/deconz-rest-plugin $ qmake && make -j2
qmake: could not find a Qt installation of ''

Smanar, 2020-05-10:
Yep, it's that.

Last chance command

```
wget -O - http://phoscon.de/apt/deconz.pub.key | sudo apt-key add -
sudo sh -c "echo 'deb http://phoscon.de/apt/deconz $(lsb_release -cs)-beta main' > /etc/apt/sources.list.d/deconz.list"
sudo apt update
sudo apt upgrade
sudo apt install deconz-dev
```

Smanar, 2020-05-10:
Je viens d'avoir un doute, Jeedom ca tourne pas sur Raspbian ? Ils aurait pas un autre systeme ? et 64 bit justement ?

rudak, 2020-05-10:
that's it I compiled ! I put the file, I reboot and I test if something still works, and hopefully if I see the new hardware appear!

my jeedom works on Raspbian GNU/Linux 10 (buster) 32bits (armv7l)

rudak, 2020-05-10:
unfortunately no new devices have arrived on jeedom.

Smanar, 2020-05-10:
The code just add the device in whitelist with

model name PST03A-v2.2.5
mac adress starting with 0x000d6f
Manu code 0xFFA0

You have changed the step 1 ?

You need to make a new inclusion, I don't think the device will appear in phoscon, but It will appear in jeedom (perhaps after a synchronisation), and you will have lot of new entries (I think 4/5), I will make explanation later.

rudak, 2020-05-10:
Yes, I've done the inclusion of material like I do with the other devices, but nothing.
[image]

I've got the .so binary in the right place so now I don't know. No error in the logs, it just doesn't appear.
sadness

Smanar, 2020-05-10:
You have restarted deconz ?
It need to restart it to use the new file, else it is using the old one already loaded.

rudak, 2020-05-10:
> You have restarted deconz ?
> you need to restart it to use the new file, else it is using the old one already loaded.

Yeah, I reboot the rpi4 and restarted services.

I'll try to check with phoscon like you said above but he usually doesn't see the stuff so I don't think that's gonna change...

rudak, 2020-05-10:
wouldn't it be possible to recover the devices even if they are not in the white list? at least for debugging purposes (in a debug branch ?)

Smanar, 2020-05-11:
Devices was blocked only in the API, you can see and act on them using deconz.
There is too a debug command line realy usefull, but not sure you will like it on jeedom ^^, so I m trying to avoid it.

I have a theorie for the code don't work.
If you look at your screenshoot, on the first one you miss "manufacture name", and It can be something blocking, my first idea was you have a problem during pairing, but some device realy miss it.

So I have modified the code to force a manufacture name
https://github.com/Smanar/deconz-rest-plugin/commit/5af3dded45b826268676ef93667be07b9b463b08

Else for information (in the worst situation) to enable debug mode, you need to add some flag, so on jeedom, If I m right, you need to close it and relaunch it with 
` sudo deCONZ –auto-connect=1 –dbg-error=1 –http-port=8484 –ws-port=20877 –upnp=0 -platform minimal –dbg-info=2 –dbg-aps=0 –dbg-zcl=0 –dbg-zdp=0 –dbg-http=0`

If you use it as command line, all logs will be send on the console, so nothing will be memorised on your drive (usefull if you are using SD card), but you need to look at them carefully ^^.

rudak, 2020-05-11:
Hi, I'm not able to tell you why, but I pulled your commit, recompiled and tried pairing again, it worked straight away!

thanks a lot for your help, I'm available if you need to make changes somewhere. thanks again.

closed 2020-05-11 by rudak

Smanar, 2020-05-11:
Already ?
And You have found all devices ?
Because I think some of them will be useless, you will have at least 1 Lux + 1 Temp + 1 presence + 2 alarm.

You pulled the commit after or before the last modification (for the manu name) ?

rudak, 2020-05-11:
I made the commit before and followed the qmake procedure to generate a new file, (it was very fast this time by the way) and everything worked directly.

not everyone will recognize because not everyone uses Jeedom but here is the information returned by the equipment
[image]

Smanar, 2020-05-11:
Yep, the first time take more time, because you need to compile all files, but in c++ you don't compile all file all the time, only needed files, and there is a link phase after.

The next not supported device, you will be able to include it yourself ^^, you have done the hardest part, if you look at my code modification, you will see I have absolute do nothing.

So I have let the code as it was (realy few change) so the API have created device according the cluster list (the blue number).
Everytime, the UniqueID is composed by
- the device mac adress
- the endpoint (01 or 02 for you)
- the cluster (the blue number)

There is some magic in the deconz code, so not all cluster create entries, and some of them can be special. Lot of things are automatic, like for values return but not in all situations, so take time to check all entries.

For exemple the Entry "Boutton 01-0006", I don't think this one will be usefull.
I don't think It will work, because I haven't finished this part, but IDK if you need it, I think you will prefer to use motion/contact entry.

You have too some setting or battery level to check, but this device seem realy compliant with zigbee stuff, nice, less work to do.

rudak, 2020-05-11:
Now I just installed it on a door, I'm going to make some tests to be able to make a return on the good functioning in a few days, it could be useful to add a maximum of details for those who will follow the issue one day or the other...
I can see the modifications made and indeed, it's not huge but you always have to know where to implement them and that's another music, especially blind like you did. good work.

Smanar, 2020-05-12:
Ok, so I will wait before making the PR.

rudak, 2020-05-12:
So, after a few tests, it works quite well but then the material is not of great quality. I compare it with other sensors with identical functionality and it's clearly not great I'm disappointed.
Apart from this subjective point, yes it goes up well so you can do the PR without any problems.

Smanar, 2020-05-12:
Ok. I do it now.
But wich one problem have you ? This device seem realy nice.
Distance detection ? weak ? perhaps it's a setting problem ?

rudak, 2020-05-12:
in fact there is an anti-tearing sensor behind, but once it's engaged, (when it's fixed to the wall) it doesn't detect the presence, the brightness sensor is bad, the values are really bad compared to the equivalent material of xiaomi.
for the temperatures, it can still be ok even if I find it badly calibrated.
I do not recommend the purchase of this device ^^

Smanar, 2020-05-13:
And I don't see setting for temperature or brightness sensor on your capture, so not possible having better ....

And I m agree, I stop to buy Xiaomi, because I have too much mesh problem with them, but it's realy a pity, because theses device works realy good.