dresden-elektronik/deconz-rest-plugin #2754: Heiman Siren HS2WD-E

martinschm, 2020-05-09:
Hi,
i can add the Heiman Siren to the Phoscon App but there is no way of controlling it.
Only when editing the name the sound will go off.

Any chance to integrate it properly?

Best
Martin

Smanar, 2020-05-09:
As said here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/565
Have you tried with API ?

```
/api/<apikey>/lights/<id>/state -X PUT -d '{ "alert": "select" }'
/api/<apikey>/lights/<id>/state -X PUT -d '{ "alert": "lselect" }'
/api/<apikey>/lights/<id>/state -X PUT -d '{ "alert": "none" }'
```

thogens, 2020-06-03:
Hi!
I bought the same siren, but I'm not able to connect it to Phoscon using Conbee 2 Stick.
I'm very new to this topic, so please forgive me.
My setup: 
Synology running Docker.
In Docker is running the following:
- Deconz==Phoscon
- ioBroker

How can I evaluate why the device is not found?
Any idea?
Thanks,
Thomas

Smanar, 2020-06-03:
Are you using phoscon for inclusion ? (you can try with light and sensor).
The led blinks on device just after reset ?

thogens, 2020-06-03:
What do you mean with "using phoscon for inclusion"?
I'd like to include several sensors and one siren into my home setup. This is my first device, the sensors are on their way, so unfortunately I can't try another one now.
I just noticed one interesting thing:
The phoscon app shows:
Version: 2.05.76 / 22.05.2020
**Firmware not connected.**

Does this help?!

Smanar, 2020-06-03:
Yes ^^, it doesn't work atm, so you can't include something.

But unfortunately, docker is not for me, sorry.
You can try hardware tests like using an USB extension cable, don't use USB3.0.
But for software solution, I think it's better make a new issue if no one answer you here.

thogens, 2020-06-03:
You are my hero of the day!
A simple, old USB extension worked. 
THANKS
Thomas

Mimiix, 2020-06-12:
Moving to Phoscon git repo, as it has to do with that.