dresden-elektronik/deconz-rest-plugin #2755: New Device Request - Sage Doorbell Sensor

Puntoboy, 2020-05-09:
Hello, I've recently bought this sensor and I would like to integrate it with my Home Assistant. Below are what I think are the required screenshots.

[image]

[image]

MattL0, 2020-05-12:
Just ordered mine.

Puntoboy, 2020-05-12:
I've only bought this sensor as my video doorbell won't send notification to Alexa so I want to use HA to do it instead.

MattL0, 2020-05-19:
is there any new hardware integration on deconmz? it is calm since a while @manup

MattL0, 2020-06-05:
Up @manup ?

Mimiix, 2020-06-06:
@SwoopX

ManicGTI, 2020-06-10:
+1 here as well, this would be useful for my Nest Hello

Puntoboy, 2020-07-01:
Any update on this?

MattL0, 2020-07-01:
+1 would really like to add this to my setup @manup

SwoopX, 2020-07-09:
I have no clue if this will work since I don't have the device. Chances are very high that the button map is not correct. I'd require some debug output then.

Puntoboy, 2020-07-09:
> 
> 
> I have no clue if this will work since I don't have the device. Chances are very high that the button map is not correct. I'd require some debug output then.

Tell me what you need and I can send it. You'll have to break it down though as I am new to this.

SwoopX, 2020-07-09:
Dunno, they did it with 2 buttons, I do it with one. Should have pretty much the same effect

Puntoboy, 2020-07-09:
So what I've found is that I can pair it to Deconz, it appears in the VNC viewer but not in the web app. When I look at the logs in HA there is no input from the device.

SwoopX, 2020-07-09:
> When I look at the logs in HA there is no input from the device.

No surprise since it's not supported yet.

Puntoboy, 2020-07-09:
> 
> 
> > When I look at the logs in HA there is no input from the device.
> 
> No surprise since it's not supported yet.

Clearly...

So where do we go from here?

SwoopX, 2020-07-09:
Wait for the new release and check it out. Changed the button map over here to mimic zha. I feel the device could behave funny otherwise (and possibly brick the intended functionality). Worst case is that you'd require 2 buttons for your home automation, 1 for open and one for close.

Puntoboy, 2020-07-09:
Great thanks. I'll give it a try.

SwoopX, 2020-07-27:
Closing this as support has been added with version .79

closed 2020-07-27 by SwoopX

ManicGTI, 2020-07-27:
Whoop!  Thanks.

Puntoboy, 2020-07-27:
Pity I can’t upgrade to get it :(

Kind regards,
Chris Jeffrey

Get Outlook<https://aka.ms/qtex0l> for iOS

SwoopX, 2020-07-28:
I need to have some feedback if it works alright though. If not, please raise a seperate issue for that :wink:

ReX1983, 2020-08-11:
> I need to have some feedback if it works alright though. If not, please raise a seperate issue for that 😉

Good stuff @SwoopX: it works (integration in Home Assistant).
I have connected only the front wire (it has also a rear wire) and what I get in Home Assistant is event 1002 when the doorbell button is pressed. There is no release event, but that's fine.

For those interested: it works also with Nest Hello. I don't have an internal chime, was hoping to use the Nest notifications played on a Google speaker but it has too much delay to be used in a real life scenario.

ManicGTI, 2020-08-11:
Excellent, It's the Nest Hello I want to use it with.  Have you hooked it up via the Chime Connector?  I'm assuming I can run it in parallel with a chime when I eventually get one.

ReX1983, 2020-08-11:
> Excellent, It's the Nest Hello I want to use it with. Have you hooked it up via the Chime Connector? I'm assuming I can run it in parallel with a chime when I eventually get one.

I've used the Chime connector even if I don't have any chime. The wiring is: one white/grey couple from the chime connector is wired to the zigbee sensor and the other couple is in series with the power source and Nest Hello.

The chime connector ensures that the zigbee sensor is not constantly powered (that would be the case if installed in series or parallel of Nest Hello, without the connector).

My theory is that the chime connector detects the button press (by measuring the tension on the series) and powers the other line connected to the chime or the zigbee sensor.

Puntoboy, 2020-08-12:
I would love to start testing this but the newer versions of Deconz won't work with my HA install on Docker. :( So until I move my HA/Deconz install to a different system I cannot update.

Mimiix, 2020-08-12:
@Puntoboy If you are on discord, we can take a look together?

Puntoboy, 2020-08-12:
> 
> 
> @Puntoboy If you are on discord, we can take a look together?

Thanks but I'm not on Discord. 

Once I move my HA install from Docker on Synology I'll be able to update the Deconz install. I had already upgraded but then found that Deconz stopped working. I posted here about it https://github.com/home-assistant/hassio-addons/issues/1503#issuecomment-665001517

Mimiix, 2020-08-12:
IT is useful to be there ;)

Anyway, I have deconz on a VM with HA. My advice would be run a VM with HA Supervised (https://github.com/DubhAd/HA-Notes/blob/main/supervised-on-linux.md) . Then run a second VM with raspbian/ubuntu and run deCONZ there.

Way more stable in my case.

Puntoboy, 2020-08-14:
> I've used the Chime connector even if I don't have any chime. The wiring is: one white/grey couple from the chime connector is wired to the zigbee sensor and the other couple is in series with the [...]

Thanks very much for this. I have finally migrated my HA installation to a virtual machine and upgraded Deconz to the latest version, I now have the Sage sensor working with Node Red and using Alexa to announce when the doorbell is ringing :) Handy when I'm away from the chime in the garage or something. :)

ReX1983, 2020-08-14:
> Thanks very much for this. I have finally migrated my HA installation to a virtual machine and upgraded Deconz to the latest version, I now have the Sage sensor working with Node Red and using Alexa [...]

Good stuff!

I have an automation to play a sound on the Google Home speakers and one automation to play a sound on the Xiaomi gateways. Eventually I have decided to stick with the latter because it all works in local: it is faster and more reliable.

MattL0, 2020-08-15:
Thanks for this .

Is it possible to add the battery attribute? I only get null.

MattL0, 2020-08-15:
I do not see it in phoscon, but I can acess it from the Homeseer or json api. 

 all i get is the 1002 value when pressed.. Is that ok? 

thanks

SwoopX, 2020-08-16:
I assume not, but I leave it for the moment. If all goes well, you as users can make a change, if required, on your own without touching the code.

Puntoboy, 2020-08-16:
Yes this is another device that doesn't show in the webapp but does show in HA. Mine is working fine. Just using the front connection so have it in node red to send an announcement via Alexa. 

[image]

Chr1stian, 2020-08-26:
How do you use the sensor when connected trough DeCONZ? It shows as "Bell" for me in HA, but no entities and listening to `deconz_event` reports nothing when I put voltage on the pins of the sensor :/ Appreciate any help on the setup process of Sage + Conbee II + DeCONZ + HA 🙏

MattL0, 2020-09-07:
Anyone knows how long the battery will last with this?

MattL0, 2021-02-14:
We just moved to a new place.  I reinstalled the sage doorbell, but it was always ''unreachable'' and it didn't send anything to jowihue homeseer plugin. 

I tried to pair a brand new sage doorbell but got the same result. 

This had been flawless since the almost a year ( I added the sensors at that time and never touched it) . I do not understand . I also tried to remove any trace of this sensor in the zll.db  and repair it again, but no luck. Still not reachable and does not send anything to homeseer when the doorbell is pressed. 

anyone have any insight?

MattL0, 2021-02-14:
**************************** Object info *****************************
{
"config": {
"group": "2",
"on": true,
"reachable": false
},
"ep": 18,
"etag": "89b725909280f2e73ba5e0c2d299c305",
"lastseen": null,
"manufacturername": "Echostar",
"mode": 1,
"modelid": "Bell",
"name": "Doorbell Sensor",
"state": {
"buttonevent": null,
"lastupdated": "none"
},
"swversion": "3.2.2",
"type": "ZHASwitch",
"uniqueid": "00:08:89:00:01:d9:dc:13-12-0006"
}

MattL0, 2021-02-14:
i 'll open a new issue for this

oywino, 2021-03-20:
> We just moved to a new place. 
> Anyone have any insight?
I've read that you need at least 12VDC to trigger the sensor, so if your doorbell at the place differs from the one you had at the old place, then that might be the reason.

oywino, 2021-04-11:
> i 'll open a new issue for this

Did you ever manage to solve this issue?
I'm having exactly the same problem.  Pairing it using Phoscon was easy, and it shows up in Phoscon like this:
[image]
But as you can see, it's reported as "unreachable".
In HA, it shows up as a "Device" with no "Entities" and there's nothing on the event bus.
[image]

I'm running Marthoc/deConz 2.10.4 in a Docker Container with a connected ConBee II with Firmware 0x026660700
I need all the help I can get on this one 😊

Adjrobinson, 2021-04-16:
I bought one of these and can't get it to show up in the phoscon webpage, I think this is it when I vnc in?
[image]. I would be happy to buy one and send it to the dev's as I would love to have it working.

oywino, 2021-04-16:
This is what mine looks like in deConz;
[image]

Can you check what firmware version is in your DoorBell sensor?
[image]

Adjrobinson, 2021-04-16:
It doesn't seem to show: 
I tried deleting it and adding it again with no difference.
[image]

oywino, 2021-04-17:
I think you have to click in the field "SW Build ID" to bring up the Attribute Editor, and in this click on "Read" (or Read Config) - I don't remember exactly.

Adjrobinson, 2021-04-17:
ah yes, looks like the same, weird, I deleted it again and turned off the routers and paired it in the same room as the combee, still no joy: 
[image]

Mimiix, 2021-04-17:
It's normal it doesn't show in phoscon. They need to add support for it.

Adjrobinson, 2021-04-17:
Ok, I have ordered https://www.tindie.com/products/ErikLemcke/doorbell-modernizr-2/ instead so if one of the dev's want the sage one to add it let me know as it's pretty pointless at the moment lol, looks like zigbee2mqtt might support it but I don't want to go down that route.

oywino, 2021-04-17:
> It's normal it doesn't show in phoscon. They need to add support for it.

Incorrect. Mine does show in Phoscon as well as in Home Assistance! It's just listed an "unreachable". Using deConz, I can read attribute values, but I cannot activate it. But I have read elsewhere that others have been able to make it work using deConz until they one day disconnected it and could no longer successfully reconnect it again after updating.

Smanar, 2021-04-17:
Deconz version 2.05.77, idk in wich one version this device was added, but can you try with a more recent version ?

oywino, 2021-04-17:
I'm using deConz 2.10.04.
I'd love to share my JSON but I don't know how. Please explain.

Smanar, 2021-04-17:
You can find it.
- using third app
- using phoscon with dev console
- checking direclty the api

`http://IP:PORT/api/KEY/lights`
and
`http://IP:PORT/api/KEY/sensors`

IP and PORT are the same than for phoscon
Key is an api key https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

oywino, 2021-04-17:
Here it is:

`"12": {
        "config": {
            "group": "20",
            "on": true,
            "reachable": true
        },
        "ep": 18,
        "etag": "75031fec22a9760f05a1604edebac11b",
        "lastseen": "2021-04-17T06:19Z",
        "manufacturername": "Echostar",
        "mode": 1,
        "modelid": "Bell",
        "name": "Doorbell Sensor",
        "state": {
            "buttonevent": null,
            "lastupdated": "none"
        },
        "swversion": "3.2.2",
        "type": "ZHASwitch",
        "uniqueid": "00:08:89:00:01:d9:af:d2-12-0006"
    },`

Smanar, 2021-04-18:
Oups sorry, I have read "doorbell" so in my mind was a "siren" not a button.

Do you have access to the GUI ? I need log when you press the button, deconz/help/log with "info" and "info_l2"

oywino, 2021-04-18:
Well, that's the whole problem. Because it appears as "unreachable", nothing happens when I press the button.
We first have to make it appear correctly and reachable in deConz.

Smanar, 2021-04-18:
No I don't say, phoscon but deconz, the application.

Puntoboy, 2021-04-18:
Sorry I haven’t read all the replies but my sage doorbell is showing in Phoscon and working great with HA. I have it triggering Alexa when someone rings the doorbell.

Kind regards,
Chris Jeffrey

Get Outlook<https://aka.ms/qtex0l> for iOS

oywino, 2021-04-18:
> Sorry I haven’t read all the replies but my sage doorbell is showing in Phoscon and working great with HA. 

Could you please share the Entity details that the Sage Doorbell is giving you in HA ?

Puntoboy, 2021-04-18:
Not sure what’s that going to show you?

[Image.jpeg]

[Image.jpeg]

Kind regards,
Chris Jeffrey

Get Outlook<https://aka.ms/qtex0l> for iOS

oywino, 2021-04-18:
(You forgot the images)

Puntoboy, 2021-04-18:
I didn't :) just the email didn't upload them. 

[image]

[image]

[13 more replies]