dresden-elektronik/deconz-rest-plugin #2756: Tuya RH3040 is not consistent

CarpeDiemRo, 2020-05-10:
I have a PIR RH3040 and I have the following issue.
It is successfully integrated into HA via DECONZ, not displayed in the DECONZ UI interface but appears as an entity in HA. So far, so good.

I observed the following unnatural behavior.
a. The PIR doesn't detect motion continuously. EX. Detect the first movement and, even though the movement Is continuous in front of the sensor the PIR state turns to "off"
b. After this transition, the PIR doesn't change to 'on' even though the movement reappears in front. If I press the side button the sensor sent the message to DECONZ and it is triggering again, but only for a short period of time.
c. It is clear that the relationship with DECONZ as a coordinator is ok and has no problem. My assumption is maybe the PIR is broken. Do you have the same issue? Or maybe I did something wrong.
d. After several tests, I realized that the PIR  state is changing to "on" after a few seconds and sometimes after 5 minutes. Very, very inconsistent. 

_In conclusion the two issues: 
1. The transition from "off" to "on" is very inconsistent
2. The PIR is not able to identify continuous movement and it's transitioning to "off" even though is movement in front_

[image]

SwoopX, 2020-05-10:
Hi. Would you have a look at the IAS Zone cluster? Please push the button on the device and than read the the attributes. Is the sensor enrolled and has the IAS CIE address of the coorinator written?

CarpeDiemRo, 2020-05-11:
Hi, I made a sensor reading and I obtain the following data. Unfortunately I don't know how to interpret. Could you help me, please?

[image]

SwoopX, 2020-05-16:
In your last screenshot, you must click on the `IAS Zone` cluster, scroll to the bottom and click read directly after you've woken up the device (by waving your hands in front of it).

closed 2020-06-15 by stale[bot]

seryeb, 2021-04-26:
I have exactly the same error. It detects motion for 3 times after every 50 secs of timeout for rescan. After third, no more detections despite I was moving in front of the sensor.

[image]

Could be related to firmware of PIR RH3040??