dresden-elektronik/deconz-rest-plugin #2760: [Request device support] PGnE electic meter.

adhvikp, 2020-05-12:
Hi Team,

Need help to integrate the PGnE electric meter data in conbee II.

PGnE is an electric company in north california and has support for smart meter. To enable the data stream from the smart meter to home we can use the few approved devices. Example

https://rainforestautomation.com/rfa-z114-eagle-200-2/

you are read the full integration @ https://www.pge.com/en_US/residential/save-energy-money/analyze-your-usage/your-usage/view-and-share-your-data-with-smartmeter/reading-the-smartmeter/stream-your-data.page?WT.mc_id=Vanity_streammydata

Few issues earlier which are here for reference. (stale and closed)

- #502 (Zigbee-HAN supported)

Please help me on the following topic to integrate the smart meter to conbee II.

1. Device type
2. mac address (is it the same as the MAC address shown in Node list for the coordinator ?)
3. Install code ( is it the same as the Install code that appears in deConz application plugins > rest api plugin and in the modal dialog box at the end) 

Device type options.

[image]

SwoopX, 2020-05-16:
Well, I'm afraid there's a boatload of questions to be answered upfront (similarly to the issue you referenced).

> To enable the data stream from the smart meter to home we can use the few approved devices.

I currently highly doubt that Conbee is one of the chosen devices? The questions/remarks raised back then are still applicable. What zigbee profile does the smart meter require?

Unfortunately, it's not an every day device.

adhvikp, 2020-05-20:
> I currently highly doubt that Conbee is one of the chosen devices? 
No, we have a option to put new device.

> The questions/remarks raised back then are still applicable. What zigbee profile does the smart meter require?
ZigBee SEP1.0 or 1.1 

Source : https://www.pge.com/en_US/residential/save-energy-money/analyze-your-usage/your-usage/view-and-share-your-data-with-smartmeter/reading-the-smartmeter/information-for-device-vendors.page

PGnE does say that customers can try to connect new device.
All i need to understand is how to get the mac address/install from conbee II  test this scenario.

Smanar, 2020-05-21:
PGnE does say that customers can try to connect new device or connect it to a device ?
It seem this device is a coordinator no ?

adhvikp, 2020-05-22:
> PGnE does say that customers can try to connect new device or connect it to a device ?

Quoting from the PGnE

> device can successfully connect ("join") with PG&E's SmartMeter™ in order to provide device users with near real-time energy use. PG&E supports issues relating to device connectivity to the meter; however, PG&E does not validate or provide support for other device functionality. Customers must troubleshoot any functionality issues with the device vendor directly.

> It seem this device is a coordinator no ?

Not sure that is where i need the support of this group to understand the usage.

ebaauw, 2020-05-22:
> ZigBee SEP1.0 or 1.1

Afaik, deCONZ only supports the ZHA and ZLL protocols.

closed 2020-06-20 by stale[bot]