dresden-elektronik/deconz-rest-plugin #2762: Some devices suddenly don't connect

StoricU, 2020-05-12:
Hi,

My setup have been working great for the last few months, but when I woke up today I  was not able to operate my Hue dimmer switch.
The battery is ok, it flashes when I'm pressing a button, but there is no reaction in deConz.
[image]

As you can see og the picture, I have also lost contact with a light, that is just beside the light below. So there is no distance/signal problem.

Anyone have a suggestion for troubleshooting or a way to fix it?

Smanar, 2020-05-12:
And the light don't come back after a power cycle ?

StoricU, 2020-05-12:
> Hello, and the light don't come back after a power cycle ?

Did not think of trying that.
My main focus was why the remote did not work, since the light is always on anyway.

But testing this also fixed the remote problem. After the power cycle, the light connected and 1 second later the remote connected.
[image]

Don't know what happened, but it works now! :)

closed 2020-05-12 by StoricU

Smanar, 2020-05-12:
Lol, yep but the light wasn't the remote parent.
Zigbee magic :)