dresden-elektronik/deconz-rest-plugin #2763: New device request

NicolasBoulanger, 2020-05-12:
Hello,
i bought 2 zigbee osram mini switch, i made a pull request to make it working on zigbee2mqtt, but now i use conbee :)
i have 1 blue and 1 white
i like these remotes for my roller shutter
https://www.amazon.fr/dp/B074PYT9R4/ref=twister_B07MVNX6V6?_encoding=UTF8&psc=1
the device is detected in deconz but can only switch on and off all lights
here screenshot of the 2 remote (maybe different)
if you need more informations tell me :)
thanks !

Blue: 
[image]

White:
[image]

Smanar, 2020-05-12:
Hello.
What is the mystery with this device ^^ ?

There is so much issue about it
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2082
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/374
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/294
ect ...

Some people say it's working, some other not.

And I don't see the problem, I m missing something ?

This device is detected in deconz ? I don't see it in supported device on the code, you have it as light or sensor ? Can you show me the device JSON pls ?

You want to use it for your roller shutter ? But how it work ATM, I m seeing only 3 button on the device and according to the cluster it can make on/off+ level control + color control, seriously ???

NicolasBoulanger, 2020-05-12:
its not really working, i cannot see remote in switch category on deconz software
it is detect but cannot assign light or whatever
remote can switch on all light with up button
turn on life with down button
i cannot configure button in deconz

i use it with:
short up press : open my living room cover 
short circle pres:  stop  living room cover
short down press: close  living room cover

long press up: open desktop window cover
long press circle:  stop desktop window cover
long press down: close desktop window cover

my second remote, i use it for garden light

which json logs you need ? which command? 
this logs ?

15:52:24:069 APS-DATA.indication srcAddr: 0x532a, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 215, rssi: -6715:52:24:069 APS-DATA.indication srcAddr: 0x532a, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 215, rssi: -67

15:54:00:255 APS-DATA.indication srcAddr: 0x50b9, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 215, rssi: -67

thanks !

Smanar, 2020-05-12:
Ok so you are using it with zigbee broadcast (or group) command ?
You can't see in phoscon , ok it is normal
But can you see it in deconz app (the desktop application)

Are you using an home assistant application ? HASS openhab or other ? (and have an API key ?)

NicolasBoulanger, 2020-05-12:
i'm using unicast: default configuration in deconz docker version
[image]
im using home assistant only to display my entities, but 
my automations are in node-red vm
my configuration : 
proxmox server :  
deconz vm
node-red vm
hassio vm

all in bridged network, managed by my unifi router
i use deconz api in node-red with node-red-contrib-deconz palette
i cannot see remote in hassio

NicolasBoulanger, 2020-05-12:
if can help
this is my commit on zigbee2mqtt repo
https://github.com/NicolasBoulanger/zigbee-herdsman-converters/commit/bd4463b2a370b47e6654006d8363a3f63e513886?fbclid=IwAR3F1yndD9fTJZn1ShBC_P3vqy6BHgwLUYNk9_R7z9F7jQV2ot6su9akhv4

Smanar, 2020-05-12:
Ok so I think this device is not in deconz at all.

It will be something I will ask again soon so to see your device in deconz API, you can use this kind of url
`http://IP:PORT/api/API_KEY/sensors`
For ip and port, it's the same you are using than for phoscon
For API_KEY I think you can find it somewhere on the deconz plugin in HASS

Now I can try make a ZHAswitch for this device, so you will have a value for all fonctions (IDK how much yet).

But if you want to make "direct connexion" between your shutter and the remote, not sure if it will work. You can make try using phoscon.
IDK if your shutter support "direct connexion", wich one cluster they need, and why you can use it with bulb and not the shutter ....

But if you want a buttonevent value like for ikea remote, I think it will be possible.

NicolasBoulanger, 2020-05-12:
ok i'll check with the  api
my shutter is in wifi (shelly2.5 module) so, event, yes and button configuration in web interface, it will be the solution

NicolasBoulanger, 2020-05-12:
i think this is the action for remote switch?   
[image]

bphermansson, 2020-05-12:
> Ok so I think this device is not in deconz at all.
>
> It will be something I will ask again soon, so to see your device in deconz API, you can use this kind of url
> `http://IP:PORT/api/API_KEY/sensors` [...]

You can get the API key by locking up the gateway and run a Rest client, or Curl:
curl --header "Content-Type: application/json" --request POST --data "{ \"devicetype\": \"\"}" http://192.168.1.15/api
The API gets back as username, "{"username":"D7nnnnnn"}}]".
This is explained at https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/.

NicolasBoulanger, 2020-05-12:
yes i have my key :)

NicolasBoulanger, 2020-05-12:
> You can get the API key by locking up the gateway and run a Rest client, or Curl:
> curl --header "Content-Type: application/json" --request POST --data "{ "devicetype": ""}" http://192.168.1.15/api [...]
yes i have my key :)

Smanar, 2020-05-13:
Ok so I have started https://github.com/Smanar/deconz-rest-plugin/commit/55ecd9a1dc7b72e48f41e2f922d47aafcb5d6371

But I have some problems for the table sensor.

Have you an Unix machine 32 bit to compile the code and make tries ?

The procedure is explained here > https://github.com/dresden-elektronik/deconz-rest-plugin at "Install deCONZ development package (optional, Linux only)"

You just need to change the step 1 by
`git clone --single-branch --branch osram https://github.com/Smanar/deconz-rest-plugin.git`
or use 
`git checkout -b livolo HEAD`
at step 2

But I don't know github too much, I can be wrong, but I already have a PR so I have make a special branch for this device.

Ok so the new code will include the device, you will probably have some issue with braodcast command (some bulbs in your network will react to the remote). You will have a new sensor, a ZHAswitch with "bouttonevent" field.

If you enable the deconz log (like in your second post) you will see something like 
>14:54:41:654 no button handler for: XXXXXX sensor ep: 0x01 cl: 0x0006 cmd: 0x42 pl[0]: 0x00
Everytime you press a button (for the missing one, but I think only 2 will work)
I just need the button (short/long/release) you have pressed and the message

bphermansson, 2020-05-14:
I've tried this and it's working. I installed Smanars patch and started a search for new switches in Phoscon. I see the switch in Deconz and in the logs:

13:11:05:886 0x00124B0014B71F47 onOff 0 --> 1
13:11:05:887 no button map for: PSMP5_00.00.03.16TC ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000
13:11:05:887 ZCL attribute report 0x00124B0014B71F47 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000

Smanar, 2020-05-14:
But to make it working I need all button value, I have some problem to guesss them, and it's easier with debug.
Just press a button and give me the log with "no button map" for missing one
You have Short press / long press / release for all 3 buttons.

NicolasBoulanger, 2020-05-15:
@bphermansson can you did it ? you have this device too ?
thanks

Smanar, 2020-05-15:
I can make explanation if you need @NicolasBoulanger ?
You have some problem with your OS ?

bphermansson, 2020-05-15:
I have the Osram remote that is black in rubber. It has three button, large and small arrow and a circle. The output I get from Deconz is noisy and I only see one of the buttons, the small arrow. It gives this on a short press:
18:08:35:642 no button map for: PSMP5_00.00.03.16TC ep: 0x01 cl: 0x0702 cmd: 0x0A pl[0]: 000
Can't see no difference on a long press. Can some values from Zigbee2Mqtt help?

Smanar, 2020-05-15:
I think I have values from zigbee2mqtt @NicolasBoulanger give a link for that https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2763#issuecomment-627491001

But it's realy faster using debug ^^. I will try better tommorow

But I don't understand why you have only the small arrow and not the big, I can understand for the "O" but the arrow ...
Perhaps It is already working, this debug message is only for not working button.
And there is something strange too

`cl: 0x0702 `

I will try to make better using this mapping

Smanar, 2020-05-15:
And I m stupid.

Better to check working values before the no working one ^^.

Pls can you check in the json (or somewhere else) the "buttonevent" value, if the value is "none" I m totaly wrong, if you have something like X00Y I have at least I good value in my table.
This field is the last pressed button.

Smanar, 2020-05-16:
So I have used this message
`13:11:05:887 no button map for: PSMP5_00.00.03.16TC ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000`

So now I m using 1 endpoint by button, but always cluster 0006 and command 0x0A.
I m almost sure I miss the cluster 0008 but IDK wich one command to use.

NicolasBoulanger, 2020-05-18:
is good for you with these logs ? 

11:01:41:569 no button map for: SML001 ep: 0x02 cl: 0x0400 cmd: 0x0A pl[0]: 000
11:01:48:142 no button map for: SML001 ep: 0x02 cl: 0x0406 cmd: 0x0A pl[0]: 000
11:01:53:081 no button map for: SML001 ep: 0x02 cl: 0x0400 cmd: 0x0A pl[0]: 000
11:02:01:120 no button map for: RWL021 ep: 0x02 cl: 0x0001 cmd: 0x0A pl[0]: 021
11:02:17:712 no button map for: RWL021 ep: 0x02 cl: 0x0001 cmd: 0x0A pl[0]: 021
11:02:34:469 no button map for: RWL021 ep: 0x02 cl: 0x0001 cmd: 0x0A pl[0]: 021

Smanar, 2020-05-18:
Bas luck, RWL021 is a philips dimmer and SML001 is a HUE motion sensor ^^.

And now I m looking better, I have based my table on 
>13:11:05:887 no button map for: PSMP5_00.00.03.16TC ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000
13:11:05:887 ZCL attribute report 0x00124B0014B71F47 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000

But It's not the good device too 0x00124B0014B71F47 is not in the good MAC range adress.

The log I need will contain "no button handler for: Lightify Switch mini" (not sure for typo).

Smanar, 2020-05-20:
No-one can compile the code and give me some logs ?

bphermansson, 2020-05-20:
Maybe later tonight I can give it a try.

bphermansson, 2020-05-21:
Some new findings. Now I see my switch as 0x75B7 in Deconz. It's a Osram Lightify Switch Mini. Pushing the circle gives: 

07:57:23:092 APS-DATA.indication srcAddr: 0x75b7, srcEp: 0x03 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 167, rssi: -70
07:57:24:083 APS-DATA.indication srcAddr: 0x75b7, srcEp: 0x03 dstAddrMode: 2, profile: 0x0104, cluster: 0x0300, lqi: 167, rssi: -70

(and my Hue lamp switches on)

Pushing the large arrow:

08:01:20:906 APS-DATA.indication srcAddr: 0x75b7, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 151, rssi: -72
08:01:21:002 APS-DATA.indication srcAddr: 0x192c, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -34
08:01:21:003 no button map for: PSMP5_00.00.03.16TC ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000

The small arrow:
08:03:33:218 APS-DATA.indication srcAddr: 0x75b7, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 159, rssi: -71
08:03:33:300 APS-DATA.indication srcAddr: 0x192c, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -35
08:03:33:302 no button map for: PSMP5_00.00.03.16TC ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000
08:03:33:302 ZCL attribute report 0x00124B0014B71F47 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000

NicolasBoulanger, 2020-05-21:
> Some new findings. Now I see my switch as 0x75B7 in Deconz. It's a Osram Lightify Switch Mini. Pushing the circle gives:
>
> 07:57:23:092 APS-DATA.indication srcAddr: 0x75b7, srcEp: 0x03 dstAddrMode: [...]

Did you try long press on both buttons?
Thanks !

Smanar, 2020-05-21:
But are you sure PSMP5_00.00.03.16TC is the osram mini switch ?
Because on a previous test ha has  0x00124B0014B71F47 as MAC adress and Osram mini start with 0x000d6f

Smanar, 2020-05-21:
Hello, what is the MAC adress of your osram remote ?

NicolasBoulanger, 2020-05-21:
[image]
i found this image for phoscon app ?

Smanar, 2020-05-21:
But phoscon is closed source.
We can't make something on our side.

bphermansson, 2020-05-23:
I made some log files now. First I press the buttons one by one, long press and short press. The second log is just one short push on one button.

https://pastebin.com/Pb8j85z3

https://pastebin.com/VmFCnvEU

[image]

Smanar, 2020-05-23:
Nice thx.
So there is a problem on my code, button event are not working at all. Need to check more closely the log.

Smanar, 2020-05-23:
Ok, sorry for manipulation, but I have a problem in the code.

I have make a new version with some debug line. But theses lines will be displayed only if you use --dbg-info=2 in your command.

Idk if you have a slider on your application or something to raise the log level ?

If you can send me the same log when you press some button ?

olicooper, 2020-05-23:
No idea if it'll help but I started attempting to add support for the Smart+ mini switch too (I gave up as I didn't have the time). Anyway, here is my commit which might prove useful? https://github.com/olicooper/deconz-rest-plugin/commit/70e42e9da35a6774dd79970a4a55e987a3e7d238

Smanar, 2020-05-23:
Yep, thx, I will take some lines to use in mine.

ATM I have some result on the 4 buttons mini switches, so I don't see why the 3 button will not work.

bphermansson, 2020-05-24:
I use the command line as described in the Deconz doc for adding new devices. I ran the below command and pushed all buttons, long press and short. From what I can see in the product documentation you can use the remote for six functions, three buttons, long or short push.
The command:

deCONZ --dbg-info=2 --dbg-aps=1 --dbg-zcl=1 --dbg-zdp=1 --http-port=80 > osram.txt

[osram.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4673144/osram.txt)

Smanar, 2020-05-24:
Ok so now I understand why I can never see the good debug logging.
There is something that missed, your device is not in the API, so it doesnt handled button event.

Can you try to re-include the remote using Phoscon pls ?

bphermansson, 2020-05-24:
Ok, I pretty sure I'm getting correct values now. 

```
Circle long - 21:28:14:228 APS-DATA.indication srcAddr: 0x89ef, srcEp: 0x03 dstAddrMode: 2, profile: 0x0104, cluster: 0x0300, lqi: 103, rssi: -78
Circle short - 21:29:34:605 APS-DATA.indication srcAddr: 0x89ef, srcEp: 0x03 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 119, rssi: -76
Little arrow short: 21:31:07:949 APS-DATA.indication srcAddr: 0x89ef, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 87, rssi: -80

```
The cluster-value indicates which button is pressed?

olicooper, 2020-05-25:
@Smanar I am trying to get the switch mini to work so I have created my own repo from your code  (see [my deconz repo](https://github.com/olicooper/deconz-rest-plugin/commits/master)) and run the compiled code, but now the clusters are not showing in the deconz GUI for the device and I don't have any useful logs showing the device commands etc.
This is an example log when pressing a button:
```
08:51:03:276 APS-DATA.indication srcAddr: 0xfda1, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 151, rssi: -72
08:51:03:276 APS-DATA.indication from child 0xFDA1
08:51:03:277 verify 0x000d6f00XXXXXXXX is child node after 839916 s
08:51:03:299 MyDebug 1
08:51:03:453 Node 0x000d6f00XXXXXXXX is known by 1 neighbors, last seen 0 s
```
I am not familiar with what is happening in the code, so my additions may have caused an issue. Do you know what might be wrong? 
How can I help to get the button mapping configured?

Smanar, 2020-05-25:
@bphermansson yes, but I need more than the cluster, I need to the command and the parameter too, It's for that I m using the "no button map" output debug, all informations are here.

>08:03:33:302 no button map for: XXXX ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000

@olicooper on the file sensor.cpp line 1315, you have disabled all the sensor value table.
And the easier way is looking for "no button map", like I have said before, you will have all value in one debug line.

I have advanced more for the 4 button, I think I have all value for the table, and I have started to mix the created device, if you want to test the code, I think it will be the same for the 3 button device.

olicooper, 2020-05-25:
Thanks for taking a look! I've started understanding it a bit more and I was hoping that by disabling sensor.cpp line 1315 I would see 'no button map' in the logs. I will re-enable it.

For some reason inside `da_web_plugin.cpp` this line isn't returning the sensor node `sensorNode = getSensorNodeForAddress(ind.srcAddress());` ([see this](https://github.com/olicooper/deconz-rest-plugin/blob/00376fcfd59e7e63ea8b60ff4a8aa976f6b7961f/de_web_plugin.cpp#L731-L735)). It isn't getting to the "MyDebug 0.2" log. I've tried resetting the device.

Smanar, 2020-05-25:
Yep but this is not trigger in your situation (for the moment).
For the moment when this part of code is trigger you have a sensorNode, so all the part that start at line 728 is not used.

I have started to put code in it for future ^^
On second step (It's that I have do for the 4 button) I will mix all endpoint, so it's possible the API don"t find a sensorNode, so it will search on the other endpoint to find it. But this part is not used yet for the 3 buttons.

olicooper, 2020-05-25:
@bphermansson Do you also have a "Climax Power Switch" connected? I've looked at your logs and I think the "no button map" logs are related to this device rather than the Lightify Switch Mini. During my own testing I am finding that I can't find any "no button map" logs for my switch - but I get it for other devices.
The MAC address `0x00124B0014B71F47` is for a Texas Instruments device. I think you should find logs with a mac address that starts with `0x000D6F` which is the switch mini (the chip is made by Ember Corporation).

olicooper, 2020-05-25:
@Smanar I don't think it has found a sensorNode by line 728 though from what I see in my log files:
```
16:24:58:814 APS-DATA.indication srcAddr: 0xc6f2, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 191, rssi: -67
16:24:58:814 APS-DATA.indication from child 0xC6F2
16:24:58:814 verify 0x000d6f00XXXXXXXX is child node after 839916 s
16:24:58:843 MyDebug 0.1
16:24:58:843 MyDebug 1
```

Smanar, 2020-05-25:
Can you show me the device created by the API for this device (the JSON) after the inclusion, you will have at least 3 devices.

olicooper, 2020-05-25:
Its working! This is the latest mapping: https://github.com/olicooper/deconz-rest-plugin/commit/5aeb0a89276d2771f2ca398d76df6e7279acb6c5

Unfortunately the `down long release` mapping isnt working. the logs say it is not bound even though it is correct in the button map. Any thoughts?

```
19:36:52:722 APS-DATA.indication srcAddr: 0x92c2, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 119, rssi: -76
19:36:52:723 APS-DATA.indication from child 0x92C2
19:36:52:724 MyDebug 1
19:36:52:724 MyDebug 2
19:36:52:724 MyDebug 3
19:36:52:724 MyDebug 4
19:36:52:724 MyDebug 6
19:36:52:724 Force binding of attribute reporting for sensor Lightify Switch Mini
19:36:52:724 no button handler for: Lightify Switch Mini ep: 0x02 cl: 0x0008 cmd: 0x03 pl[0]: 0x00
```

Smanar, 2020-05-25:
Yes, I have exactly same problem on the 4 buttons one > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/374#issuecomment-633254527

I think we need to use more debug line in the fonction checkSensorButtonEvent(), and for me the problem can be from 

```
            else if (ind.clusterId() == LEVEL_CLUSTER_ID &&
                       (zclFrame.commandId() == 0x03 ||  // stop
                        zclFrame.commandId() == 0x07) )  // stop (with on/off)
            {
                ok = false;
                if (buttonMap->zclParam0 == sensor->previousDirection) // direction of previous move/step
                {
                    sensor->previousDirection = 0xFF;
                    ok = true;
                }
            }
```

We have cluster 0008, command = 3 so this part can set ok to false again. And I have same bug on same cluster same command.

Yep, to disable broadcast message, if I m right we can do that with biding the remote to the coordinator, we can check with Xiaomi opple method.

And for the last question, IDK if it's a a good idea, ATM my objective is making a device in the api that can send all buttons used, to use the remote with an application. For the moment I haven't enabled the group feature

olicooper, 2020-05-25:
For the broadcast issue, a developer from zigbee2mqtt said they had fixed it: https://github.com/Koenkk/zigbee-herdsman-converters/commit/7756df8ccb314c9c52c89b0640107573f1aa9bdc

I added payload debugging. When I first press the down button (hold action) I noticed there is payload data `0126`. There is no payload for the release action. What does this mean and can we use this? I assume the first byte is the value of `pl[0]` but what is the second byte `26`?
```
22:15:24:114 APS-DATA.indication srcAddr: 0x92c2, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 103, rssi: -78
22:15:24:115 APS-DATA.indication from child 0x92C2
22:15:24:115 verify 0x000d6f00XXXXXXXX is child node after 839916 s
22:15:24:116 MyDebug 1
22:15:24:116 MyDebug 2
22:15:24:116 MyDebug ZCL attribute report 0x000D6F00XXXXXXXX for cluster: 0x0008, ep: 0x02, frame control: 0x11, mfcode: 0x0000
22:15:24:116    MyDebug payload: 0126
22:15:24:116 MyDebug 3
22:15:24:116 MyDebug 4
22:15:24:117 MyDebug 6
22:15:24:117 button 3001 Down long
22:15:24:117 Force binding of attribute reporting for sensor Lightify Switch Mini
```

olicooper, 2020-05-26:
@Smanar I have got the down long release working. There were two ways we could fix it - see [my latest commit](https://github.com/olicooper/deconz-rest-plugin/commit/bc2fc1071ed4265712bcc19032c3a1792f7c5bc6) - but I have gone for changing the buttonMap param to `0x01` which seems cleaner?

olicooper, 2020-05-26:
@bphermansson if you install my patch the button mappings should be working now and it'll be good to see if it works for you too. This is the repo: https://github.com/olicooper/deconz-rest-plugin

bphermansson, 2020-05-26:
@olicooper I see your debug rows, should I see anything more?
```
11:30:09:359 APS-DATA.indication srcAddr: 0x53ce, srcEp: 0x03 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 183, rssi: -68
11:30:09:359 APS-DATA.indication from child 0x53CE
11:30:09:360 verify 0x000d6f000e1b8d80 is child node after 785732 s
11:30:09:368 MyDebug 0.1
11:30:09:368 MyDebug 1
11:30:09:382 verify neighbor status: APP_SUCCESS (0x00) 
11:30:09:409 Node 0x00124B00167E5F97 is known by 1 neighbors, last seen 0 s
11:30:09:648 Mgmt_Lqi_req zdpSeq: 201 to 0x086BD7FFFE064F13 start index 3
11:30:09:648 APS-DATA.request id: 8, addrmode: 0x03, addr: 0x086bd7fffe064f13, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
11:30:09:696 APS-DATA.confirm id: 8, status: 0x00 SUCCESS
11:30:09:696 APS-DATA.confirm request id: 8 -> confirmed, timeout 29914440
11:30:09:776 APS-DATA.indication srcAddr: 0xf0d8, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8031, lqi: 151, rssi: -72
11:30:09:776 APS-DATA.indication request id: 8 -> finished
11:30:09:776 APS-DATA.request id: 8 erase from queue
11:30:09:776 ZDP status = 0x00 -> SUCCESS
11:30:09:776 void deCONZ::zmNode::setFetched(deCONZ::RequestId, bool) fetched item: 8, node: 0xF0D8
11:30:09:776 ZDP Mgmt_Lqi_rsp zdpSeq: 201 from 0x086BD7FFFE064F13 total: 5, startIndex: 3, listCount: 2
11:30:09:776     * neighbor: 0x000D6FFFFEDABE85 (0xBFAE), LQI: 35, relation: 0x02 rxOnWHenIdle: 1
11:30:09:776     * neighbor: 0x000D6FFFFEAAC57B (0x5DE6), LQI: 95, relation: 0x01 rxOnWHenIdle: 0
11:30:09:889 Node 0x000D6FFFFEDABE85 is known by 5 neighbors, last seen 34 s
11:30:09:980 poll node 00:21:2e:ff:ff:04:0c:91-01
11:30:09:980 Poll light node Unknown 5
11:30:10:327 MAC Poll 0x02 0x53CE
11:30:10:346 APS-DATA.indication srcAddr: 0x53ce, srcEp: 0x03 dstAddrMode: 2, profile: 0x0104, cluster: 0x0300, lqi: 159, rssi: -71
11:30:10:347 APS-DATA.indication from child 0x53CE
11:30:10:356 MyDebug 0.1
11:30:10:356 MyDebug 1
11:30:10:369 Node 0x000D6F000E1B8D80 is known by 1 neighbors, last seen 0 s
11:30:10:477 Daylight now: goldenHour1, status: 160, daylight: 1, dark: 0
11:30:10:848 Node 0x00178801026FF283 is known by 4 neighbors, last seen 6 s
11:30:10:885 poll node 00:12:4b:00:14:b7:1f:47-01-0702
11:30:10:885 Poll ZHAConsumption sensor node Consumption 2
11:30:11:146 Poll APS request to 0x00124B0014B71F47 cluster: 0x0702 dropped, values are fresh enough
```

olicooper, 2020-05-26:
I've attached an example of me pressing the down button.
The line after `MyDebug 6` will look like this: `button <button-number> <button-command>` .
If you press each button you should see the correct button being described.
```
02:56:30:052 APS-DATA.indication srcAddr: 0x7565, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 175, rssi: -69
02:56:30:053 MyDebug 1
02:56:30:053 MyDebug 2
02:56:30:053 MyDebug ZCL attribute report 0x000D6F00XXXXXXXX for cluster: 0x0008, ep: 0x02, frame control: 0x11, mfcode: 0x0000
02:56:30:053    MyDebug payload: 0126
02:56:30:054 MyDebug 3
02:56:30:054 MyDebug 4
02:56:30:054 MyDebug 5.3 mode: 1 ep: 0x02 cluster: 0x0008 cmd: 0x01
02:56:30:054 MyDebug 6
02:56:30:058 button 3001 Down long
02:56:30:059 Force binding of attribute reporting for sensor Lightify Switch Mini (2)
```

bphermansson, 2020-05-26:
Hmm I don't see that...
```
$ git checkout bc2fc1071ed4265712bcc19032c3a1792f7c5bc6
Note: checking out 'bc2fc1071ed4265712bcc19032c3a1792f7c5bc6'.
...
HEAD is now at bc2fc10 Osram switch mini - Alternative long press down release handling
$  qmake && make -j2
make -f Makefile.Release
make[1]: Entering directory '/home/pi/Downloads/olicooper/deconz-rest-plugin'
make[1]: Nothing to be done for 'first'.
make[1]: Leaving directory '/home/pi/Downloads/olicooper/deconz-rest-plugin'
$  sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
$ deCONZ --dbg-info=2 --dbg-aps=1 --dbg-zcl=1 --dbg-zdp=1 --http-port=80 
```

```
12:47:40:625 Node 0x00124B0014B71F47 is known by 1 neighbors, last seen 0 s
12:47:41:104 Node 0x000D6FFFFEAAC57B is known by 1 neighbors, last seen 17 s
12:47:41:189 APS-DATA.indication srcAddr: 0x53ce, srcEp: 0x03 dstAddrMode: 2, profile: 0x0104, cluster: 0x0008, lqi: 255, rssi: -55
12:47:41:189 APS-DATA.indication from child 0x53CE
12:47:41:189 verify 0x000d6f000e1b8d80 is child node after 785732 s
12:47:41:190 MyDebug 0.1
12:47:41:190 MyDebug 1
12:47:41:206 verify neighbor status: APP_SUCCESS (0x00) 
12:47:41:325 APS-DATA.indication srcAddr: 0x2e61, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0x0500, lqi: 255, rssi: -46
12:47:41:326 IAS Zone Status Change, status: 0x0031, zoneId: 100, delay: 0
12:47:41:326 MyDebug 0.1
12:47:41:326 MyDebug 1
12:47:41:472 poll node 00:21:2e:ff:ff:04:0c:91-01
12:47:41:472 Poll light node Unknown 5
12:47:41:584 Node 0x000D6FFFFEF1370B is known by 5 neighbors, last seen 10 s
12:47:42:065 Node 0x000B57FFFE99F616 is known by 1 neighbors, last seen 32 s
12:47:42:158 MAC Poll 0x02 0x53CE
12:47:42:174 APS-DATA.indication srcAddr: 0x53ce, srcEp: 0x03 dstAddrMode: 2, profile: 0x0104, cluster: 0x0300, lqi: 255, rssi: -55
12:47:42:174 APS-DATA.indication from child 0x53CE
12:47:42:175 MyDebug 0.1
12:47:42:175 MyDebug 1
12:47:42:336 poll node 00:12:4b:00:14:b7:1f:47-01-0702
12:47:42:336 Poll ZHAConsumption sensor node Consumption 2
12:47:42:545 Node 0x000B57FFFE27551E is known by 0 neighbors, last seen 0 s
12:47:42:577 Poll APS request to 0x00124B0014B71F47 cluster: 0x0702 dropped, values are fresh enough
```

olicooper, 2020-05-26:
Maybe try re-adding your device? I was able to use Phoscon web UI.

This is the steps I used to re-add the device:
1. Press and hold the middle and down buttons for ~10 seconds until the light stops flashing
2. Click add new sensor in Phoscon
3. Press and hold the middle and up button for ~3seconds until the light stops flashing
4. Wait for Phoscon to say the sensor is added

Smanar, 2020-05-26:
So much thing to read, sorry if I forget something.

So natively the remote make as bind
- Power Configuration to the gateway
- Poll Control to the gateway
- On/Off to the bulb
- Level Control to the bulb

And to disable braodcast they make
- Power Configuration to a group for endpoint 1
- On/Off to the gateway for endpoint 1 and 2
- Level Control to the gateway for endpoint 1 and 2
- Color control to the gateway for endpoint 3

ATM I m making only the 3 lasts but for all endpoint, so for me it's enought to stop brodcast command ...

But I m checking the code for the relase problem, you have made a rollback of your code to chnage only the parameter, it's enought to make the device working ?

bphermansson, 2020-05-26:
Oh, I have trouble with the vocabulary... I tried to add it as a switch in Phoscon.
But now, great success! It shows up in the logs and I can even see it in Home Assisstant :)

```
Event 1 fired 6:44 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "lightify_switch_mini_8",
        "unique_id": "00:0d:6f:00:0e:1b:8d:80",
        "event": 3002
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-26T16:44:41.667421+00:00",
    "context": {
        "id": "9cfc44280cd945b3a30176aad6062c0a",
        "parent_id": null,
        "user_id": null
    }
}
Event 0 fired 6:44 PM:
{
    "event_type": "deconz_event",
    "data": {
        "id": "lightify_switch_mini_7",
        "unique_id": "00:0d:6f:00:0e:1b:8d:80",
        "event": 2002
    },
    "origin": "LOCAL",
    "time_fired": "2020-05-26T16:44:10.616804+00:00",
    "context": {
        "id": "4c54cd080d92402ca54a200fd433193e",
        "parent_id": null,
        "user_id": null
    }
}
```

bphermansson, 2020-05-26:
Now I just wonder why the Osram switches on and off my Climax device.

Smanar, 2020-05-26:
>08:01:20:906 APS-DATA.indication srcAddr: 0x75b7, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 151, rssi: -72
08:01:21:002 APS-DATA.indication srcAddr: 0x192c, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -34
08:01:21:003 no button map for: PSMP5_00.00.03.16TC ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000
The small arrow:
08:03:33:218 APS-DATA.indication srcAddr: 0x75b7, srcEp: 0x02 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 159, rssi: -71
08:03:33:300 APS-DATA.indication srcAddr: 0x192c, srcEp: 0x01 dstAddrMode: 2, profile: 0x0104, cluster: 0x0006, lqi: 255, rssi: -35
08:03:33:302 no button map for: PSMP5_00.00.03.16TC ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000
08:03:33:302 ZCL attribute report 0x00124B0014B71F47 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000

He is sensible to Cluster 0006 so it s because broadcast command, and I don't think a group one.
Do you know how to bind / unbind device using deconz ?

olicooper, 2020-05-26:
>So natively the remote make as bind
> * Power Configuration to the gateway
> * Poll Control to the gateway
> * On/Off to the bulb
> * Level Control to the bulb

This is what the official Osram mobile app does when the switch buttons are configured by the user. I found out this information [from here](https://github.com/Koenkk/zigbee2mqtt/issues/962#issuecomment-478301484).

>But I m checking the code for the relase problem, you have made a rollback of your code to change only the parameter, it's enought to make the device working ?

Yes, just by changing the down button release `param` binding to `0x01` instead of `0x00` it has fixed the issue. This might be the same for the 4 button switch. Add [this debug output](https://github.com/olicooper/deconz-rest-plugin/blob/bc2fc1071ed4265712bcc19032c3a1792f7c5bc6/de_web_plugin.cpp#L3552) to your code and look at the part that says `sPrevD:` to see if it output `0x01`. I really think it should be `0x00` but it isn't and that is why the binding doesn't work.

When I bound the clusters (OnOff,LevelControl, etc.) in deCONZ it stopped the switch broadcasting to all lights (I could control a single light directly) but it also stopped the debug logs appearing so something wasn't right. I had to delete and re-add the switch to get the debug messages to appear again.

Smanar, 2020-05-26:
So yes Gz, using 0x01 is realy better that the "hack" I had in my mind ^^.

The code use all theses 3.
And the code need them, if you look at the device it havent usefull input cluster (the blue one), all usefull cluster on/off, level, color are outputcluster, so we need to bind the conbee and the remote, so it send the command to the conbee and the conbee can manage them.

If you disable all binds, and make them to a bulb, the conbee can't detect them.

[89 more replies]

closed 2020-07-10 by manup