dresden-elektronik/deconz-rest-plugin #2764: Xiaomi Aqara temperature/humidity/pressure stops reporting humidity and pressure

x0010, 2020-05-12:
I have 10+ of these Aqara sensors (the square ones) and a couple of them has stopped updating some of their values. 

The temperature value is changing and updating but the humidity and air pressure value are stuck since a week back. The other sensors are reporting as they should. I'm using node-red and I alway get 3 values after each other (temp/hum/pres) when I listen to these on the websocket node. But with these broken ones I only get temperature.

I have tried changing to new batteries but that didn't make any difference. I will try to remove them and add them again to my conbee1 to see if that makes any difference but it feels weird if it would be such.

I just wanted to know if anyone else had any similar problem and if it can be solved or if I should see them as permanently broken and replace them?

SwoopX, 2020-05-12:
Eventually, the've not been paired correctly. Check the REST API if all Xiaomi wether sensors have 3 sensors listed under `/sensors`. If not delete and re-pari them.

x0010, 2020-05-12:
> Eventually, the've not been paired correctly. Check the REST API if all Xiaomi wether sensors have 3 sensors listed under `/sensors`. If not delete and re-pari them.

They are there, all 3, but only the temperature has been updated today while the 2 others has not since the 7th May, 5 days ago.
I've had those 2 working fine for 1½ year and now suddenly they both stopped updating humidity and pressure.
Funny thing is that they those 2 sensors that stopped working, both of them stopped the exact same time, within a few minutes so my feeling is that it can't be hardware error. Maybe some update on my server? I will try some more research on that, but thank you for your idea!

Smanar, 2020-05-13:
And other thing, Xiaomi device use 2 notifications.
The normal one, one by cluster, and a periodic one using cluster ff01 and on this one there is all values (temperature/pression/humidity) on the same request, not possible having only temperature.

When you are saying values are not updated, it mean, they have not changed or the device haven't send values since long time (and perhaps always the same value) ?

SwoopX, 2020-05-13:
I recall that I had a similar experience once with one of my sensors. Have just reset and paired it again instead of searching for the root cause. Pure pragmatism to spare hours of searching and trying.

x0010, 2020-05-13:
> I recall that I had a similar experience once with one of my sensors. Have just reset and paired it again instead of searching for the root cause. Pure pragmatism to spare hours of searching and trying.
> 
> > The normal one, one by cluster, and a periodic one using cluster ff01 and on this one there is all values (temperature/pression/humidity) on the same request, not possible having only temperature.
> 
> @Smanar That condition is indeed possible. Beside the special attribute reporting of the sensors, they also send a regular for each value individually.

I was thinking about that it was this and I tried it tonight, deleted them from phoscon and added paired them again and yepp, now every value is reporting again. 
So I guess my problem is solved and I know what to do if it happens again, but..it's interesting that two sensors, that both have been working well for 1½ year, within the same hour decide to stop reporting humidity and pressure (and not reporting them at all for 6 days) and only report temperature. It feels like it must have been something on the server/conbee side since the sensors can't talk to each other.

About the other reporting you are talking about, I guess the rest/websocket node in node-red only get the individual reporting (even though it get it directly after each other), it would be good to have one that includes all three values at the same time, as of now I'm using a merge node to collect all three values so I can save all of them into one post in my database.

SwoopX, 2020-05-16:
Well, you should consider it that way: There's an (more or less) ensured, let's say hourly reporting with all 3 values included and pushed by the API. The individual value reporting is dependent on the relative value changes, and therefore those are push irregularly by the API. The threshold is something like 5% or so. Not configurable as far as I can tell.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix