dresden-elektronik/deconz-rest-plugin #2767: Conbee 2 - updates

TungstenE2, 2020-05-13:
Hi all,

According to gateway menu my version 2.05.71is up to date.
I thought there is OTA update available. 

[image]

But now I see manual instructions and several versions.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

http://deconz.dresden-elektronik.de/deconz-firmware/?C=M;O=D

[image]

Which version is latest stable to be used with Conbee 2 on Raspberry Pi?

deCONZ_ConBeeII_0x26580700.bin.GCF
deCONZ_Rpi_0x26350500.bin.GCF

I am a bit confused about the updates. 

thx

Gizmo-Ger, 2020-05-13:
Those are the firmwares for several zigbee devices. Depending on which device you are using. In your case it´s a ConbeeII, an USB zigbee stick. The other one is an addon module for a Raspberry. Latest release for Phoscon is V2_05_75.

SwoopX, 2020-05-13:
> Latest release for Phoscon is V2_05_75.

That's the release fo deCONZ (which includes Phoscon).

The reason why the newer firmware versions are not suggested as updates is that they're still tagged as beta. However, this is expected to change soon.

TungstenE2, 2020-05-14:
thx for the info.

But I see V2_05_74_stable, which is also not shown in my Phoscon. Any idea why?

https://github.com/dresden-elektronik/deconz-rest-plugin/releases

TungstenE2, 2020-05-14:
just updated to  V2_05_74, but why was this update not listed in Phoscon app?

SwoopX, 2020-05-14:
> 
> 
> just updated to V2_05_74, but why was this update not listed in Phoscon app?

You meant to V2_05_75, right? Presumably, .75 is still tagged as beta. When you use deconz' repository via APT, it depends on which channel you've chosen as source, beta or stable.

TungstenE2, 2020-05-14:
As stated I updated to V2_05_74 as I subscribed to main.

My question was: while beeing on V2_05_71 the app sated I am on the latest version, while V2_05_74 was already available. So there is a bug in the info displayed regarding the latest version.

SwoopX, 2020-05-14:
ok, clear now. I'm not too sure though if the update notification is just for the firmware, deconz or both. If it's just firmware, it would be indeed beneficial to extend that to deconz version as well.

djashjones, 2020-05-15:
I thought there was going to be a major update this week?

danli003, 2020-05-18:
Yes, there hasn't been much updates lately. Does anyone have any info on what is going on?

Mimiix, 2020-06-12:
This is a phoscon related question. Moving it to their git.