dresden-elektronik/deconz-rest-plugin #2768: Installation deconz

Smanar, 2020-05-13:
Hello, ATM when you install deconz you have theses messages

```
Important: enable_uart=1 should be set in /boot/config
Important: /boot/cmdline must not contain console=serial0
```

But this is not only for Raspbee ?

manup, 2020-05-13:
Yes that's only relevant for RaspBee, I think a (RaspBee only) should be mentioned.

Smanar, 2020-05-13:
Same, because some user are realy surprised by this comment, and this modification can be compliced for some of them for the console command for exemple because of synthax, especially if it's useless.

Mimiix, 2020-10-21:
Re-opened. Still the case. Also: It provides a faulty url: 

https://github.com/dresden-elektronik/deconz-rest-plugin#software-requirements

I got this from a user today: 
`root@raspberrypi:/etc/apt/sources.list.d# apt-get reinstall deconz
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 0 B/6,093 kB of archives.
After this operation, 0 B of additional disk space will be used.
(Reading database ... 95288 files and directories currently installed.)
Preparing to unpack .../deconz_2.05.86-raspbian-buster-beta_armhf.deb ...
Important: enable_uart=1 should be set in /boot/config

Important: /boot/cmdline must not contain console=serial0

use raspi-config to configure serial port access as described in user manual
Unpacking deconz (2.05.86-raspbian-buster-beta) over (2.05.86-raspbian-buster-beta) ...
Setting up deconz (2.05.86-raspbian-buster-beta) ...
Important: enable_uart=1 should be set in /boot/config

Important: /boot/cmdline must not contain console=serial0

use raspi-config to configure serial port as described in documentation:
https://github.com/dresden-elektronik/deconz-rest-plugin#software-requirements`

Can we get this fixed? @manup

manup, 2020-11-17:
The text in v2.6.0 was changed to to inform that this applies only to RaspBee devices.

closed 2020-11-17 by manup