dresden-elektronik/deconz-rest-plugin #2770: disable router function

LukaszP2, 2020-05-13:
Hi

I got problem with my wall switchs. In VNC I see it's a router. When I add new device it connecting to them but for few minuts/hour its disconnected. 
How to turn off router function in it?

SwoopX, 2020-05-13:
Typically, mains powered devices are routers. The device type is determined base on the information provided in the node descriptor response packages. The corresponding data comes from the device firmware, so no way to change it.

The behaviour you're experiencing should have nothing to do with the device type. Please provide more information about the devices in question (Brand, Model, modelID, etc.).

Smanar, 2020-05-14:
Just by curiosity, it' s a legrand or Osram device, and the device that disconnect are Xioami ?

It's not possible to turn on or off the router feature but @pipiche38 have find a good tips.
Instead of sending the permit join request in broadcast, it's possible to send it at only one router, so you can select wich one device you want to use as router.

LukaszP2, 2020-05-14:
End devices its aqara. Switch is lonsonho (
https://a.aliexpress.com/_BOtzDk ) 
[image]
[image]

Smanar, 2020-05-14:
Ha, tuya and Xiaomi ? First time I m seing this issue
You have some informations here https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623
But nothing about tuya.

LukaszP2, 2020-05-17:
I disconnect all this switch and everything was ok.  I reconnected aqara sensor direct to conbee. After 12h I turn on switch. Only motion detectors (7 from 9) reconnected to switch. The rest of sensors: windows (15), temperature (8) and battery switch (2) works ok.
A have two Aqara wire switch (LLKZMK11LM) maybe this help if I add them somewhere...

Smanar, 2020-05-18:
>Only motion detectors (7 from 9) reconnected to switch
You mean they was directly connected to the conbee, and have leave it to reconnect them using the switch, 12h after ?

LukaszP2, 2020-05-18:
Exactly.. 1-12h and they change conbee to switch

Smanar, 2020-05-18:
WTF ?
And it's Aquara sensor ? You are looking the mesh on deconz ?

LukaszP2, 2020-05-18:
Yes aqara motion.
Im looking on vnc

Smanar, 2020-05-18:
If you don't know where is it, I don't think you have used it ^^.

But it's realy strange, xiaomi device never do that, they prefer leave than using another way.

LukaszP2, 2020-05-18:
i forgot about permint join. I will try add with that.

Smanar, 2020-05-19:
No, don't worry, you can't add device without that, when you use in phoscin "add new device" it will enable the permit join for some minut.

LukaszP2, 2020-05-20:
still nothing ;/ I think I sell this switch and buy some think else....

Smanar, 2020-05-20:
I haven't find similar issue with Xiaomi<>Tuya.
But you are the first one I see having Xiaomi device that change router .....

LukaszP2, 2020-05-23:
I buy CC2531 with Zigbee2MQTT and have the some problem. I add montion detector and about 11:30 we go outside. We back about 13:30

[image]

this is link quality ... 

[image]

Smanar, 2020-05-23:
Ok, so we can add tuya to problematic device with Xiomi ....

LukaszP2, 2020-05-23:
I just upgrade 2.05.76 maybe this can help...

Mimiix, 2020-06-12:
@LukaszP2 Is this fixed?

LukaszP2, 2020-06-12:
X3 33xd w d

closed 2020-06-12 by LukaszP2