dresden-elektronik/deconz-rest-plugin #2772: operating distance - mash network

horchi, 2020-05-14:
Hello,

not directly an issue with deconz but may you have a idea and can help regarding this.

I try to use a OSRAM gardenpole like this https://www.amazon.de/Osram-Gartenleuchte-tageslicht-Schutzklasse-kompatibel/dp/B074Q144J1/ref=sr_1_3?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&dchild=1&keywords=gardenpole&qid=1589435290&sr=8-3.
Inside the house it works fine, but when I position it outside (in about 25m distance to the ConBee I with direct view to it thru a window) it stops working. But a other Zigbee device (TRADFRI bulb GU10 W 400lm) at the same location is working fine.  
Shouldn't this bulb repeat the signal? Or did I have to adjust a setting or something else to activate this signal repeating functionality?
Thx and best regards Jörg

ebaauw, 2020-05-14:
> in about 25m distance to the ConBee

25m is about 2.5 times the direct line-of-sight range of a Zigbee device.  I don't know what glass your window has, but HR+ glass will impact the radio signal.

> Shouldn't this bulb repeat the signal?

Yes, it should.  It's probably not in direct range of the ConBee either, so there's probably one or more ZigBee routers in between, relaying the messages from the ConBee to this bulb.

I have more or less the same setup as you, and experience the same issue.  I have an OSRAM GardenSpot Mini RGB on my terrace.  Not 25m, but the radio unit is outside.  I also have a Hue LTW010 light outside, above the terrace door.  The Hue light can be controlled without any issue, the GardenSpot rarely.

I find the GardenSpot does react reliably to broadcasts.  Recalling scenes or issuing group commands works as expected.  This is indicative of routing issues on the Zigbee network: the ConBee doesn't know how to reach the GardenSpot through unicast.  Unfortunately, different vendors handle routing differently, and the ConBee doesn't always find a new route.  OSRAM (Ledvance) and IKEA seem to be least compatible.  Hopefully, some more workarounds can be created in the ConBee firmware, see #1261.

horchi, 2020-05-14:
thx, than I can't use this light actually if there's now workaround.
The window has normal standard glass.
I just started the deConz with the GUI to see the mashed network but is isn't meaningful for me until I can't identify the devices without there name or type.
It seems deconz looks a bit beta at all and a detailed documentation is missing. 
Ma be I find a way to connect zigbee devices directly to homeassistent without the way round over deconz

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

horchi, 2020-06-28:
Yes is resolved, I actually have no problems anymore since I have deactivated Phoscon and the ConBee Stick and switched over to  [zigbee2mqtt.io](https://www.zigbee2mqtt.io/). This also provides a log file and I can see what is going on :) !
Best regards Jörg