dresden-elektronik/deconz-rest-plugin #2773: Sonoff Basiczbr3 issues

TungstenE2, 2020-05-14:
Hi all,

I am using V2_05_74 and Conbee2 with Sonoff Basiczbr3.

Issue 1: Basiczbr3 is listed as light but I should be switch. I know this was discussed before, but I does not make it better.

Issue 2: In Phoscon App state of Basiczbr3 is not correct. I am using FHEM and Android App HUE Essentials. All state switchening in HUE Essentials is successfull and shown in FHEM correct. In Phoscon App state in light group is not updated, and toggeling state by pressing the bulb symbol is not always changing state. Bulb symbol is only sometimes changing state, but is also changing symbol to off again after a couple of sec, while Basiczbr3 is still turned on. So Phoscon App is not showing correct state of connected device.

[image]

cs42, 2020-05-14:
> Issue 1: Basiczbr3 is listed as light but I should be switch. I know this was discussed before, but I does not make it better.

No, in Zigbee it fits better into "lights" because it is an output device. Maybe the category should be named rather "Outputs" than "Lights".

A switch on the other hand is an input device, this could even be merged with "Sensors"...

TungstenE2, 2020-05-14:
I do not agree with this definition.

There are lights. You can turn them on/off, dim, change color.

There are sensors. They give temperatur, motion, light, etc.

There are switches. They only turn on/off. Maybe with reading power details.

Smanar, 2020-05-14:
The code make only difference with "light" (generaly router) and "sensor" (generaly end device). Switch or sensor it's same.

You can send command to "light"
You can only receive command from "sensor/switch"

If the switch is battery powered, you can send order to it > sensor
If the switch is wired, you can send it order > light

And I know it s realy "popularized", in reality it's more complex than that.

cs42, 2020-05-14:
No. Switches **send a signal**, else it would be a relay. And there are lights that can't be dimmed or/and have a fixed color. So here "light" fits better as it does for switchable outlets. There might be better names for the categories and maybe there should be more categories but a device like the basiczbr3 is a very different device compared to things that are grouped in "switches" currently.

I think this is a historical remain from ZLL.

Smanar, 2020-05-14:
BTW for issue 2, you have problem in phoscon, but in FHEM too?

TungstenE2, 2020-05-14:
> 
> 
> BTW for issue 2, you have problem in phoscon, but in FHEM too?

yes.

Couple of weeks ago I integrated the basiczbr3 successfully into FHEM as HUEDevice combined with a doif and a Osram smart+ motion sensor.

First everything was fine. Then it started to go wired.
In past days I was not able to set it on or off any more via the device.
Today I played around a bit and found out that having a device on the phoscon group and not on the phoscon device did the trick. Strange enough, toggeling the group also toggels the device, but not vice versa. Also the state in phoscon does not change most of the time, while in HUE essentials app or FHEM state is correct. Also device appears as not reachable in phoscon while I am able to toggle it via FHEM and app.

Phoscon is not reliable at all in my setup.

I already point out an issue regading old controller showing up and not beeing able to get rid of it.

Also aquara window sensors are not reliable. State in phoscon and FHEM is different.

TungstenE2, 2020-05-15:
seems I found the reason of strange things happening recently:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2273

cllatser, 2020-05-24:
Good morning, I just started and BASICZBR3 recognizes me as a switch. It works very well. I have the HA in Rpi4 and the usb Conbee 2.
[image]
[image]

TungstenE2, 2020-05-24:
me not

[image]
[image]

cllatser, 2020-05-24:
The firmware version is different from mine. But the number and the date are the same. Thats weird.

Mimiix, 2020-06-12:
As this is Phoscon, i move it to their Git.