dresden-elektronik/deconz-rest-plugin #2774: New Device in Phoscon App: Blitzwolf/Tuya Temperature/Humidity with Heimann TS0201 #2442

kaybau, 2020-05-14:
Hi,
I have same device connected via Phosconn App like #2442 , After connecting App is telling me "Ready" but not showing the device.
In deConz adapter of Iobroker I see the values of temperature and humidity, but they are seldom updated. Deconz-App itself is seeing the device more regularly.
Will the device be included in Phosconn App in one of next releases? 
What can I do to get the transmitted data regularly into deConz adapter?

Best Regards,
Kay

SwoopX, 2020-05-14:
> Will the device be included in Phosconn App in one of next releases?

That is up to DE, as Phoscon is closed source. High chance that it won't.

> Deconz-App itself is seeing the device more regularly.
How that?

> What can I do to get the transmitted data regularly into deConz adapter?

That more or less depends on the capabilities of the device in question. If it supports attribute reporting (and also tis configuration), than it probably can be changed as desired. Check https://discord.com/channels/494922323518947329/704252638727897188/709120485862146161 (2nd part).
However, there are device which issue attribut reports, but are not configurable in this regard. In those cases, the answer is no.

kaybau, 2020-05-14:
Hi SwoopX,
thanks for quick reply. I assumed this github channel is covering all the dresden-elektronik stuff also phoscon .

Can you copy the part of discord chat to here or provide me access ?
Thanks
Kay

SwoopX, 2020-05-15:
There you go
https://discord.gg/QFhTxqN

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

Smanar, 2020-08-15:
Hello, there is some one else with battery problem and this device ? (less than 1 month)

kaybau, 2020-08-24:
Hi @Smanar ,
even if it is not related to conbee:
Yes, I have 2 devices. One is stable at 100% battery for appx.8-10 weeks. the second one I have to replace battery ervy 3-4 weeks. strange

Smanar, 2020-08-24:
@kaybau 
Yep, exactly same issue, I m seing 2 reasons
- Hardware problem, I have already this kind of bug on Xiaomi stuff.
- There is more pooling/reporting on one device than the other.

You can see one device more active than the other in deconz (with the LED on node title)
Have you some log enabled to check if there is something bad ?

jinjie, 2020-08-26:
Just got this device. Shows ready but not showing on phoscon. At least it got recognised in HA.

Not really a deal breaker, as long as I get the sensor values in HA, it works for me.

Smanar, 2020-08-26:
Yep, this is normal, not all device appear in Phoscon.

jinjie, 2020-08-26:
> Yep, this is normal, not all device appear in Phoscon.

Why is this considered normal? Don't understand your definition of normal in this case.

Smanar, 2020-08-26:
lol.
Phoscon is like HA, it s just an app that use the API.
To avoid problem with unwanted device, phoscon use a whitelist, and they autorise less device than HA, I think in HA all device will be visible.