dresden-elektronik/deconz-rest-plugin #2775: Some groups in Phoscon are missing

cs42, 2020-05-14:
Hi,

I know, this isn't a tracker for Phoscon, but maybe someone can help nevertheless.
 Since some time some of my groups are **missing** from Phoscon. This does not bother me too much as those groups are still working (I'm using Home Assistant as my smart home interface and I'm just using Phoscon for the very basic configuration). Those groups are still shown in the old web interface as well!

So my questions:
- What and how could I debug?
- Is there a better place to report such an issue with Phoscon?

Smanar, 2020-05-14:
Phoscon is closed source, so you can't do something.
But IDK if there is a better place for Phoscon issue ...

BTW, the missing group is perhaps voluntarily missing.

cs42, 2020-05-14:
Well, the groups have been there once (and created in Phoscon as well). ;-)

Smanar, 2020-05-14:
Ha yep, shutter group ?
Yes, I know this bug, hoping for correction in future version, realy boring.

cs42, 2020-05-25:
With 2.05.77 all groups are visible again.

closed 2020-05-25 by cs42