dresden-elektronik/deconz-rest-plugin #2776: How to change MAC

bastshoes, 2020-05-14:
I've moved from Conbee I to Conbee II via backup restore. Now to sticks have same MAC. It have negative impact on discovery process.

How can I change MAC on stick?

Smanar, 2020-05-14:
You can in deconz (In the GUI) in network setting.
But I don't understand the problem, your 2 conbee don't work in same time ? So they can have the same MAC.

bastshoes, 2020-05-14:
They work some how. But when I try to replace rests_api lib in my installation I can't login to web GUI. pwa app discovers second stick instead of my.

Smanar, 2020-05-14:
Have you tried with deleting browser cache ? or using another browser (or using iconito mode) ?

bastshoes, 2020-05-14:
Yes, I've tried all this. 3 different browsers. Incognito mode in all of them.
Starnge ting that https://phoscon.de/discover returns data of sescond stick only when I try to use new rest_api lib. If return old one back in 5 minutes in starts return rmy data

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix