dresden-elektronik/deconz-rest-plugin #2778: Phoscon didn't find the gateway

Galiwyx, 2020-05-15:
Hello, I'm trying to use a Conbee II USB stick on my RB+2 with Homebridge.

- I installed DeConz
- I plugin the Conbee II USB
- I rebooted the Raspberry

But when I go to : https://phoscon.de/pwa/ after ten minute I have this : 

[image]

I tried to go to http://192.168.X.X/PWA but it redirects to the homebridge login : 
[image]

I'm a beginner, Maybe I have forgot a step ? 
Thank you :)

Smanar, 2020-05-15:
And just by curiosity https://phoscon.de/discover ?

Galiwyx, 2020-05-15:
I tested with a pre-built deConz image for raspberry : https://phoscon.de/en/conbee2/sdcard

Discover returns : 
[image]

134 returns the same loading page:
[image]

Galiwyx, 2020-05-15:
I retested with a prebuilt deConz image from Phoscon and it seems okay now ! 
[image]
Thanky ou :à

Smanar, 2020-05-16:
And what is the difference beetween "pre-built deConz image for raspberry" and "prebuilt deConz image from Phoscon", it haven't worked on first, but ok on second ?