dresden-elektronik/deconz-rest-plugin #2780: Smart Switch PTM 215Z (EnOcean) not repeated by Hue Bulbs to Raspbee

sprauerc, 2020-05-16:
I would like to set up a Niko Smart Switch (based on EnOcean PTM 215Z) far away from the Controller. I have tried with Hue G10 bulbs, Innr bulbs and Ikea bulbs and the signal of the smart switch is not repeated.

I couldn't find much information of how Zigbee green power (ZGP telegrams) are repeated by light bulbs. Could this be linked to the bulb firmware ? Or the lack of green power support somewhere ?

The switch works very well when in direct connection with the Rasbpee II controller but when I go a few meters further from the controller (yet still close to light bulbs), it doesn't work.

This may not be Deconz related but thought I might ask.

ebaauw, 2020-05-16:
Oh, it is deCONZ related.  deCONZ doesn’t configure the bulbs as ZGP proxy, unlike the Hue bridge.  I don’t understand the details of what’s preventing deCONZ from doing this, apparently it’s related to obtaining the encryption key.

Effectively, ZGP switches work only when in direct reach of the coordinator.

manup, 2020-05-16:
@koenkk has figured out the encryption details https://github.com/dresden-elektronik/deconz-rest-plugin/issues/436#issuecomment-610545110 :rocket: 

The related commands are handled in the REST-API plugin but calculating the proper ZGP key isn't implemented yet. I think here we need to use some OpenSSL functions to calculate the key.

https://github.com/dresden-elektronik/deconz-rest-plugin/blob/master/permitJoin.cpp#L151-L157

ebaauw, 2020-05-16:
@manup, what's the ETA?  Anything I can do to help or test?

The link to the source in Koen's comment is broken.

Smanar, 2020-05-16:
@ebaauw I think the new link is here https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/greenPower.ts

manup, 2020-05-16:
> @manup, what's the ETA? Anything I can do to help or test?

The missing piece seems to be generating the key with OpenSSL / libcrypto. I figure it's only a few lines of code but dealing with these libraries isn't a joy :) I can't provide an ETA but would be glad if you can help figure out the puzzle.

On Unix the crypto libraries can be added to `de_web.pro` like this:

```
packagesExist(libssl libcrypto) {
    DEFINES += HAS_OPENSSL
    PKGCONFIG += libssl libcrypto
}
```

Then in `permitJoin.cpp` the following need to ported to C++:
https://github.com/Koenkk/zigbee-herdsman/blob/master/src/controller/greenPower.ts#L20-L42

I think the node.js crypto modules also just use libcrypto / libssl, so it would be interesting which functions they call under the hood.

Smanar, 2020-05-17:
Someone have started something ?
I have find an exemple with same code in js and c++ https://stackoverflow.com/questions/53918059/getting-different-values-when-encrypting-in-nodejs-versus-openssl-c
So I can make a try, but I have realy by memory with Openssl ...

Smanar, 2020-05-17:
Ok so I have finished a code that mimic the JS code, but I compile it on windows ATM

I have exactly same result on nodejs or C++

https://github.com/Smanar/Zigbee_firmware/blob/master/Encryption.cpp

Smanar, 2020-05-18:
I have cleaned the code, and make a fonction, just need to convert type to use Qt type.

manup, 2020-06-05:
Cool can you share the code?

raphirm, 2020-06-25:
@manup : @Smanar has updated his github file on the 18th. I assume the link is still the same 

https://github.com/Smanar/Zigbee_firmware/blob/master/Encryption.cpp

ebaauw, 2020-07-28:
Bloody bot!

posixx, 2020-07-29:
Time to implement then:-)

n30V13l, 2020-09-18:
Nice to see this issue in the backlog.
Just dropped my Hue Bridge in favor of RaspBee2/deConz and now our many FoH Switches installed in the house do not work properly any more, especially when my 2.4g router decides to switch to channel 11. My wife kills me.
Of course I looked into [compatibility list](https://phoscon.de/en/conbee2/compatible) before I did this. FoH Switches were in it. Would have been nice if there was a "*limited compatibility" hint.

Smanar, 2020-09-18:
Unfortunately, I think the GreenPower is not implemented yet.

posixx, 2020-09-18:
Yeah this is an issue which is almost 3 years old now.. Hopefully more and more people going to nag about it so it gets proper attention..

n30V13l, 2020-09-18:
> Yeah this is an issue which is almost 3 years old now.. Hopefully more and more people going to nag about it so it gets proper attention..

3 years? Especially the subject "Green Power" should be considered to have more attention in every company. Maybe I should tell Greta ;)

Mimiix, 2020-09-22:
Smanar asked in the dev channels. Manuel has this on the list for the next beta.

ICTSMART, 2020-11-15:
As there was a new stable release of Phoscon (addon in HA) version 2.05.88 I decided to retest this issue again.
Still, for this specific model (the Niko Smart Switch PTM 215Z)  the ZGP telegrams are still not relayed through the network.
My setup consists of a Raspbee I - latest firmware 26390500 running Phoscon/deCONZ version 2.05.88 / 10/15/2020.
When I increase the distance between the Niko Switch and the Raspbee, although multiple routers are in between, no messages are relayed.

Mimiix, 2020-11-16:
@ICTSMART Let's keep it all in the same place: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/436

Add your comment there if you can. I'll get Manup there again.

ICTSMART, 2020-11-16:
@Mimiix Thanks for pointing out. I added my comment on the location you suggested.

thomas-svrts, 2022-06-02:
was this defect now fixed or not? ( https://github.com/dresden-elektronik/deconz-rest-plugin/issues/436 was closed and merged so I assume this defect can be closed )
Got some issues with my Niko FoH switch and bumped into this open issue...

Mimiix, 2022-06-03:
> was this defect not fixed? #436 was closed and merged so I assume this defect can be closed.

Nice! Thanks :)

closed 2022-06-03 by Mimiix