dresden-elektronik/deconz-rest-plugin #2781: Deconz Combee 2 Update with proxmox

Ageo, 2020-05-16:
Hi All,

I have an issue to update my combee 2 key. It's a brand new one.

I'm on Debian 10 with proxmox also updated. I have this error:

I stop the service and do this command for update:
sudo GCFFlasher_internal -d /dev/ttyACM0 -t 60 -f deCONZ_ConBeeII_0x26580700.bin.GCF

**ERROR:
781: Error: failed to open device, status: Permission error while locking the device (2)**

Information:
GCFFlasher_internal -l
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM0     | 0x1CF1 | 0x0030  | DE2191969  | ConBee II

s -l /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 mai   16 02:10 /dev/ttyACM0

Root is into the dialout group.

So everything seems to be setup correctly!

Another error that happened sometimes:
[image]

I didn't find how to solve that, so I hope someone will help me :)
Many thanks in advance for your help 👍 

Best regards,
Ageo

Smanar, 2020-05-16:
I don't have idea, bu have you check this issue https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1292 ?

Ageo, 2020-05-16:
Hello Smanar,

Thanks for your answer.

I read your post but i don't have the same error:
root@home:~# sudo GCFFlasher_internal -l
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM0     | 0x1CF1 | 0x0030  | DE2191969  | ConBee II

No error message serial is ok. I don't have any serial port and I don't have a RPI.

I follow this installation for ubuntu:
https://phoscon.de/en/conbee2/install#ubuntu

If i try to have logs:
deCONZ --dbg-info=1 --dbg-error=1 --dbg-aps=1 --dbg-zdp=1

Error:
qt.qpa.screen: QXcbConnection: Could not connect to display
Could not connect to any X display.

Nothing work... what a nightmare seriously! It's an alpha !?

Smanar, 2020-05-16:
Just to be sure you can try "ps ax | grep deCONZ" tu be sure theres isn't another deconz instance running.

But this message is generaly because something else are using the port.

Ageo, 2020-05-16:
ps ax | grep deCONZ
 1343 ?        S      0:00 sudo deCONZ --auto-connect=1 --dbg-error=1 --http-port=8484 --ws-port=20877 --upnp=0 -platform minimal --dbg-aps=2 --dbg-info=2 --dbg-zcl=2 --dbg-zdp=2 --dbg-http=2
 1345 ?        Sl     0:02 deCONZ --auto-connect=1 --dbg-error=1 --http-port=8484 --ws-port=20877 --upnp=0 -platform minimal --dbg-aps=2 --dbg-info=2 --dbg-zcl=2 --dbg-zdp=2 --dbg-http=2
 5552 pts/0    S+     0:00 grep deCONZ

I use Jeedom for  your information.

Smanar, 2020-05-16:
Haaa ok.
Jeedom use command line to run deconz, not service, so maybe you haven't correctly closed it.

Try to turn off the plugin, you need have nothing when using "ps ax | grep deCONZ".

Ok so you want to try the beta firmware, but all is working on your machine ? I understand better. On jeedom I know some users using a windows machine to make firmware update (even with command line)

Ageo, 2020-05-16:
Ok, i stop everything i have now: 
Jeedom plugin 
root@home:~# sudo systemctl stop deconz
root@home:~# sudo systemctl stop deconz-gui
ps ax | grep deCONZ
19597 pts/0    S+     0:00 grep deCONZ

Update try:
sudo GCFFlasher_internal -d /dev/ttyACM0 -t 60 -f deCONZ_ConBeeII_0x26580700.bin.GCF
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
retry, failed

I have update my firmware into windows at the begening but I really would like to check is that work on debian before put it into production. Just to be sure that everything work well.

Ageo, 2020-05-16:
Ok i reboot with the plugin into jeedom disable.

Result:
root@home:~# GCFFlasher_internal -l
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Path             | Vendor | Product | Serial     | Type
-----------------+--------+---------+------------+-------
/dev/ttyACM0     | 0x1CF1 | 0x0030  | DE2191969  | ConBee II
root@home:~# sudo GCFFlasher_internal -d /dev/ttyACM0 -t 60 -f deCONZ_ConBeeII_0x26580700.bin.GCF
GCFFlasher V3_13 (c) dresden elektronik ingenieurtechnik gmbh
Reboot device /dev/ttyACM0 (ConBee II)
deCONZ firmware version 26580700
failed reset -9223371981416
retry, failed

2139: Error: uart reset failed, check retry

Smanar, 2020-05-16:
I have already find this issue, generaly some of them have solved the problem using windows or with "luck" https://community.jeedom.com/t/update-key-combee-2-error/12719.

You can update the firmware on windows but use the key on debian.

Ageo, 2020-05-16:
But how we can talk to luck in this kind of thing :o 

I should be able to update the firmware with debian no ? I have tried to reboot several times, reset the GW  but same issues and same error.

Smanar, 2020-05-16:
Yeah I know, but I can't say why, but there is less issue with windows.

SwoopX, 2020-05-16:
Maybe that could help? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2493#issuecomment-591395981

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix