dresden-elektronik/deconz-rest-plugin #2782: Legrand - Cable Outlet integration

Frederic7391, 2020-05-17:
Hi,

I try to communicate with Legrand Cable Outlet, but it's not working totally.
When I plug the key on my laptop and use the software Deconz, it works perfectly, I can change mode, I turn ON, turn OFF.
But I don't see any "sensor" , "switch" or "light" in webapp, nothing appear.
How is it possible ?

Thanks, 

[image]

Smanar, 2020-05-17:
Yep, Phoscon is just an application that use the API.
It don't display all devices.
But if you use the on/off mode, you will have a visible device in light.
For the "fil pilote" mode you need to use another application or command line.

This device create 3 device
- 1 light that can can be used with on/off if enabled
- 1 sensor ZHAthermostat to set the "fil pilote" mode if enabled.
- 1 sensor ZHAconsumption for powe consumption.
Only the first one can be visible in Phoscon. And you need to update the device for the 2 sensor can be enabled.

Frederic7391, 2020-05-18:
Where I should see light? ZHAthermostat and ZHAConsumption?
Once i see it in deconz, if I do a command rest like "GET LIST" I should see my equipments? right?

Smanar, 2020-05-18:
You can see them only by the API.
Wich one application are you using (other than phoscon) ? Jeedom/HA/OpenHab/Domoticz/... ?

If you don't use other application I will give you a curl command to change the mode.

Frederic7391, 2020-05-18:
I use Jeedom.
But in jeedom I don't see any devices and I try to figure out why.
In deconz everything good and in jeedom nothing appear, no devices.

Smanar, 2020-05-18:
Ok So an easy way to check your sensor is using a web Browser with this url

`http://IP:PORTapi/API_KEY/sensors`

IP and PORT are the same you are using to go to Phoscon, and you can use the same api_key you are using in Jeedom

If you realy miss the sensors :
- set Phoscon in permit join (add new light)
- return in deconz select the node , select the cluster 0000 "Basic"
- on the cluster tab, clic on the "read" button (to read attribute)

Frederic7391, 2020-05-19:
Thanks, I followed your procedure and it works, I can operate with this device.

closed 2020-05-19 by Frederic7391