dresden-elektronik/deconz-rest-plugin #2783: Centralite 3323 -Geu Door-/Window Sensor and Thermometer

babajun12, 2020-05-17:
Request for support of following Sensor:
Centrallite 3323-Geu

[image]

[image]

Best Regards,
Gerhard

babajun12, 2020-05-17:
[image]

SwoopX, 2020-05-17:
Please follow https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support and provide all mentioned screenshots. Thanks!

babajun12, 2020-05-17:
With pleasure:
PowerConfiguration:
[image]
TempertureMeasurement:
[image]
IAS Zone:
[image]
Simple sensor clusters are empty:
[image]

Poll control:
[image]
Diagnostics:
[image]
OTAU:
[image]

Basic cluster and node iofo above.

Please reply if you need more information,
Thanks Gerhard

babajun12, 2020-05-21:
I added the IAS_CIE_Address

Mimiix, 2020-06-05:
Case closed!

closed 2020-06-05 by Mimiix