dresden-elektronik/deconz-rest-plugin #2784: Migrated from raspbee to conbee - firmware update fails

tosion, 2020-05-17:
I just migrated my old homeassistant system using official Deconz addon running Raspbee in RPI3B+ to Conbee in Intel NUC.  I used cloning to get new Conbee to run my old existing zigbee mesh with 67 nodes. It works like a charm and I am very pleased with the result.

However I have one open question. In Phoscon Gateway it says there is a new firmware update available. I run the update process and it look it updates. But the result is that it shows firmware did not change?

Could this be related to cloning or something else? Any help is appreciated.

[image]

[image]

[image]

SwoopX, 2020-05-17:
If memory serves, you must either restart deconz or just wait.

tosion, 2020-05-17:
Thanks for your response!

I updated first time yesterday - I have restarted the whole system couple times with totally power of.
I have done the update procedure couple times  ...

Now after the latest restart it says that version is up to date but firmware is 26320500 not 26330500.

BTW which one it should be?

tosion, 2020-05-18:
One more observation. In Deconz GUI behind VNC the node list still show the Coordinator as Raspbee with 26330500 firmware. In Phoscon it shows Conbee with 26320500 firmware.

Propable result of this cloning method - should I worry about this?

[image]

SwoopX, 2020-05-20:
Trust deconz GUI

tosion, 2020-05-21:
OK - thanks!

Probably there is no reason to worry about this?

closed 2020-05-21 by tosion