dresden-elektronik/deconz-rest-plugin #2785: Xiaomi plug ZNCZ02LM always report a 0 power value

ragefan, 2020-05-17:
Xiaomi plug ZNCZ02LM always report a 0 power value.
(On & Off is working with Deconz).

SwoopX, 2020-05-17:
Apparently, it is different than the ZNCZ04LM. Can you please provide the screenshots as described here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

Also, please run deconz with `--dbg-info=2` for at least 20 mins to chatch the Xiaomi special attribute reporting. It should look similar to this https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2572#issuecomment-599820045

ragefan, 2020-05-17:
How to get those screeshot ? i have a headless server.
is this the attributes you are looking for ?

{
    "1": {
        "etag": "f8423e30c00637b07a35e1fbd7c0cc0c",
        "hascolor": false,
        "manufacturername": "LUMI",
        "modelid": "lumi.plug",
        "name": "Smart plug 1",
        "state": {
            "alert": "none",
            "on": true,
            "reachable": true
        },
        "swversion": "09-04-2018",
        "type": "Smart plug",
        "uniqueid": "00:15:8d:00:03:9b:b0:b6-01"
    },
    "2": {
        "config": {
            "on": true,
            "reachable": true,
            "temperature": 4600
        },
        "ep": 2,
        "etag": "bfa6c367b9d229c40b3a5ff87214681d",
        "manufacturername": "LUMI",
        "modelid": "lumi.plug",
        "name": "Power 2",
        "state": {
            "lastupdated": "2020-05-17T21:53:11",
            "power": 2
        },
        "swversion": "09-04-2018",
        "type": "ZHAPower",
        "uniqueid": "00:15:8d:00:03:9b:b0:b6-02-000c"
    },
    "3": {
        "config": {
            "on": true,
            "reachable": true,
            "temperature": 4600
        },
        "ep": 3,
        "etag": "bfa6c367b9d229c40b3a5ff87214681d",
        "manufacturername": "LUMI",
        "modelid": "lumi.plug",
        "name": "Consumption 3",
        "state": {
            "consumption": null,
            "lastupdated": "none"
        },
        "swversion": "09-04-2018",
        "type": "ZHAConsumption",
        "uniqueid": "00:15:8d:00:03:9b:b0:b6-03-000c"
    }
}

ragefan, 2020-05-17:
[image]

[image]

Smanar, 2020-05-18:
Ha, you have 2 sensors
- ZHAConsumption with "null" as consumption, idk if it works on other device
- ZHAPower with 2 , so this sensor have a value.

The power seem working ?
The good unique ID is 00:15:8d:00:03:9b:b0:b6-02-000c

ragefan, 2020-05-18:
Thanks!  Sorry, I was speaking about consumption. Should it work?  I cannot get it trough the api and I actually don't understand you answer.

Smanar, 2020-05-18:
Ok, so for the other one I think @SwoopX need more information.
Can you restart deconz but with adding "--dbg-info=2" in the command line ? You will have some usefull debug line.

ragefan, 2020-05-18:
Thanks ! Here are the logs.
I can ses some dropped message and powre + consumption that are not understood (it seems)
[putty.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4645471/putty.log)

SwoopX, 2020-05-18:
Cool, that helps to double check if we got everything. Thanks!

Regardless, the upcomming release should have everything on board to have consumption finally working.

Smanar, 2020-05-18:
@ragefan and Jeedom is not an home automation application fo you ? ^^

SwoopX, 2020-05-18:
Double cheked the special attribute values and it looks all good, the code can handle all data. As indicated, consumption should be working with the next release.

Out of curiosity: the plug is new? It reports a consumption of 2W.

ragefan, 2020-05-18:
> 
> 
> @ragefan and Jeedom is not an home automation application fo you ? ^^

^^ It is just a bunch of php code i am fighting with !

ragefan, 2020-05-18:
> 
> 
> Double cheked the special attribute values and it looks all good, the code can handle all data. As indicated, consumption should be working with the next release.
> 
> Out of curiosity: the plug is new? It reports a consumption of 2W.

I received it recently from China but it does not seems to be a new model ?
I may try to compile from latest source code and report if it is working or not :)

SwoopX, 2020-05-18:
Meant new in that way as the consumption is that low.

You can always compile the plugin yourself. The instrunctions are to be found in the readme of the repo.

ragefan, 2020-05-18:
I did that : i compiled the plug-in and restarted deconz.
same result. I cannot read any consumption or power info.
Is upgrading the plugin enough ?

SwoopX, 2020-05-19:
Thanks for sharing your input. I guess we should start in clarifying if deconz announces that data first. From your log file, it looks quite ok.
```

00:45:04:572 ZCL attribute report 0x00158D00039BB0B6 for cluster: 0x0000, ep: 0x01, frame control: 0x1C, mfcode: 0x115F 
00:45:04:572 payload: 01ff422b64100103282b98398fc215409539cd84443b052113009a2000082122130727000000000000000009210001
00:45:04:573 0x00158D00039BB0B6 extract Xiaomi special attribute 0xFF01
00:45:04:573 64 on/off 1
00:45:04:573 03 temperature 43 °C
00:45:04:573 98 power 2,340000 (2)
00:45:04:573 95 consumption 0,002999 (3)
00:45:04:573 05 RSSI dB (?) 19 (0x0013)
00:45:04:573 9a unknown 0 (0x00)
00:45:04:573 08 unknown 4898 (0x1322)
00:45:04:573 07 unknown 0 (0x0000000000000000)
00:45:04:573 09 unknown 256 (0x0100)
```

Your data comes in alright...
`
00:45:09:184 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState, mode) VALUES ('3', 'Consumption 3', 'ZHAConsumption', 'lumi.plug', 'LUMI', '00:15:8d:00:03:9b:b0:b6-03-000c', '09-04-2018', '{"consumption":3,"lastupdated":"2020-05-18T22:45:04.572"}', '{"on":true,"reachable":true,"temperature":4300}', '{"d":83,"ep":3,"in":[12],"p":260}', 'normal', '1')`

... and apparently, that data also lands in deconz' database. Now, I'd recommend to check the REST API first, as you did here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2785#issuecomment-629872063 . If data for consumption shows up, it feels like deconz is doing just fine. To gain more certainty, pipe the debug output to a file and let it run for a hour or so and make sure, you apply a considerable load on the plug (100W or more). A higher load is preferred since we also had people wondering consumption is not raising with a 3W LED light (and the consumption reporting interval being in 10W steps), although it took just a reasonable amount of time. By this, there's a higher chance that the values are updated more quickly. Meanwhile, keep an eye on the consumption values. If they change, then the issue is not on deconz side and you may stop the test.

We currently do not know on what threshold the plug sends any consumption related data

ragefan, 2020-05-19:
Hi !
Here the new attributes from the API :
{
    "1": {
        "etag": "9216d980f6ba8979220e5a12c31b3958",
        "hascolor": false,
        "lastseen": "2020-05-19T21:32:20.839",
        "manufacturername": "LUMI",
        "modelid": "lumi.plug",
        "name": "Smart plug 1",
        "state": {
            "alert": "none",
            "on": true,
            "reachable": true
        },
        "swversion": "09-04-2018",
        "type": "Smart plug",
        "uniqueid": "00:15:8d:00:03:9b:b0:b6-01"
    },
    "2": {
        "config": {
            "on": true,
            "reachable": true,
            "temperature": 2500
        },
        "ep": 2,
        "etag": "abf5f6733e8f0673dfff0048f0efadc2",
        "lastseen": "2020-05-19T21:32:20.839",
        "manufacturername": "LUMI",
        "modelid": "lumi.plug",
        "name": "Power 2",
        "state": {
            "lastupdated": "2020-05-19T21:29:56.774",
            "power": 0
        },
        "swversion": "09-04-2018",
        "type": "ZHAPower",
        "uniqueid": "00:15:8d:00:03:9b:b0:b6-02-000c"
    },
    "3": {
        "config": {
            "on": true,
            "reachable": true,
            "temperature": 2500
        },
        "ep": 3,
        "etag": "abf5f6733e8f0673dfff0048f0efadc2",
        "lastseen": "2020-05-19T21:32:20.838",
        "manufacturername": "LUMI",
        "modelid": "lumi.plug",
        "name": "Consumption 3",
        "state": {
            "consumption": 3,
            "lastupdated": "2020-05-19T21:29:56.773"
        },
        "swversion": "09-04-2018",
        "type": "ZHAConsumption",
        "uniqueid": "00:15:8d:00:03:9b:b0:b6-03-000c"
    }
}

SwoopX, 2020-05-19:
Thanks. So the log shows multiple reporting updates for the power value but in that short period of time, none for the consumption. This is not necessarily unusual. As described above, a high load combined with a larger timespan should also give some updates there.

One thing to note: the plug doesn't report any current or voltage, so not having any values in jeedom is reasonable. Also, the REST API output shows the values, so I conclude deconz is working as it should.

I don't know jeedom, but maybe the logical ID is not set precise enough? I'd deduce from the screenshot, that it want to work on the right sensor (02-000c), but the state has multiple attributes in it. Also take note that the consumption value will come from 03-000c.

ragefan, 2020-05-20:
Hi,
Thanks for your support. I think i got more lucky with this log for the consumption ?
[putty.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4658324/putty.log)

Can you tell why the REST Api does not report the current or voltage ? is it something to develop or a restriction ?

I start to by upset with jeedom. I got weird behaviour.  The plug is detected but even if i change the right Logical ID nothing happens :(

Can you tell me what ZHAxxxx are ?
i see ZHAPower, ZHAConsumption ... created by Deconz (?) without knowing what it is ?

By the way, i added a light sensor and i think no value is reported either.

Smanar, 2020-05-20:
I think it's just the device that don't support voltage and current.

ZHA device are just device created by deconz they depend cluster used by device, and API code. Deconz can select some of them according to cluster device, and with the API code we can add/remove them.

Unique ID are composed by
(Mac_adress)-(endpoint)-(cluster)
And Name are according to the type of valuemesured

What is your light sensor ?

ragefan, 2020-05-20:
Ok ! Thanks a lot !
Did you have chance to look at the logs for the consumption ?

For the light sensor, i posted here :) 
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2300#issue-547457839

SwoopX, 2020-05-20:
> I think it's just the device that don't support voltage and current.

Indeed. For the rest, nothing to add ;)

My production environment was bricked for the whole day, but I'm checking your stuff now.

SwoopX, 2020-05-20:
Ok, just checked the log. I still see reports for power, but not for consumption. Kinda strange...

I've also seen that overall consumption has increased however. Maybe not high enough to issue a consumption report (e.g. if the threshold is: send a report upon a 10W increase). What does the REST API give as consumption value, still 3?

ragefan, 2020-05-20:
I got 0.017 
[image]

I move to HA (pointing to the same Deconz as before)

SwoopX, 2020-05-20:
😄 So I guess we're done here?

ragefan, 2020-05-20:
Actually .... yes ! :) Just wanted you to confirm ^^ but Ok for me. Thanks for your support :)

closed 2020-05-20 by ragefan

SwoopX, 2020-05-20:
Well, I see something reasonable ;) Keep an eye on the value changes.