dresden-elektronik/deconz-rest-plugin #2786: INNR SP120 detected as light instead of switch

Webunity, 2020-05-19:
New Innr SP120 added but adding it as a switch (by searching) led to a red LED on the innr.
Then tried adding it as a sensor and it went OK, but appeared at the lights section.

[image]

I am running 2.05.75

Mimiix, 2020-05-19:
Hi!

I don't think it is a switch, as it isn't usable as something that switches something else on or off(within Phoscon).

Webunity, 2020-05-19:
Good point; but then it should appear as sensor at least.

As a side question;
> can you also configure this plug to have a default 'on' state? i want it to measure the power from my washing machine but power cuts should not put the plug into an 'off' state.

Webunity, 2020-05-19:
Related: I think mine is a duplicate then; https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2578

closed 2020-05-19 by Webunity