dresden-elektronik/deconz-rest-plugin #2788: ConBee II as router for Hue

fubar-coder, 2020-05-19:
I followed the instructions of #21, but I encountered the problem that setting the TC address to 0xffffffffffffffff results in the TC link key being reset to 0x5a6967426565416c6c69616e63653039.

ebaauw, 2020-05-19:
See the many other issues that link to 21.  It has been broken for a long time.  You might try a firmware version that was current early 2017 from http://deconz.dresden-elektronik.de/deconz-firmware/?C=M;O=A.

fubar-coder, 2020-05-20:
That's the problem: There is no firmware that old for the ConBee II.

ebaauw, 2020-05-21:
The ConBee II wasn’t yet introduced back then.  Tbh I’m not sure if the issue is with the firmware or with deCONZ, but I doubt 2.04.35 will want to work with the ConBee II.

Never tried this myself, but you might try and ignore the TC Link key altogether, and configure the network key of the Hue bridge directly.  Of course, you’d need to recover the network key By sniffing the Hue bridge while pairing a device.  Afaik, there’s still no ZShark firmware for the ConBee II, so you’d need a different device for sniffing.

rkjnsn, 2020-06-28:
I believe this is still an issue.

fubar-coder, 2020-07-19:
Are there any news about this issue?

Mimiix, 2020-08-10:
@fubar-coder I think there is a tab in the deconz-gui settings where you can select the mode.

Then you should be able to add it to hue, but i don't know if that would work. Other than that, i will ask @manup to reply on this.

I doubt if it gets any priority, as this is a niche.

fubar-coder, 2020-09-03:
@Mimiix  I found the settings, but the problem is that the key gets reset when I try to set the address.

Mimiix, 2020-09-03:
I really have no clue on this one.

I'll ask Manup

SwoopX, 2020-09-03:
Iirc, the magic only happens when you leave the network for configuration changes. So try to press `leave` and then try again.

tkl-git, 2020-09-17:
> I really have no clue on this one.
> 
> I'll ask Manup

Hi,

any news about this subject? I would really appreciate if the developers could finally solve this issue which has been annoying people about 2 years already. It really can't be that hard to fix if you are just able to allocate some time to the task?

I bought a ConBee 2 mainly because I wanted to debug my current Zigbee network(s) (and maybe later integrate it to my current system). I have a quite large installation of Zigbee devices including 2 Hue bridges and a Cozify hub. The Cozify home automation hub has it's own Zigbee network and it controls Hue devices through those 2 Hue bridges, not directly.

My system includes over 100 Zigbee devices and the problem is that sometimes some of the Hue devices go offline without any obvious reason. I would need to get the routing tables from the devices and see a network map. That way I could maybe identify and solve the problem (adding more repeating Zigbee devices randomly is not the solution I need).

Is it currently possible to join the network of the Hue bridge as an end device instead of a router?

ConBee 2 is still missing the Zshark support too which also came as a surprise to me.

Unfortunately it is not an option for me to use ConBee as a Zigbee cordinator. I really like the Conbee 2 hardware so I don't want to get yet another Zigbee USB gateway. Hopefully the developers could fix these ConBee 2 problems soon. Thank you.

carbonelle, 2021-08-11:
hello I succeeed in joining a hue network with the conbee II  by followed the instructions of #21, but I defined the TC address with the MAC address  of the Hue bridge

carbonelle, 2021-08-11:
Defining the TC address with the mac address of the hue bridge avoid the reseting of the TC link key with  0x5a6967426565416c6c69616e63653039.  However if you remove the conbee II from the power (USB port) you lost the connection and you have to redo the configuration.

Mimiix, 2021-08-11:
@tkl-git For some reason i missed this one. 

I've asked him, but no avail. yet.
I'll ask again :)

LineF, 2021-12-11:
Same for me - it would be nice if deCONZ / Conbee-II could be used as a router in an existing Hue network.
Any news about that?

Martin

paetsky, 2022-11-10:
I'd also appreciate updates on this bug topic! Conbee II would be a good tool to force some device mappings on Hue network with non-Philips (ie. Busch-Jaeger 6710U, 6711U and 6715U) devices on it.

kavejo, 2023-04-09:
Has anybody managed to run Conbee II as a Zigbee Router (plugged into a power socket)?
I have done that with a Sonoff ZBdongle-E, but can't find a way to do that with a fully up-to-date Conbee II.

Mimiix, 2023-04-09:
@manup

BOFHELL, 2024-07-07:
Push

ebaauw, 2024-10-04:
I just joined my ConBee III to the Zigbee network by my Hue bridge.  I messed up, and the ConBee is still acting as coordinator, but that doesn't seem to cause any harm.  Note that the Hue bridge isn't a Zigbee coordinator, but just another router (with NWK address 0x0001) that doubles as gateway.

Pairing the RaspBee or ConBee with the Hue bridge doesn't work - somehow writing the ZLL _TC Link Key_ doesn't stick.  You need to configure the network parameters manually.  Typically, you need to _Leave_ the network, change the parameters, _Write_ them, _Read_ them back for good measure, and _Join_.  I like to do another _Read_ after joining, to make sure the parameters stuck.

You need to set at least the _PAN ID_, _Network Key_ and _Channel Mask_.  I also set the _TC Link Key_, but that might not be needed, and the _Ext PAN ID_ (to the Zigbee MAC address of the Hue bridge).  deCONZ bitches about that in the log, but it doesn't seem to do any harm.
```
Oct 04 11:15:12 pi15 deCONZ[2955400]: 11:15:05:127 apsUseExtPanid is 0x00178801050A55DA but should be 0, start reconfiguration
Oct 04 11:15:12 pi15 deCONZ[2955400]: 11:15:05:127 Skip automatic channel change, TODO warn user
```

Note that in general it is not recommended to use more than one gateway on your Zigbee network.  Both gateways will try and configure devices, which might cause conflicts, particularly for sensor bindings.  I currently only have a couple of Hue lights connected to this test network, to have the Hue bridge upgrade these to their latest firmware (for which the files haven't been located yet), and use deCONZ to see what has changed on the device.  As the Hue bridge doesn't setup attribute reporting for Hue lights, I don't expect much conflict in this case.  For good measure, I haven't enabled the OTAU server on deCONZ.  Also, I paired the lights to the Hue bridge, before adding the ConBee III.  I'm not sure what would happen when pairing additional devices when both gateways are active.

ebaauw, 2024-10-04:
> Has anybody managed to run Conbee II as a Zigbee Router (plugged into a power socket)?
I have done that with a Sonoff ZBdongle-E, but can't find a way to do that with a fully up-to-date Conbee II.

Never tried that myself.  You would need to connect the ConBee II to deCONZ to configure it as router and join the network where you want it to be a router in.  If that's a "normal" network, using the ZHA link key, there should be no issue pairing the ConBee to that network.  If it's a ZLL network, you'll need to sniff the network parameter and enter these manually, see my remarks above.  As far as I know, basic router functionality is handled completely by the firmware, and should continue to function, when deCONZ is down.  Of course you would need to check that on another gateway.  Then move the ConBee to a USB power adapter, and double-check it's still acting as router.