dresden-elektronik/deconz-rest-plugin #2790: Rebooting the Pi disables WLAN connection to router and activates WLAN access point

luddibro, 2020-05-20:
Rebooting the Pi disables WLAN connection to router and activates WLAN access point. After that WLAN password is also gone.

Model: Rasperry Pi B, RaspBee II
Phoscon App:
Version: 2.05.72 / 12.12.2019
Firmware: 26520700

f403, 2020-05-24:
If you do not need deCONZ to manage your wi-fi settings you can try disabling `deconz-wifi.service`.

To do so open `/usr/bin/deCONZ-init.sh`, find
```
## check wifi service ##
# enable WiFi service just for Phoscon gateways which
# have a printet label with WiFi password
```

and add following lines before or after mentioned comments:
```
## PATCH: Disable wifi management
ENABLE_SERVICE=0
```
Save your changes and do:
```
sudo systemctl daemon-reload
sudo systemctl restart deconz.service
```

Mimiix, 2020-06-06:
AS this has been answered, i close the issue.

closed 2020-06-06 by Mimiix