dresden-elektronik/deconz-rest-plugin #2792: apt doesn't update like I expect it to and phoscon app doesn't show new firmware

ABeumer525, 2020-05-23:
I'm running in to an issue. Probably I'm doing something wrong here.

Setup:
Raspberry 3b+
Raspbian (Hassbian)
Conbee II
Deconz 2.05.72
Firmware 264A0700

Issue:
Running apt-get update and apt-get upgrade only upgrades to x.74 whilest I see that 2.05.76 is tagged as stable.
My deconz.list (/etc/sources.list.d/deconz.list) has been set to main.
When I set this to beta. It does install the "correct" version.

And  the firmware available to download (https://deconz.dresden-elektronik.de/deconz-firmware/?C=M;O=D) is 26580700, this does not show up in the phoscon webapp.

What is wrong with my setup or otherwise wrong with the software.

Help much appreciated and if I can clarify I will.

SwoopX, 2020-05-23:
I'm afraid you misinterpret. .76 is beta and not yet stable, so you only get it via apt if you use beta channel. Also, latest firmware for Conbee II must be installed manually. That's also what the release notes say.

ABeumer525, 2020-05-23:
@SwoopX Thanks, I was afraid I didn’t get the memo as intended.
So stable is something else then main release I reckon.

closed 2020-05-23 by ABeumer525