dresden-elektronik/deconz-rest-plugin #2794: Unable to pair switches

boombuler, 2020-05-23:
I recieved my raspbee II yesterday to replace my tradfri bridge.

I'm using the phoscon pwa.
First I tried to pair some lights, which worked fine. After that I've tried to pair some Tradfri remotes which did not work at all. I've reset the whole thing and started over, this time I've first paired one of those tradfri remotes first, which worked, but the second remote failed to pair. Though lights were pairable. 
I also tried some other switches: Other Tradfri Remotes, a Tradfri dimmer and also a "ledvance switch mini"

I've also had a look to the deCONZ GUI app and found those incomplete nodes which seem to be the switches.
[image]

Does anyone have any hint on how to get those working?

Smanar, 2020-05-23:
The "ledvance switch mini" is from Osram ? If yes, I m on it ATM and need some help ^^

Wich one pairing procedure have you used for the ikea remote ? You have followed the procedure displayed in Phoscon ?

boombuler, 2020-05-23:
yes that should be the same as the osram but with different branding feel free to contact me if I could help you with it ;)

I've done it the same way as the one working remote. Phoscon -> Switches -> Add new switch -> Ikea -> Remote Control -> Press button 4 times -> wait -> fail...

btw. I've also reset twice and used a different tradfri remote as first pairing device, which also worked. but i can't get more then one switch paired...

Smanar, 2020-05-23:
Your ledvance device have as model id "lightify switch mini" ?

For the ikea remote, after have pressed the button 4 time, you have the led blinking ?
Make a try with pressing randomly buttons after this step, every seconds.

boombuler, 2020-05-23:
I can't find any model id on the packaging or the device and it also doesn't pair. But I'm very sure that it is the same device since ledvance is part of osram see [wikipedia](https://en.wikipedia.org/wiki/LEDVANCE)

The ikea remote has the led blinking first, after that it is slowly fading in and out and thats it. Random buttons don't seem to help.

Smanar, 2020-05-23:
Same blinking than https://www.youtube.com/watch?v=aMzKf3VNuPQ ?

boombuler, 2020-05-23:
Hi,

I've taken 2 videos while pairing a tradfri remote and a tradfri dimmer...
You can view the progress [here](https://photos.app.goo.gl/kMqE4s73QEdxbuWv9)

Smanar, 2020-05-23:
Nothing special for me ...
Perhaps battery, the led is on when you press a button ?

boombuler, 2020-05-23:
The battery is fine. No red light when I press any button. I've also tested multiple remotes and as I mentioned before: Once I reset the router, I can pick any one of my ikea remotes, pair it, and it will work. but only exactly one. Is there anything i could check? any logs or something else that will help debug the situation?

Btw: I've also tried to remove the powersupply for all routers. But that didn't work either.

Smanar, 2020-05-23:
You router don't have a maximum connexion ? On the capture you realy have few device on it, but I realy don't have idea.

Have you tried to make inclusion next another router ?

boombuler, 2020-05-23:
Since I just got the Raspbee I have not yet included all devices, since its not very usable if I can't use switches -.- 

Yes I've tried different locations, all with the same result :(

boombuler, 2020-05-23:
Hmmm I kind of paired the "ledvance" switch. It is not visible in phoscon but i can control the brightness of the living room with it.

Smanar, 2020-05-23:
Yep it s normal, this device can use broadcast command.

Can you try to run deconz with logging enabled ? Perhaps we can see something usefull ?

boombuler, 2020-05-23:
Sure, should i take a log while pairing oder just general usage?

boombuler, 2020-05-24:
Just captured a log with `--dbg-info=1 --dbg-zcl=1 --dbg-zdb=1` [deconz.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4673119/deconz.log)

boombuler, 2020-05-24:
Another Update: I was able to assign the group of one ikea remote via deCONZ UI, now i can at least control the lights of that group. non the less i am not able to see or configure it by using phoscon. Any hint on how to "fake" the normal setup?

Smanar, 2020-05-24:
Ok so deconz see the device announce, but nothing append during the "information collect".

BTW on your screenshoot you have 2 remotes, one of them have the name grayed, but the other not.
On the still connected one, can you try :
- Set phoscon in permit join
- In deconz ,select the first circle, and recheck square to resend command
- Wake the device during procedure.

boombuler, 2020-05-24:
Thanks this kind of does the trick:

1. Let the device join the network (visible in deconz)
2. Set Phoscon to Pairing mode
3. Select the 2nd Circle from deCONZ and select "Basic"
4. While Phoscon is still waiting for the device, read the attributes off the device in deCONZ.
5. deCONZ finds the switch.

Smanar, 2020-05-24:
Ok so you was in better situation than your previous capture, on it you had only 1 circle.

boombuler, 2020-05-24:
Indeed :) Btw the "Ledvance" switch identifies itself as "OSRAM" "Lightify Switch Mini". 
But I cant see it in phoscon. If I could help you with anything about that switch please let me know.

Smanar, 2020-05-24:
The code make the device usable by the API in third application but I m almost sure it will not appear in Phosocn.

For the Osram mini switch you have the issue and the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2763#issuecomment-628192423

I have a working code for the 4 button, so I m sure I can make the same for the 3 button, but I need a correspondance beetwen pressed button and values to use.

bphermansson, 2020-05-24:
I'm not sure I'm following. When the device has a green wire to the controller in Deconz it is connected? 
[image]

closed 2020-05-24 by boombuler

Smanar, 2020-05-24:
Yep, and name not grayed, because you can have node connected but without the green line.

boombuler, 2020-05-24:
@Smanar I would have loved to test your code but the raspberry is taking forever to compile it... I might have a look at it tomorrow. Thanks again for your help!