dresden-elektronik/deconz-rest-plugin #2796: Delay in reporting state

bastshoes, 2020-05-23:
After upgrading to 2.05.76 my to hue bulbs have a huge delay in reporting their state change. It take up to 1 minute to report state change.

I'm running fw 26580700.

ebaauw, 2020-05-23:
How do you change the state?

Note that Hue bulbs don't support attribute reporting.  deCONZ needs to poll the bulb for it's state.  The larger your network, the longer the time before a bulb is polled next.

bastshoes, 2020-05-23:
I'm changing state via REST_API (home-assistant). My network hasn't changed. I only upgrade deconz and fw. After upgrade this delay become visible.

ebaauw, 2020-05-23:
That's odd, the REST API should update it's cache when changing a light state.  Can you check what commands HA sends?

bastshoes, 2020-05-23:
It looks like there is some issue on HASS side. deConz sends response immediately.
```
today at 9:16 PM 2020-05-23 21:16:05 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': True}" to "172.18.0.13 /lights/9/state"
today at 9:16 PM 2020-05-23 21:16:05 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/9/state/on': True}}]
today at 9:16 PM 2020-05-23 21:16:17 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': False}" to "172.18.0.13 /lights/9/state"
today at 9:16 PM 2020-05-23 21:16:17 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/9/state/on': False}}]
today at 9:18 PM 2020-05-23 21:18:16 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': True}" to "172.18.0.13 /lights/9/state"
today at 9:18 PM 2020-05-23 21:18:16 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/9/state/on': True}}]
today at 9:18 PM 2020-05-23 21:18:38 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': False}" to "172.18.0.13 /lights/9/state"
today at 9:18 PM 2020-05-23 21:18:38 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/9/state/on': False}}]
```

Kane610, 2020-05-23:
the Hass integration only update state based on web socket events

bastshoes, 2020-05-23:
There is delay between http response and websocket update. Sometimes it's 2 sec, some times it's bigger
```
today at 10:08 PM 2020-05-23 22:08:38 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': False}" to "172.18.0.13 /lights/9/state"
today at 10:08 PM 2020-05-23 22:08:38 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/9/state/on': False}}]
today at 10:08 PM 2020-05-23 22:08:46 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"9","r":"lights","state":{"alert":null,"bri":254,"on":false,"reachable":true},"t":"event","uniqueid":"00:17:88:01:02:45:09:5b-0b"}
today at 10:10 PM 2020-05-23 22:10:00 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': True}" to "172.18.0.13 /lights/9/state"
today at 10:10 PM 2020-05-23 22:10:00 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/9/state/on': True}}]
today at 10:10 PM 2020-05-23 22:10:00 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"9","r":"lights","state":{"alert":null,"bri":254,"on":true,"reachable":true},"t":"event","uniqueid":"00:17:88:01:02:45:09:5b-0b"}
today at 10:10 PM 2020-05-23 22:10:13 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': False}" to "172.18.0.13 /lights/9/state"
today at 10:10 PM 2020-05-23 22:10:13 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/9/state/on': False}}]
today at 10:10 PM 2020-05-23 22:10:15 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"9","r":"lights","state":{"alert":null,"bri":254,"on":false,"reachable":true},"t":"event","uniqueid":"00:17:88:01:02:45:09:5b-0b"}
today at 10:11 PM 2020-05-23 22:11:39 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': True}" to "172.18.0.13 /lights/9/state"
today at 10:11 PM 2020-05-23 22:11:39 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/9/state/on': True}}]
today at 10:11 PM 2020-05-23 22:11:39 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"9","r":"lights","state":{"alert":null,"bri":254,"on":true,"reachable":true},"t":"event","uniqueid":"00:17:88:01:02:45:09:5b-0b"}
today at 10:11 PM 2020-05-23 22:11:47 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': False}" to "172.18.0.13 /lights/9/state"
today at 10:11 PM 2020-05-23 22:11:47 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/9/state/on': False}}]
today at 10:11 PM 2020-05-23 22:11:58 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"9","r":"lights","state":{"alert":null,"bri":254,"on":false,"reachable":true},"t":"event","uniqueid":"00:17:88:01:02:45:09:5b-0b"}
```

Kane610, 2020-05-23:
Sounds like a bug in deCONZ then, websocket is usually snappy

ebaauw, 2020-05-23:
Hm, looks like the API cache isn't updated when setting `on` to false.  Setting `on` to true and changing `bri` seem to be reported instantaneously.  Setting `bri` to 0 no longer turns the light off, but the light is recorded as off in the cache.

ebaauw, 2020-05-23:
It's even more fun: the cached `on` value is set to true on _Off with Effect_.  Try setting the Hue light off while it's off.

bastshoes, 2020-05-25:
Just checked 2.05.77. Everythnig works as expected. Thank you Erik!

closed 2020-05-25 by bastshoes