dresden-elektronik/deconz-rest-plugin #2797: Another Firmware not connected in Phoscon

ruabo, 2020-05-23:
I'm on hassio 0.109.6 with deconz addon 5.3.3 and raspee running on firmware 26330500 (2.05.76). Phoscon tells me a new firmware is available update to 26350500. I engaged the firmware upgrade process, I got "updated finished" , click on "done" but "firmware not connected" is displayed. 
A fews seconds later again the message a new update is available is displayed.
From the log:
18:49:14:008 Wait reconnect after firmware update
18:49:15:009 Wait reconnect after firmware update
18:49:16:009 Wait reconnect after firmware update
18:49:17:008 Wait reconnect after firmware update
18:49:18:009 Wait reconnect after firmware update
1172.30.32.2 - - [23/May/2020:18:49:24 +0200] "GET /api/2B55023BAF/config?_=1590252357262 HTTP/1.1" 200 4290 "http://hassio.local:8123/api/hassio_ingress/WwJMW9W8V3Eu9Ro0njpNym4MNaDbAfUyhlqSizE4llo/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,6,18&gwid=00212EFFFF02466D" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
172.30.32.2 - - [23/May/2020:18:49:28 +0200] "GET /api/2B55023BAF/config?_=1590252357263 HTTP/1.1" 200 4290 "http://hassio.local:8123/api/hassio_ingress/WwJMW9W8V3Eu9Ro0njpNym4MNaDbAfUyhlqSizE4llo/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,6,18&gwid=00212EFFFF02466D" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
172.30.32.2 - - [23/May/2020:18:49:35 +0200] "GET /api/2B55023BAF/config?_=1590252357264 HTTP/1.1" 200 4290 "http://hassio.local:8123/api/hassio_ingress/WwJMW9W8V3Eu9Ro0njpNym4MNaDbAfUyhlqSizE4llo/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,6,18&gwid=00212EFFFF02466D" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
172.30.32.2 - - [23/May/2020:18:49:41 +0200] "GET /api/2B55023BAF/config?_=1590252357265 HTTP/1.1" 200 4290 "http://hassio.local:8123/api/hassio_ingress/WwJMW9W8V3Eu9Ro0njpNym4MNaDbAfUyhlqSizE4llo/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,6,18&gwid=00212EFFFF02466D" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
172.30.32.2 - - [23/May/2020:18:49:45 +0200] "GET /api/2B55023BAF/config?_=1590252357266 HTTP/1.1" 200 4290 "http://hassio.local:8123/api/hassio_ingress/WwJMW9W8V3Eu9Ro0njpNym4MNaDbAfUyhlqSizE4llo/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,6,18&gwid=00212EFFFF02466D" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
172.30.32.2 - - [23/May/2020:18:49:55 +0200] "GET /api/2B55023BAF/config?_=1590252357268 HTTP/1.1" 200 4290 "http://hassio.local:8123/api/hassio_ingress/WwJMW9W8V3Eu9Ro0njpNym4MNaDbAfUyhlqSizE4llo/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,6,18&gwid=00212EFFFF02466D" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
172.30.32.2 - - [23/May/2020:18:50:00 +0200] "GET /api/2B55023BAF/config?_=1590252357269 HTTP/1.1" 200 4289 "http://hassio.local:8123/api/hassio_ingress/WwJMW9W8V3Eu9Ro0njpNym4MNaDbAfUyhlqSizE4llo/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,6,18&gwid=00212EFFFF02466D" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
172.30.32.2 - - [23/May/2020:18:50:05 +0200] "GET /api/2B55023BAF/config?_=1590252357270 HTTP/1.1" 200 4289 "http://hassio.local:8123/api/hassio_ingress/WwJMW9W8V3Eu9Ro0njpNym4MNaDbAfUyhlqSizE4llo/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,6,18&gwid=00212EFFFF02466D" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
172.30.32.2 - - [23/May/2020:18:50:09 +0200] "GET /api/2B55023BAF/config?_=1590252357271 HTTP/1.1" 200 4289 "http://hassio.local:8123/api/hassio_ingress/WwJMW9W8V3Eu9Ro0njpNym4MNaDbAfUyhlqSizE4llo/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,6,18&gwid=00212EFFFF02466D" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
172.30.32.2 - - [23/May/2020:18:50:14 +0200] "GET /api/2B55023BAF/config?_=1590252357272 HTTP/1.1" 200 4289 "http://hassio.local:8123/api/hassio_ingress/WwJMW9W8V3Eu9Ro0njpNym4MNaDbAfUyhlqSizE4llo/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,6,18&gwid=00212EFFFF02466D" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
172.30.32.2 - - [23/May/2020:18:50:21 +0200] "GET /api/2B55023BAF/config?_=1590252357273 HTTP/1.1" 200 4289 "http://hassio.local:8123/api/hassio_ingress/WwJMW9W8V3Eu9Ro0njpNym4MNaDbAfUyhlqSizE4llo/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,6,18&gwid=00212EFFFF02466D" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
172.30.32.2 - - [23/May/2020:18:50:25 +0200] "GET /api/2B55023BAF/config?_=1590252357274 HTTP/1.1" 200 4289 "http://hassio.local:8123/api/hassio_ingress/WwJMW9W8V3Eu9Ro0njpNym4MNaDbAfUyhlqSizE4llo/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,6,18&gwid=00212EFFFF02466D" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
172.30.32.2 - - [23/May/2020:18:50:30 +0200] "GET /api/2B55023BAF/config?_=1590252357275 HTTP/1.1" 200 4289 "http://hassio.local:8123/api/hassio_ingress/WwJMW9W8V3Eu9Ro0njpNym4MNaDbAfUyhlqSizE4llo/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,6,18&gwid=00212EFFFF02466D" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
172.30.32.2 - - [23/May/2020:18:50:35 +0200] "GET /api/2B55023BAF/config?_=1590252357276 HTTP/1.1" 200 4289 "http://hassio.local:8123/api/hassio_ingress/WwJMW9W8V3Eu9Ro0njpNym4MNaDbAfUyhlqSizE4llo/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,6,18&gwid=00212EFFFF02466D" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
172.30.32.2 - - [23/May/2020:18:50:41 +0200] "GET /api/2B55023BAF/config?_=1590252357277 HTTP/1.1" 200 4289 "http://hassio.local:8123/api/hassio_ingress/WwJMW9W8V3Eu9Ro0njpNym4MNaDbAfUyhlqSizE4llo/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,6,18&gwid=00212EFFFF02466D" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
172.30.32.2 - - [23/May/2020:18:50:45 +0200] "GET /api/2B55023BAF/config?_=1590252357278 HTTP/1.1" 200 4289 "http://hassio.local:8123/api/hassio_ingress/WwJMW9W8V3Eu9Ro0njpNym4MNaDbAfUyhlqSizE4llo/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,6,18&gwid=00212EFFFF02466D" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
172.30.32.2 - - [23/May/2020:18:50:51 +0200] "GET /api/2B55023BAF/config?_=1590252357279 HTTP/1.1" 200 4289 "http://hassio.local:8123/api/hassio_ingress/WwJMW9W8V3Eu9Ro0njpNym4MNaDbAfUyhlqSizE4llo/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,6,18&gwid=00212EFFFF02466D" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
172.30.32.2 - - [23/May/2020:18:50:54 +0200] "GET /api/2B55023BAF/config?_=1590252357280 HTTP/1.1" 200 4289 "http://hassio.local:8123/api/hassio_ingress/WwJMW9W8V3Eu9Ro0njpNym4MNaDbAfUyhlqSizE4llo/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,6,18&gwid=00212EFFFF02466D" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
172.30.32.2 - - [23/May/2020:18:50:55 +0200] "GET /websocket HTTP/1.1" 101 1089 "-" "Mozilla/5.0 (Macintosh; Intel Mac OS X 10.14; rv:76.0) Gecko/20100101 Firefox/76.0"
8:49:19:011 Wait reconnect after firmware update
18:49:20:007 Wait reconnect after firmware update
18:49:21:009 Wait reconnect after firmware update
18:49:22:010 Wait reconnect after firmware update
18:49:23:010 Wait reconnect after firmware update
18:49:24:011 Wait reconnect after firmware update
18:49:25:011 Wait reconnect after firmware update
18:49:26:009 Wait reconnect after firmware update
18:49:27:010 Wait reconnect after firmware update
18:49:28:009 Wait reconnect after firmware update
18:49:29:008 Wait reconnect after firmware update
18:49:30:008 Wait reconnect after firmware update
18:49:31:012 Wait reconnect after firmware update
18:49:32:009 Wait reconnect after firmware update
18:49:33:013 Wait reconnect after firmware update
18:49:34:010 Wait reconnect after firmware update
[... cut ...]
18:51:24:833 	 local time seems to be ok
18:51:52:881 verified group capacity: 15 and group count: 1 of LightNode 0x00158d0001fdeae4
18:51:52:881 0x00158d0001fdeae4 found group 0xFFF0
18:51:56:630 verified group capacity: 255 and group count: 2 of LightNode 0x000d6ffffee3e6ed
18:51:56:630 0x000d6ffffee3e6ed found group 0xFFF0
18:51:56:630 0x000d6ffffee3e6ed found group 0x0007
18:52:00:497 binding for cluster 0x0000 of 0x000D6FFFFEE7562D exists (verified by reporting

What's the problem ?
Thanks

Nebuchanezzer, 2020-05-24:
This seems to have happened to me as well.

Nebuchanezzer, 2020-05-24:
An update.

Something from the log to follow up with:
15:44:06:643 SensorNode 34 set node 0x00158d000253c6ec
15:44:06:643 SensorNode 35 set node 0x00158d000253c6ec
15:44:06:673 SensorNode 36 set node 0xec1bbdfffea307a8
15:44:06:691 don't close database yet, keep open for 900 seconds
15:44:06:692 LightNode 15: Klädkammare - Glob added
15:44:06:719 SensorNode 37 set node 0x680ae2fffe3e006f
15:44:06:740 SensorNode 38 set node 0x00158d00024e35bc
15:44:06:757 SensorNode 39 set node 0x00158d000283bc2b
15:44:06:844 Current channel 11
15:44:06:860 CTRL ANT_CTRL 0x03
15:44:06:884 Device protocol version: 0x010B
15:44:06:932 Current channel 11
15:44:06:948 CTRL ANT_CTRL 0x03
15:44:17:033 don't close database yet, keep open for 900 seconds
15:44:17:033 LightNode 6: Configuration tool 6 added
15:44:20:482 ZCL attribute report 0xEC1BBDFFFE1C9EED for cluster: 0x0006, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
15:44:29:202 binding for cluster 0x0006 of 0xEC1BBDFFFE1C9EED exists (verified by reporting)
15:44:29:202 skip configure report for cluster: 0x0006 attr: 0x0000 of node 0xEC1BBDFFFE1C9EED (seems to be active)
15:44:49:747 0xCCCCCCFFFE37C9B0 error APSDE-DATA.confirm: 0xA7 on task
15:44:54:718 Current channel 11
15:44:54:727 Device TTL 3524 s flags: 0x7
15:45:10:332 0x086BD7FFFE1E8BE3 error APSDE-DATA.confirm: 0xA7 on task
15:45:23:725 New websocket 172.30.32.1:36652 (state: 3) 
15:45:30:794 0xCCCCCCFFFE2911D9 error APSDE-DATA.confirm: 0xA7 on task
15:45:51:717 0x000D6FFFFE3239C9 error APSDE-DATA.confirm: 0xA7 on task
15:45:54:717 Current channel 11
15:45:54:725 Device TTL 3464 s flags: 0x7
15:45:57:214 0xEC1BBDFFFE1C9EED error APSDE-DATA.confirm: 0xD0 on task
15:46:11:894 0x14B457FFFE40D584 error APSDE-DATA.confirm: 0xA7 on task
15:46:22:145 ZCL attribute report 0xEC1BBDFFFE1C9EED for cluster: 0x0008, ep: 0x01, frame control: 0x08, mfcode: 0x0000 
15:46:32:603 0x14B457FFFEF67BDF error APSDE-DATA.confirm: 0xA7 on task
172.30.32.2 - - [24/May/2020:15:46:43 +0200] "GET /pwa/index.html HTTP/1.1" 200 107399 "http://hassio.local:8123/core_deconz" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/81.0.4044.138 Safari/537.36"
172.30.32.2 - - [24/May/2020:15:46:43 +0200] "GET /pwa/developer.js HTTP/1.1" 404 142 "http://hassio.local:8123/api/hassio_ingress/-83DQ-AkHebA8c_-hzO7jT7cjVnMMDEggFxcKsKnhBU/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/81.0.4044.138 Safari/537.36"
172.30.32.2 - - [24/May/2020:15:46:43 +0200] "GET /api/F34A5B85B4/config?_=1590328004751 HTTP/1.1" 200 3585 "http://hassio.local:8123/api/hassio_ingress/-83DQ-AkHebA8c_-hzO7jT7cjVnMMDEggFxcKsKnhBU/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/81.0.4044.138 Safari/537.36"
172.30.32.2 - - [24/May/2020:15:46:43 +0200] "GET /api/F34A5B85B4/config?_=1590328004752 HTTP/1.1" 200 3585 "http://hassio.local:8123/api/hassio_ingress/-83DQ-AkHebA8c_-hzO7jT7cjVnMMDEggFxcKsKnhBU/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/81.0.4044.138 Safari/537.36"
172.30.32.2 - - [24/May/2020:15:46:43 +0200] "GET /api/F34A5B85B4/groups?_=1590328004753 HTTP/1.1" 200 4552 "http://hassio.local:8123/api/hassio_ingress/-83DQ-AkHebA8c_-hzO7jT7cjVnMMDEggFxcKsKnhBU/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/81.0.4044.138 Safari/537.36"
172.30.32.2 - - [24/May/2020:15:46:43 +0200] "GET /api/F34A5B85B4/lights?_=1590328004754 HTTP/1.1" 200 5610 "http://hassio.local:8123/api/hassio_ingress/-83DQ-AkHebA8c_-hzO7jT7cjVnMMDEggFxcKsKnhBU/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/81.0.4044.138 Safari/537.36"
172.30.32.2 - - [24/May/2020:15:46:43 +0200] "GET /api/F34A5B85B4/sensors?_=1590328004755 HTTP/1.1" 200 13289 "http://hassio.local:8123/api/hassio_ingress/-83DQ-AkHebA8c_-hzO7jT7cjVnMMDEggFxcKsKnhBU/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/81.0.4044.138 Safari/537.36"
172.30.32.2 - - [24/May/2020:15:46:43 +0200] "GET /api/F34A5B85B4/config?_=1590328004756 HTTP/1.1" 200 3585 "http://hassio.local:8123/api/hassio_ingress/-83DQ-AkHebA8c_-hzO7jT7cjVnMMDEggFxcKsKnhBU/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/81.0.4044.138 Safari/537.36"
15:46:44:656 New websocket 127.0.0.1:53078 (state: 3) 
172.30.32.2 - - [24/May/2020:15:46:48 +0200] "GET /pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,0,15&gwid=00212EFFFF042821 HTTP/1.1" 200 53614 "http://hassio.local:8123/api/hassio_ingress/-83DQ-AkHebA8c_-hzO7jT7cjVnMMDEggFxcKsKnhBU/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/81.0.4044.138 Safari/537.36"
172.30.32.2 - - [24/May/2020:15:46:48 +0200] "GET /websocket HTTP/1.1" 101 0 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/81.0.4044.138 Safari/537.36"
15:46:48:261 Websocket disconnected 127.0.0.1:53078 (state: 0) 
172.30.32.2 - - [24/May/2020:15:46:48 +0200] "GET /pwa/developer.js HTTP/1.1" 404 142 "http://hassio.local:8123/api/hassio_ingress/-83DQ-AkHebA8c_-hzO7jT7cjVnMMDEggFxcKsKnhBU/pwa/settings-gateway2.html?_v=8ce76350306a90b00436da47b8fda4f439ecd46a,4,0,15&gwid=00212EFFFF042821" "Mozilla/5.0 (Windows NT 1 [... cut ...] 4044.138 Safari/537.36"
15:47:52:715 0x00158D0003283774 error APSDE-DATA.confirm: 0xA7 on task
15:47:54:716 Current channel 11
15:47:54:725 Device TTL 3344 s flags: 0x7
172.30.32.2 - - [24/May/2020:15:47:55 +0200] "GET /api/F34A5B85B4/groups/2?_=1590328046561 HTTP/1.1" 200 367 "http://hassio.local:8123/api/hassio_ingress/-83DQ-AkHebA8c_-hzO7jT7cjVnMMDEggFxcKsKnhBU/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/81.0.4044.138 Safari/537.36"

Hedgehog57, 2020-05-24:
Similar problem
Raspberry PI 3B+
RaspBee I (26330500)
Raspbian
Docker
Hassio supervised 0.110.0
Deconz addon 5.3.3 (2.05.76)

Phoscon show new FW available 26350500. Update is succesfull in phoscon. but FW is not updated.
docker logs -f <container_id> show similar information.

Additional information about flasher inside container:
```
root@core-deconz:/tmp# ldd /usr/bin/GCFFlasher_internal.bin
/usr/bin/GCFFlasher_internal.bin: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by /usr/lib/libwiringPi.so)
        linux-vdso.so.1 (0x7ebb0000)
        libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76f05000)
        libwiringPi.so => /usr/lib/libwiringPi.so (0x76ee7000)
        libQt5SerialPort.so.5 => /usr/lib/arm-linux-gnueabihf/libQt5SerialPort.so.5 (0x76ecf000)
        libQt5Core.so.5 => /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5 (0x76a6c000)
        libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76a43000)
        libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0x768fb000)
        libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x7687c000)
        libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x7684f000)
        libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76710000)
        /lib/ld-linux-armhf.so.3 (0x76f18000)
        librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x766f9000)
        libcrypt.so.1 => /lib/arm-linux-gnueabihf/libcrypt.so.1 (0x766ba000)
        libudev.so.1 => /lib/arm-linux-gnueabihf/libudev.so.1 (0x7669d000)
        libz.so.1 => /lib/arm-linux-gnueabihf/libz.so.1 (0x76676000)
        libicui18n.so.57 => /usr/lib/arm-linux-gnueabihf/libicui18n.so.57 (0x76466000)
        libicuuc.so.57 => /usr/lib/arm-linux-gnueabihf/libicuuc.so.57 (0x762fb000)
        libpcre16.so.3 => /usr/lib/arm-linux-gnueabihf/libpcre16.so.3 (0x7628d000)
        libdouble-conversion.so.1 => /usr/lib/arm-linux-gnueabihf/libdouble-conversion.so.1 (0x7626b000)
        libglib-2.0.so.0 => /lib/arm-linux-gnueabihf/libglib-2.0.so.0 (0x76163000)
        libicudata.so.57 => /usr/lib/arm-linux-gnueabihf/libicudata.so.57 (0x748d6000)
        libpcre.so.3 => /lib/arm-linux-gnueabihf/libpcre.so.3 (0x7485d000)
```

palitu, 2020-05-25:
hmmm... i too have this issue.  I also cannot connect to the device in DECONz now.  Any help would be great!

[image]

morfei1, 2020-05-25:
I've flashed my RasBee I with firmware 0x26350500 this way, about 2 months ago:

**Update in Raspbian**
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually

No problems so far.

Hedgehog57, 2020-05-25:
@morfei1 , thanks for a information.
Unfortunatelly, this page does not contain information about flashing under docker hassio addon. It only say that if you are using " marthoc's deCONZ Docker image" there is another instruction.
I use "standart" deCONZ container.

morfei1, 2020-05-25:
Sorry, I'm using my RaspBee I on a separate Rpi3, than my Hass.io.

Did you check this?

https://github.com/marthoc/docker-deconz#updating-conbeeraspbee-firmware

Regards!

Hedgehog57, 2020-05-25:
@morfei1,
Yep. I have read it. This is a way to updatre firmware with alternative marthoc container.

pietia7, 2020-05-25:
@Hedgehog57 this worked for me on HA Supervised on Ubuntu:

# this get you in the docker container:
sudo docker exec -it addon_core_deconz /bin/bash

# find PID od deCONZ:
ps -ax | grep deconz

# kill deCONZ:
kill _pid_number_from_command_above_

# flash firmware (make sure to put correct device, and firmware file (example is for Conbee I)
GCFFlasher_internal -d /dev/ttyUSB0 -f /usr/share/deCONZ/firmware/deCONZ_Rpi_0x26350500.bin.GCF

# reboot addon from HA

Hedgehog57, 2020-05-25:
@pietia7,

Cool idea, will try to do so. If You look at my first comment you will see that I did similar thing.
As I see GCFFlasher_internal is inoperational in this addon. ldd shows GCC error.
```
root@core-deconz:/tmp# ldd /usr/bin/GCFFlasher_internal.bin
/usr/bin/GCFFlasher_internal.bin: /lib/arm-linux-gnueabihf/libc.so.6: version `GLIBC_2.28' not found (required by /usr/lib/libwiringPi.so)
```

pietia7, 2020-05-25:
@Hedgehog57,
I missed your info about GCFFlasher error :/

Looking at this comment you can try updating libc6:
https://github.com/marthoc/docker-deconz/issues/153#issuecomment-633212639

1. Run bash in docker
sudo docker exec -it addon_core_deconz /bin/bash

2. echo "deb http://ftp.de.debian.org/debian sid main" | sudo tee /etc/apt/sources.list.d/libc6.list
apt-get update && apt-get install libc6

Hedgehog57, 2020-05-25:
@pietia7,

You are right. I saw this thread and this is exactly a way I plan to do. But MAYBE this a core of the issue we are speaking?

meremortals70, 2020-05-27:
sudo: GCFFlasher_internal: command not found

This seems to be a pain point for most people not sure why this cant be fixed

pietia7, 2020-05-27:
@meremortals70 

if your running GCFFlasher in Docker addon container you don't use sudo, use GCFFLasher command with args only

meremortals70, 2020-05-27:
Please carefully read my post the Deconz is an add on to HA. HA is running in docker. Deconz is in the same container not a separate one.

I tried it without sudo and received the same result

pietia7, 2020-05-27:
@meremortals70 

Hmm, I don't see any mentions here how you have your HA set up.
As I have written: "If your running in Docker addon (meant as official Docker addon in HA Supervised install)"

haniffm, 2020-05-27:
FYI. 
I created similar issue on the HA-addon repository: https://github.com/home-assistant/hassio-addons/issues/1347

Mimiix, 2020-06-12:
Is this still a issue?
If not, need more info on what the actual issue is after all comments.

ruabo, 2020-06-13:
Yes it is still an issue ! Please also see:
https://github.com/home-assistant/hassio-addons/issues/1347#issuecomment-636286711
I'm on raspberry 3b+

Mimiix, 2020-06-13:
I think it is a error in HA. 

If you can try: Put another SD with raspbian and install deCONZ following: https://phoscon.de/en/conbee2/install#raspbian

ruabo, 2020-06-13:
thanks for supporting, but i'm on hass.io and raspbee I

Mimiix, 2020-06-13:
As i said, that might be a error in Home-Assistant (HA) . So in order to debug, get another SD card for the raspberry pi and install raspbian on there.

Smanar, 2020-06-13:
Or you can just install deconz on a PC to make tries and update.

ruabo, 2020-06-18:
> As i said, that might be a error in Home-Assistant (HA) . So in order to debug, get another SD card for the raspberry pi and install raspbian on there.

Well, I'll have to suspend HA/Hassio services for a while, I'll try if possible. Anyway, how may it help if I were successful ? The original problem, if in hassio/ha, would remain

ruabo, 2020-06-18:
> Or you can just install deconz on a PC to make tries and update.

I own a raspbee, so PC is not an option

Mimiix, 2020-07-10:
@ruabo I apparently missed your reply.

Did you succeed by now? Let me knw! :)

ruabo, 2020-07-16:
@Mimiix sorry for the delay; I've been not able to follow your advice. Anmyway I tried something different. I borrowed another raspberry pi 3b+ and a conbee II usb stick, installed from zero Hassio 0.112.3 (hassos 4.11 later), deconz 5.3.6 and tried to update from firmware 264A0700  with no luck (I had problem similar to my raspbee I experience). So I moved the usb stick to windows 10, installed deconz 2.05.77 and I upgraded to 26580700 firmware from the command line. HTH

morfei1, 2020-07-16:
> @Mimiix sorry for the delay; I've been not able to follow your advice. Anmyway I tried something different. I borrowed another raspberry pi 3b+ and a conbee II usb stick, installed from zero Hassio [...]

You can't update firmware in HA-addon.

ruabo, 2020-07-17:
@morfei1 In the past I did !

ruabo, 2020-07-17:
@morfei1 using the HA official add-on

morfei1, 2020-07-17:
> @morfei1 using the HA official add-on

Join discord DE/Ha-addon channel - https://discord.gg/X56uMs

Mimiix, 2020-07-17:
> @morfei1 In the past I did !

Today it doesn't work unfortunately. However, they are working on it : https://github.com/home-assistant/hassio-addons/issues/1347

Mimiix, 2020-08-10:
I think this can be closed. IF not, reach out to me @ruabo ;)

closed 2020-08-10 by Mimiix