dresden-elektronik/deconz-rest-plugin #2798: deCONZ v2.05.76 breaks some bulbs integration with Home Assistant

stickpin, 2020-05-23:
Hi, 

It seems that after logic refactoring (https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2553) Home Assistant working with some bulbs (Ikea for example) incorrectly.

It seems that Home Assistant trying to set CT but it's no longer available.
Here is the home assistant log:
```
2020-05-23 19:31:18 ERROR (MainThread) [homeassistant.components.automation] [Test][TS] Turn On Light: Error executing script. Unexpected error for call_service at pos 1: /lights/17/state parameter, ct, not available
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 153, in _async_step
    self, f"_async_{cv.determine_script_action(self._action)}_step"
  File "/usr/src/homeassistant/homeassistant/helpers/script.py", line 623, in _async_call_service_step
    *self._prep_call_service_step(), blocking=True, context=self._context
  File "/usr/src/homeassistant/homeassistant/core.py", line 1253, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1288, in _execute_service
    await handler.func(service_call)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 213, in handle_service
    self._platforms.values(), func, call, required_features
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 453, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 597, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 484, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/light/__init__.py", line 242, in async_handle_light_on_service
    await light.async_turn_on(**params)
  File "/usr/src/homeassistant/homeassistant/components/deconz/light.py", line 175, in async_turn_on
    await self._device.async_set_state(data)
  File "/usr/local/lib/python3.7/site-packages/pydeconz/deconzdevice.py", line 35, in async_set_state
    await self.async_set(field, data)
  File "/usr/local/lib/python3.7/site-packages/pydeconz/api.py", line 112, in async_set
    await self._request("put", field, json=data)
  File "/usr/local/lib/python3.7/site-packages/pydeconz/gateway.py", line 106, in request
    _raise_on_error(response)
  File "/usr/local/lib/python3.7/site-packages/pydeconz/gateway.py", line 220, in _raise_on_error
    raise_error(data["error"])
  File "/usr/local/lib/python3.7/site-packages/pydeconz/errors.py", line 58, in raise_error
    raise cls("{} {}".format(error["address"], error["description"]))
pydeconz.errors.RequestError: /lights/17/state parameter, ct, not available
```

Light 17 on deCONZ side:
```
{
    "ctmax": 454,
    "ctmin": 250,
    "etag": "937ad074234534534557bb3d4a23d",
    "hascolor": true,
    "lastseen": "2020-05-23T17:24:14.993",
    "manufacturername": "IKEA of Sweden",
    "modelid": "TRADFRI bulb E27 CWS opal 600lm",
    "name": "Table RGB Lamp",
    "state": {
        "alert": "none",
        "bri": 254,
        "colormode": "xy",
        "effect": "none",
        "hue": 0,
        "on": false,
        "reachable": true,
        "sat": 0,
        "xy": [
            0.4592,
            0.4107
        ]
    },
    "swversion": "1.3.009",
    "type": "Color light",
    "uniqueid": "------"
}
```

Thanks for help in advance.

stickpin, 2020-05-23:
The issue also opened on Home Assistant side: https://github.com/home-assistant/core/issues/36044

tubalainen, 2020-05-23:
Stange -- ive got the same bulb type and firmware of bulb here. Ive got no issues at all working that bulb. Tried all features with full functionality and response.
[image]
[image]

Just a thought - does this bulb support CT? It do support XY...

What happens if you remove the bulb from `/config/.storage/core.entity_registry` and let hass re-populate it automatically?

ebaauw, 2020-05-23:
The CWS bulb doesn’t support colour temperature.  Pre-v2.05.76 there was a hack to translate `ct` for IKEA lights into `xy`.  This hack caused issues with the TW lights, and has been removed in v2.05.76.

stickpin, 2020-05-23:
> The CWS bulb doesn’t support colour temperature. Pre-v2.05.76 there was a hack to translate `ct` for IKEA lights into `xy`. This hack caused issues with the TW lights, and has been removed in v2.05.76.

Thanks @ebaauw for clarification. 
I will change it to XY.

closed 2020-05-23 by stickpin

ebaauw, 2020-05-24:
Was actually not related to removing the hack, but due to more stringent checking in the API that attributes set were actually exposed.  As you can see above: the `/lights` resource does not contain `state.ct`.  Made an exception for IKEA, so `ct` can once again be set for this light.
```
$ ph put -v /lights/10/state '{"ct": 454}' 
{
  "ct": 454
}
$ ph get /lights/10/state
{
  "alert": "none",
  "bri": 1,
  "colormode": "xy",
  "effect": "none",
  "hue": 0,
  "on": true,
  "reachable": true,
  "sat": 0,
  "xy": [
    0.5053,
    0.4153
  ]
}
$ ph put -v /lights/10/state '{"ct": 250}' 
{
  "ct": 250
}
$ ph get /lights/10/state
{
  "alert": "none",
  "bri": 1,
  "colormode": "xy",
  "effect": "none",
  "hue": 0,
  "on": true,
  "reachable": true,
  "sat": 0,
  "xy": [
    0.3805,
    0.3767
  ]
}
```

cben0ist, 2020-07-08:
Hi @ebaauw 
I have just updated my deconz and conbee II firmware and Iam having the exact same problem with the Osram bulbs and led strings now (no problem with Philips bulbs):
```
Light 1	OSRAM	LIGHTIFY A19 RGBW
Light 2	LEDVANCE	A19 RGBW
Light 3	OSRAM	LIGHTIFY FLEX OUTDOOR RGBW
```
How can I create the same exception for these ?
Thank you

tumtumsback, 2020-07-13:
> Hi @ebaauw
> I have just updated my deconz and conbee II firmware and Iam having the exact same problem with the Osram bulbs and led strings now (no problem with Philips bulbs):
>
> ```
> Light 1	OSRAM [...]

I am also having the same issue. I lost the ability to change the color temperature of my Ledvance lights. I run circadian lighting and now my Ledvance fixtures won't stay in sync with the rest of the CT lights in my house. HELP!