dresden-elektronik/deconz-rest-plugin #2803: no swtich group created for newly added switch

andreibanaru, 2020-05-24:
I've just added a new Ikea TRADFRI Remote via Phoscon.
I can see it in deconz and in the "Switches" menu of Phoscon but it didn't create a remote group for me like below:

[image]

[image]

Smanar, 2020-05-24:
Hello, have you tried to delete it and re include it ?

andreibanaru, 2020-05-24:
@Smanar Yes, multiple times.
The workaround was to go to the old webapp interface, nagivate to the Groups tab and then edit the members from there. After adding the light, I managed to see it in Phoscon as well.

Smanar, 2020-05-24:
Ha ok, so it' s probably phosvon related.
Can you take a look in the API if you have the group ? If you know how to use the API ?

You need an url like
`http://IP:PORT/api/API_KEY/groups/`

To have the group list

andreibanaru, 2020-05-24:
Seems this is the case. It's Phoscon related.
I have a "TRADFRI on/off switch" present with no lights assigned in the API but not in the Phoscon groups. If I assign a light to it, then it will appear in Phoscon groups.

I used the "Hue Essentials" app for api navigation if anyone's interested.

morfei1, 2020-05-24:
> @Smanar Yes, multiple times.
> The workaround was to go to the old webapp interface, nagivate to the Groups tab and then edit the members from there. After adding the light, I managed to see it in Phoscon as well.

Thank you for sharing it.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

ploebb, 2020-06-20:
Hey, I don't think this is resolved. I see no way to show or create the switch-groups. Those are important, since normale groups do not work if phoscon is down. Switch-groups however create direct bindings between the switches and lights.
Furthermore, even the proposed workaround seems to not work consistantly for me.

sushifrick, 2020-06-30:
The Problem is not resolved for me either....

Mimiix, 2020-06-30:
@ploebb @sushifrick This has been removed in Phoscon earlier this year. I re-opened this issue and i will move this issue to the Phoscon repository after this comment.