dresden-elektronik/deconz-rest-plugin #2805: Heiman Gas Sensor HS1CG no state change

karsie1994, 2020-05-24:
I connected 2 Heiman Gas Sensor HS1CG sensors on 2 different Home Assistant instances with Deconz, adding the sensor is succesful and the sensor is visible in Home Assitant. But no state changes are reported when the testing button is pressed on the device. The state will remain off. Anybody knows how to solve this?

SwoopX, 2020-05-24:
This is not exactly how it works. The state will definitively not change, but probably the alert value.

Haven't actually paid too much attention if the test woul result in any value changes though, to be honest.

karsie1994, 2020-05-26:
Well, I also own the smoke sensor of Heiman and when pressing the test button on the device the state in HA does change to on/true. I thought it was also supposed to happen on the carbon sensor?

SwoopX, 2020-05-26:
Which modelID ist that (from deconz GUI)? Could have a look out of curiosity.

closed 2020-07-11 by stale[bot]