dresden-elektronik/deconz-rest-plugin #2806: Tuya Smart ZigBee Door Window Contact Sensor adding/pairing

panenka, 2020-05-24:
I'm novice and I've trying to add devices to deCONZ, HA. These sensors working in Tuya Zigbee wire gateway. 
When I press reset hole, for a 5 seconds, red LED is blinking for a few second, maybe 3 second. In this time Tuya gateway adds devices without any problem. 
Phoscon recognition is more difficult and I don't know what to do. But I think, that I see device in VNC console now.

[image]
[image]
[image]

https://www.aliexpress.com/item/4000475399821.html?spm=a2g0s.9042311.0.0.27424c4dl8B4HD
Brand Name: REHENTRONIX
Model Number: RH-ZDS2

panenka, 2020-05-24:
I have two same devices as above from the same order, I'v been tried only first one.
From previous order I have another device, VNC shows same information except NWK and IEEE.
Looks similar.
https://www.aliexpress.com/item/10000008534724.html?spm=a2g0s.9042311.0.0.27424c4dl8B4HD
Model Number: WKD-D02
Zigbee variant oon that page.

Thanks a lot.

SwoopX, 2020-05-24:
Please read the attributes from the basic cluster and share the modelId for both devices.

panenka, 2020-05-24:
There was no other information, but a few minutes ago has been shown. Both
different devices has same output. Attached.

panenka, 2020-05-24:
[image]

SwoopX, 2020-05-24:
Thanks, that was the info I was after. What is the exact definition of "difficult" and what's the actual issue anyway?

panenka, 2020-05-24:
As I wrote above. I have 2 same sensors from one order (only one of them
has been connected to Phoscon) and another one from another order. But I
think that all sensors are the same.

SwoopX, 2020-05-24:
so?

panenka, 2020-05-25:
Sorry. Too late, tired.

1. Pairing. Why are some devices shown only in VNC view? Why are not shown
as others?
2. Why are switches between lights?
3. Why is needed control buttons as prepared. For example one press means
light on AND long press means adding intensity. Why is not possible to
control this independently? There is possible to control double click,
tripple click, quad click?
4. What to do, when something is not working? Where is correct place for
discussion and support? GitHub? This discussion? Not sure.

SwoopX, 2020-05-25:
Now we're talking.

1. That's the way it is. Phoscon is closed source and devices need to be added independently by DE. There's nothing we can do here.
2. Does that really matter? I assume random pick.
3. You mean in Phoscon? Not sure if I fully get your point. Generally, you can distinguish between the button presses as you described if the device supports it. Not sure though how that is reflected in Phoscon, as I don't use it.
4. Really depends on your issue. This is only for REST API and not for the other components. Chime in the Discord server: https://discord.gg/QFhTxqN

panenka, 2020-05-26:
Please correct me, if I've understood this clearly. My sensors are the same like in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2559, ie. 0x0004 is Tuyatec, 0x0005 is RH3001? So these sensors has been added to newest release 3 days ago.
I've been updated to this version yesterday, but I don't see any changes. Sensors are only in mesh network (VNC) and nothing has been changed. What to do now? Pair it again?

SwoopX, 2020-05-27:
I cannot really follow what you're after. The device(s) seem to have been paired alright. What do you expect?

panenka, 2020-05-28:
After new pairing (two pairing for each device, first device needs 3 pairing, it doesn't show battery sensor) is everything ok.

closed 2020-05-28 by panenka