dresden-elektronik/deconz-rest-plugin #2809: IKEA Fyrtur stops responding

tyr84, 2020-05-24:
Setup:
- Hassio
- Deconz plugin 2.05.76
- Fyrtur without repeater, latest firmware

I've recently installed a Fyrtur curtain, and successfully added it to deconz.
There is no issues with the functionality, i can issue commands through both VNC and Home Assistant. Cover position and battery status also works as expected.

However, after some time (30-60min?) the curtain stops responding to commands (both in vnc and ha), and it is no longer possible to read cluster info. The node does not "gray out", which is strange given no response. Pressing "0" does nothing. The node acting as router is an IKEA Traadfri Socket in the same room.

The curtain keeps responding as expected to the physical buttons on the unit itself. It seems to me the zigbee module falls asleep.

If i start the "add light" in phoscon and do a factory reset on the curtain, it starts responding again, but after a time it falls back to sleep like before.

Any ideas?
How can i debug this?

ebaauw, 2020-05-24:
Does it still react to the group commands sent by the open/close remote?

If not, most likely it's been kicked out by its parent.  You might try and make the (former) parent leave and rejoin the network, by pressing `L` in the GUI, when its node is selected.  Otherwise, simply power cycling the FYRTUR (disconnecting the battery for 5 seconds) should bring it back.  No need to reset it.  On rare occasions, I need to open the network (search for devices), while power cycling the blind.

If so, most likely the coordinator has lost the route to the FYRTUR.  Pressing `0` on the parent might remedy this, otherwise `L` for leave/rejoin, or power cycling the parent.

mgrn0, 2020-05-25:
@ebaauw is there a manual / documentation for all these GUI commands you mentioned? I did not find anything in the pdf manual..

ebaauw, 2020-05-25:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-for-Dummies#keyboard-shortcuts

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix

xxLeoxx93, 2022-05-01:
@ebaauw 
Is there any documentation on how you can add the blinds to deconz but also couple a remote directly to them?
I tried 5h yesterday to get them to work at all ( now only 2/6 are updating their status…).
Thanks!