dresden-elektronik/deconz-rest-plugin #2810: Cannot change a color temperature on a Gledopto bulb

anatolyk82, 2020-05-24:
Every release brings something new :/ When I change a color on a gledopto bulb so it goes to an rgb mode then there is no way to return the bulb back to the color temperature mode. It ignores all changes of color temperature and keeps the set color. Home Assistant gives an error.

Version of deconz is 2.05.76

ebaauw, 2020-05-24:
What type of GLEDOPTO do you have?  Can you list the resource that the API has created for it here?

anatolyk82, 2020-05-24:
Hi @ebaauw, thanks for the response. After updating to 2.05.76 I can control a color temperature for ikea bulbs (tanks for the fix) but couldn't do any longer for Gledopto bulbs. I tried Home Assistant, pwa, they simple didn't react anyhow. I reset the bulbs, no effect. When I remove them from the database, I guess, and paired again, it worked. The gledopto bulbs are GL-B-008Z.
 Thanks anyway. We can close the issue. Hopefully it will help some people who have the same problem.

closed 2020-05-24 by ebaauw