dresden-elektronik/deconz-rest-plugin #2811: Unable to locate package deconz-dev

olicooper, 2020-05-25:
I am on a Raspberry Pi 1B+ (armhf) running raspbian buster. 

I would like to compile my own version of the application, so I tried following [the official install guide](https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only) but I get the error `Unable to locate package deconz-dev`. 

I tried installing the ubuntu version [found here](http://deconz.dresden-elektronik.de/ubuntu/beta/) but that complained it was for x64 architecture so I have run out of ideas how to get the package. Does anyone know how to fix this?

olicooper, 2020-05-25:
Okay, so I spent hours trying to get it to work, and shortly after posting this issue I managed to acquire the package. I think the dev package is just not visible on [here](http://deconz.dresden-elektronik.de/raspbian/beta/).

I had to run the following:
```
wget -O - http://phoscon.de/apt/deconz.pub.key | sudo apt-key add -
sudo sh -c "echo 'deb http://phoscon.de/apt/deconz (cat /etc/os-release | grep -oP '(?<=VERSION_CODENAME\=).*')-beta main' > /etc/apt/sources.list.d/deconz.list"
sudo apt update
sudo apt upgrade
sudo apt install deconz-dev
```

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

Oderik, 2020-11-24:
Observed the same issue, solved it based on https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2811#issuecomment-633328779

doit4fun, 2021-09-21:
I've tried it but I get this... 
sh: 1: Syntax error: "(" unexpected

is the command not good or something is wrong?

Smanar, 2021-09-21:
Perhaps https://www.phoscon.de/en/conbee2/install

doit4fun, 2021-09-21:
I've managed to install it on Odroid XU4. I run Ubuntu on XU4 and the installation instruction form the website don't work because Odroid XU4 is using arm based linux not Ubuntu x64 versions so the package manager doesn't know to lookup for the right package as described in the main website.
So, to bypass this you must install the package manually 

1. get the latest version localy with
    wget https://deconz.dresden-elektronik.de/raspbian/deconz-latest.deb
....yes the right package is the raspian .deb version since it's working with ARM architecture 
2. install the deb package
    sudo dpkg -i deconz-latest.deb
3. install missing dependencies
sudo apt update
sudo apt install –f

for me worked fixing the dependencies with
apt --fix-broken install
and after the installation process finishes without error

4. next try to launch the Phoscon gateway software from VNC or RemoteDesktop and launch the desktop app... you will find the icon in the menu

P2W2, 2021-11-15:
I have the same issue. I tried to install it on a raspberry arm & an Intel processor. 
None of the solutions above helped me.
I get the same `sh: 1: Syntax error: "(" unexpected` error or `de_web_plugin.h:16:10: fatal error: deconz.h: No such file or directory
   16 | #include <deconz.h>
      |          ^~~~~~~~~~` Error if I try to compile the project with the solution from @doit4fun.

Smanar, 2021-11-15:
```
pi@raspberrypi:~ $ sudo apt install deconz-dev
deconz-dev est déjà la version la plus récente (2.13.01-raspbian-buster-beta).
```