dresden-elektronik/deconz-rest-plugin #2812: Issue with home assistant and legrand cover shutter celiane netatmo

plum59, 2020-05-25:
Hello
I need support on integration deconz in HA.
In facte all seems good except for Legrand celiane netatmo cover shutter switch.
The switch is present in deconz and in home assistant too  but in HA it remains in closed state continuously

The stop command works, the open command works
but the home assistant closing command does not work, it activates the opening, not closing.

If i use deconz api rest i can open and closed the cover normaly

http://1*******:40850/api/********/lights/3/state
{ "open": false }

http://1*******:40850/api/********/lights/3/state
{ "open": true }

it looks like the entity "cover" in home assistant is badly configured for closing

SwoopX, 2020-05-25:
So it sounds to me it's a HA issue then. You described the REST API works correctly, right?

plum59, 2020-05-25:
Hello , i don't no because when i had discover my switch in deconz (with HA deconz integration) this entity is automaticaly create in HA .

Smanar, 2020-05-25:
Heu ? the command in the api is not state/open but state/on or state/bri.
I have never tried the "open" command, perhaps it's from the last modification for shutter, I need to take a  look ....

Smanar, 2020-05-25:
Ok so there is some change since the last @ebaauw modification.
Now theses devices support "open", but "open" is the opposite at "on"
"open" > targetOpen = map[param].toBool();
"on" > targetOpen = !(map[param].toBool());

ebaauw, 2020-05-25:
See https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2475#issuecomment-609485739

Smanar, 2020-05-25:
Ok So I have found archive ^^

On previous version State = on  > Close the shutter, so it's good with last modification 
```
$ ph put -v /lights/327/state '{"on": true}'
{
  "open": false
}
```

So I realy think the problem is from HA, ca you see the command used by HA ?

plum59, 2020-05-25:
Hello,

like mentionned on discord the problem is when i use cover.closed it seems post on deconnz this intsruction

2020-05-25 18:19:19 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': True, 'bri': 254}" to "172.30.33.8 /lights/3/state"
2020-05-25 18:19:19 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/3/state/lift': 100}}]

or if i use api rest with lift: 100 my cover is opening non closing

Smanar, 2020-05-25:
>2020-05-25 18:19:19 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': True, 'bri': 254}" to "172.30.33.8 /lights/3/state"

And using directly the api, this command is working ?

If not, try with only "on" or only "bri"

Smanar, 2020-05-26:
Ok, so I realy don't see why the code is not working ?
It's a new device ?
Can you check if the model id is still exaclty "Shutter switch with neutral" ?
It have already worked ?

I know how to repair it, but I don't want to break the installation on others users ?

And As you have Hass, I think you can't test code modification ?

I will make an ask on legrand issue.

plum59, 2020-05-26:
Good evening.
I'm sad , bad news :(

No it is no new device but before i use homekit integration in HA with legrand control hub and it's work fine , but my hub is broken and i try to use deconz intead off.

 the model id is still exaclty "Shutter switch with neutral"

In VNC :
[image]

In Deconz:
[image]

With deconz it is my primary test never test before but with homekit or with the physical button is ok

I have hassio and i don't know how i can try it because it is just a container in docker but , i don't now how i can build the image specific for hassio

[image]

To resume , to be sure you have all the good information 
When i use service close.cover in hassio
the cover try to opening with this log
2020-05-26 18:14:09 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': True, 'bri': 254}" to "172.30.33.1 /lights/3/state"
2020-05-26 18:14:09 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/3/state/lift': 100}}]

When i use open.cover
the cover opening too with this log 
2020-05-26 18:16:14 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'on': False}" to "172.30.33.1 /lights/3/state"
2020-05-26 18:16:14 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/3/state/open': True}}]

When i use stop action 
the cover stop with log
2020-05-26 18:17:01 DEBUG (MainThread) [pydeconz.gateway] Sending "put" "{'bri_inc': 0}" to "172.30.33.1 /lights/3/state"
2020-05-26 18:17:01 DEBUG (MainThread) [pydeconz.gateway] HTTP request response: [{'success': {'/lights/3/state/lift': 'stop'}}]

When i use api rest 
{ "open": false } >> closing
{ "open": true } >> opening
{ "on": true } >> closing
{ "on": false } >> opening
{ "lift": 0} >> closing
{ "lift": 100} >> opening
{"bri": 0} >> closing
{"bri": 254} >> opening
{ "on": true, "bri": 254 } >> opening (like i use cover.close see log before)

If i check on cluster info:
When cover is open 
Current lift percentage : 0

When cover is close 
Current lift percentage : 100

Thx for help

Smanar, 2020-05-26:
Yep, It's clear.
I just need to reverse the "bri" command. the command don't work in HA because you are using on and bri in the same command, and ATM at least for your device, this command is reversed.

I have just 2/3 lines to remove, but I realy think the difference is from your device, perhaps a recent firmware update.

I think I will need to check something (perhaps a firmware version, to know if I need to reverse the command or not.

Or If realy I don't found clean solution I will disable the "bri" parameter for legrand.

plum59, 2020-05-27:
Ok thx 
I await your news

plum59, 2020-05-27:
If you build *.deb with modification here or here or and other place i will able to  try on hassio i think
http://deconz.dresden-elektronik.de/raspbian/alpha/
http://deconz.dresden-elektronik.de/raspbian

plum59, 2020-05-27:
Hello , 
can you give me the modification you think should be do and where (name file) ?
i will try to make a new version in my own side and create a personnal integration with this in hassio.
I will test an if it is ok i will told you

i think is probably around this :
rest_lights.cpp

  else if (taskRef.lightNode->modelId() == QLatin1String("Shutter switch with neutral"))
        {
            // Legrand invert bri and don't support other value than 0
            targetLiftZigBee = targetLift == 0 ? 100 : 0;
        }

but i don't know what exactly i should modify

thx a lot

Smanar, 2020-05-27:
Just remove the line ^^.
Or you can use targetLiftZigBee = targetLift == 0 ? 0 : 100; (Better to have only 2 value)

Or you can too just disable the "bri" command or ignore it according to the device id.

Or you can too use "open" and "close "in HA.

There so much method possible, but I didn't choose one yet.

plum59, 2020-05-27:
Ok, i will try one of your solution, but i need more help, i will clone repo off github make the modification but  how i can make *.deb after when modification will be done ?

Smanar, 2020-05-27:
Are you using HA or Hass.io ?

plum59, 2020-05-27:
Hass.io

plum59, 2020-05-27:
but i think is not a probleme 
i have create my own deconz integration  i just need deb file with correction to include in docker file
actually docker file is :+1: 

ARG DECONZ_VERSION
RUN if [ "${BUILD_ARCH}" = "armhf" ]; \
        then \
            curl -q -L -o /deconz.deb http://deconz.dresden-elektronik.de/raspbian/beta/deconz-${DECONZ_VERSION}-qt5.deb; \
        elif [ "${BUILD_ARCH}" = "aarch64" ]; \
        then \
            curl -q -L -o /deconz.deb http://deconz.dresden-elektronik.de/raspbian/alpha/deconz_${DECONZ_VERSION}-debian-stretch-beta_arm64.deb; \
        else \
            curl -q -L -o /deconz.deb http://deconz.dresden-elektronik.de/ubuntu/beta/deconz-${DECONZ_VERSION}-qt5.deb; \
        fi \
    && dpkg -i /deconz.deb \
    && rm -f /deconz.deb \
    && chown root:root /usr/bin/deCONZ* \
    && sed -i 's/\/root/\/data/' /etc/passwd

COPY data/ika-otau-dl.sh /bin/
COPY data/ledvance-otau-dl.sh /bin/
COPY data/nginx.conf /etc/nginx/nginx.conf
COPY data/run.sh data/discovery.sh /

Smanar, 2020-05-27:
Ok, so sorry, I have already tried to helpo someone with Hass.io, I can nothing for you

There is window, unix, mac and Hass.io.

plum59, 2020-05-27:
:(  it is C language , i m not be sure i will be able to do that :D

plum59, 2020-05-27:
and i m not sure i have right to do that , it seem to be hacking :)

plum59, 2020-05-27:
thank you anyway, i will wait for you to get feedback from other users to maybe update

Smanar, 2020-05-27:
On your version you can do all you want, you can freely edit the code.
But act on system on Docker system is just hell for me.

ATM, I think you can act on HA directly ? waiting for patch ? you can't edit the command send by the addon ?

plum59, 2020-05-28:
I'm sorry for insisting like that. but if I can't find a solution I will have to buy a new hub and it is expensive for me.

I think that in home assistant I cannot modify anything the code of the cover service is integrated.

the only solution for me is to modify the sources of the deconz application. I tried to see but I do not know anything in assembly and therefore I can not modify what I want and redo a deb file.

By cons I know docker and I have no problem with it

plum59, 2020-05-28:
oh , my bad it is explain here
https://github.com/dresden-elektronik/deconz-rest-plugin

Smanar, 2020-05-28:
Ha yep, the compiling procedure ?

But in HA, you can to use "virtual device" with command line, remember decone works with simple REST command, it's easy to use.

plum59, 2020-05-29:
Hello,

thx a lot i discover the power of rest_command 
https://www.home-assistant.io/integrations/rest_command/

I would have preferred an integrated solution, but it works perfectly well

thanks for spending time for me

have a nice week

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

Smanar, 2020-06-06:
Not yet ^^
But for the moment there is only one person with this problem, so I m waiting.

smesguich, 2020-06-08:
Okay so, I think I have the same issue ...
I trying to compare HA and jeedom right now.
Jeedom is not very reactiv to update plugin for deCONZ integration ...

and here the issue
[image]

Smanar, 2020-06-08:
Ha nice (if I can say that ^^)

Have you make a try with the API ? You have too the "bri" command reversed ? with device updated or not ?

> In fact you can close the shutter with :
> - bri = 255
> - on = true
> - open = false

> But It seem on the last version the device OPEN the shutter with bri = 255, no changes for others.

smesguich, 2020-06-08:
In fact : via API or Jeedom :
{"bri": 0} >> closing
{"bri": 254} >> opening
{"bri_in": 0} >> stop

Via HA UI : 
UP==> UP, 
Stop ==> Stop
Close ==> UP

I'm running deCONZ in docker on my Synology and HA with Hass.io
[image]

smesguich, 2020-06-08:
BTW : My shutter is open but :
`{
  "etag": "3b5d3f03ac18414340116bf317c15797",
  "hascolor": false,
  "lastseen": "2020-06-08T16:26:29.127",
  "manufacturername": "Legrand",
  "modelid": "Shutter switch with neutral",
  "name": "Volet roulant Salon",
  "state": {
    "alert": "none",
    "bri": 254,
    "lift": 100,
    "on": true,
    "open": false,
    "reachable": true
  },
  "swversion": "0021",
  "type": "Window covering device",
  "uniqueid": "00:04:74:00:00:99:c9:bf-01"
}`

smesguich, 2020-06-08:
This code open the shutter

```
{
    "bri": 255,
    "lift": 100,
    "on": true,
    "open": false
}
```

smesguich, 2020-06-08:
Okay, so an other issue related I think : whatever way I used, in all case : the bri value is not updated back. I mean, previously, we had immediat feedback about the last bri value. Right now, the value didn't look like updated

Smanar, 2020-06-08:
Ok so I will do 2 thing
- Reverse Bri command
- Ignore "bri" if there is too "on"

Someone wana make a try https://github.com/Smanar/deconz-rest-plugin/commit/0f452f65a13f726ff7bf92147d331c4f62d6e542 ?

The code is on a special branch, not the master.

Smanar, 2020-06-12:
No one can try the modifications ?

smesguich, 2020-06-14:
Yeah, I'm trying ! But I'm still facing issue to bind a custom lib on the https://github.com/marthoc/docker-deconz image ....

Smanar, 2020-06-14:
Ha yes, I know, I have so much problems with this kind of system.

smesguich, 2020-06-15:
It's really borring because my Synology doesn't have lot of HW ressources and Docker is a good way to up/downgrade an APP with data persistance and co ...

smesguich, 2020-07-03:
> 
> 
> Ok so I will do 2 thing
> 
>     * Reverse Bri command
> 
>     * Ignore "bri" if there is too "on"
> 
> 
> Someone wana make a try [Smanar@0f452f6](https://github.com/Smanar/deconz-rest-plugin/commit/0f452f65a13f726ff7bf92147d331c4f62d6e542) ?
> 
> The code is on a special branch, not the master.

Compiliation right now ... I will try at lunch

smesguich, 2020-07-03:
It didn't change anything ...

Smanar, 2020-07-03:
Seriously ?
You have tried the code on my fork in legrand branch, and the official version 77 or 78 ?

> In fact : via API or Jeedom :
> {"bri": 0} >> closing
> {"bri": 254} >> opening
> {"bri_in": 0} >> stop
> 
> Via HA UI :
> UP==> UP,
> Stop ==> Stop
> Close ==> UP

So for you on your last compilation, it still broken ?

You have used the command
`git clone --brach legrand https://github.com/dresden-elektronik/deconz-rest-plugin.git`

smesguich, 2020-07-03:
Well not exactly. 
On  HA :
UP and STOP was Okay. DOWN wasn't working and grey.
But after try it manuly on the shutter and next uusing the position button, down button was usable (black) but up wasn't ....

So IDK why / what the issue ...

So in fact : position 0 = down
position 100 = up
stop = stop

UP/DOWN may not works, depends on the last buton use ...
So related on the last status ?

Mimiix, 2020-07-03:
The HA buttons are often dodgy.

What happens if you call the cover.open_cover and cover.close_cover services manually?

Smanar, 2020-07-03:
Why HA grey the button ? It disable "up" if the cover is already open for it ?

smesguich, 2020-07-06:
> 
> 
> The HA buttons are often dodgy.
> 
> What happens if you call the cover.open_cover and cover.close_cover services manually?

IDK how to do that

> Why HA grey the button ? 
IDK but it's seems that HA think it's not possible to do.

>Perhaps it have reversed values ?
Looks like ! But not exactly. As far as I said, the status shown in HA looks like related to the last physical use of the shutter ...
It's strange because, previously, the API was able to know what was the state of the shutter, no matter if it was by API or physicaly. ...

Mimiix, 2020-07-06:
@smesguich https://www.home-assistant.io/docs/scripts/service-calls/

Smanar, 2020-07-06:
>It's strange because, previously, the API was able to know what was the state of the shutter, no matter if it was by API or physicaly. ...

Yes, You haven't it anymore ?

It can be a good test, you haven't changed your wiring, so we can take it as base.
If you move it manualy, what is the result on HA ?

BTW you haven't access to http debug (request send at/from the gateway) on HA ?

closed 2020-08-06 by stale[bot]

smesguich, 2020-08-10:
> 
> 
> @smesguich https://www.home-assistant.io/docs/scripts/service-calls/

Hum ... I have to test as soon I'll be back of holidays !

> Yes, You haven't it anymore ?
>
> It can be a good test, you haven't changed your wiring, so we can take it as base.
> If you move it manualy, what is the result on HA ?
>
> BTW you haven't access to http [...]

No, I didn't have this function anymore .....a

smesguich, 2020-08-14:
Okay, so long days after ...
Up, stop and close services are working on HA with your release @Smanar

Smanar, 2020-08-14:
I haven't updated the files yet ^^
I think it s because HA is using now the last ebaauw code, without "on" and "bri"

plum59, 2020-08-14:
Hello
i have test on my side with last  ha version and is not working
i have exactly the same cheat , when i try to open is opening when a try to close is opening

Smanar, 2020-08-14:
But HA don't use open = true/false for open close ?

I can remake change for bri/on, but it don't think it will be on next version, the alpha is already in test.

@smesguich you have compiled the branch legrand on my old fork ? It s the one you are using ATM ?
Just to be sure can you use pls the command "bri", h_ttp://IP:PORT/api/KEY/lights/ID/state with { "bri": 0} (or using HA if you are sure it s the command used)
And check in the device if bri = 0 when the command is finished (to know if I need to reverse too the returned value.

smesguich, 2020-08-17:
> 
> 
> wich one release are you using ?
> 
> I have cleaned the fork yesterday, and I think I have deleted all files ....
> 
> You will have soon the deconz version 80, just tell me if you still have a problem with it, long time I have started this issue, don't remember all the history.

Sorry, I haven't precised : 
I was running with [Smanar@0f452f6](https://github.com/Smanar/deconz-rest-plugin/commit/0f452f65a13f726ff7bf92147d331c4f62d6e542). 
Right now, i'm running on : 

- Home Assistant Core 0.114.1  with HassOS 4.12
- deCONZ 2.05.80 / 14/08/2020 in Docker
- Conbee II 26580700

smesguich, 2020-08-17:
> 
> 
> But HA don't use open = true/false for open close ?
> 
> I can remake change for bri/on, but it don't think it will be on next version, the alpha is already in test.
> 
> @smesguich you have compiled the branch legrand on my old fork ? It s the one you are testing ATM ?
> Just to be sure can you use pls the command "bri", h_ttp://IP:PORT/api/KEY/lights/ID/state with { "bri": 0} (or using HA if you are sure it s the command used)
> And check in the device if bri = 0 when the command is finished (to know if I need to reverse too the returned value.

Okay, I will test :)

Smanar, 2020-09-04:
BTW someone else have tried, only with API, with new device and haven't problem on his side > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/883#issuecomment-687203640

samsam-rolon, 2020-10-11:
Well, a little bit long after, To be clear : still issue right now on HA _**(Home Assistant 0.116.2 on HassOS 4.13)**_ with Deconz ( Conbee II 2.05.81 and Firmware 26660700).

Open : up and down are working properly
BRI : 0 UP (up) and 255 down (close)

BUT

BRI option looks like overiding OPEN option.
So 
open: false and bri: 255 will open up the shutter
open: true and bri: 0 will close down the shutter

Mimiix, 2020-10-12:
@samsam-rolon That is something for the HA Addon repo or the HA integration repo.

[47 more replies]