dresden-elektronik/deconz-rest-plugin #2814: REST API - Create sensor documentation incorrect?

jasperslits, 2020-05-25:
I am following http://dresden-elektronik.github.io/deconz-rest-doc/sensors/ to workaround the limitation in #90 (Sensor config.on not implemented?) . I want presence detection of my Philips Motion Sensor to be turned off when I am at home as it spams my Home Assistant log all the time . This was easy with the Hue Hub but not possible. 

I decided to play around with the REST API to delete the sensor during the day and add it at night (to delete it again upon first detection). 

Delete worked well, adding not. Payload: 
```
{"config": {
        "on": true,
        "reachable": true
    },"manufacturername": "Philips", 
	"modelid": "SML001", 
	"name": "Motion Sensor woonkamer", 
	"swversion": "6.1.1.27575",
	"type": "ZHAPresence", 
	"uniqueid": "blablabla"}
```

This gives an undocumented 501 response with "Not allowed to create sensor type" error. ZHAPresence is a documented allowed type. 

However the code in rest_sensors.cpp checks against CLIP type: 

 ```
  if (!type.startsWith(QLatin1String("CLIP")))
    {
        rsp.list.append(errorToMap(ERR_NOT_ALLOWED_SENSOR_TYPE, QString("/sensors"), QString("Not allowed to create sensor type")));
        rsp.httpStatus = HttpStatusBadRequest;
        return REQ_READY_SEND;
    }
```

However the example code in the documentation for creating sensors uses CLIPSwitch so no wonder it works there. 

Question: is the documentation misleading on creating sensors or am I missing something? 

Is there a better way via REST API or other means to prevent presence detection / state updates?

Smanar, 2020-05-25:
I think ZHA__________ are reserved to deconz for it's own real sensor.

jasperslits, 2020-05-25:
I managed to solve it - for now - in a different way by changing the Python code of Home Assistant so it only fires an event if the config.on == true

I can create the sensor when changing ZHA to CLIPPresence but then it doesn't show up with any attributes (motion detected, temperature, lux) in Deconz / Phoscon.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix