dresden-elektronik/deconz-rest-plugin #2815: Osram Smart+ Plug can't be added to deConz

Superwuzy, 2020-05-25:
Hi there,

I'm using several Smart+ Plugs that were successfully connected to a Philips Hue Bridge v2 in the past. I have did a reset on all of the plus (10+ seconds till the relay does a click).

I'm not able to add these plugs via the lights dialog to deConz on a Raspi 4 using my Confee II stick (firmware 26580700, deConz: 2.05.77). The stick searches for a minute or so but doesnt fine any of the plugs. I've added several Xiaomi sensors to deConz without any problems. The plugs have different firmware versions (1.04.12 & 1.04.90).

I can add them to the Hue bridge without any problems.

Any hints?

Is this maybe an issue with the new deConz 2.05.77?

Thanks!

Smanar, 2020-05-25:
Hello, have you a GUI, to see if you can see the device in deconz ?

Superwuzy, 2020-05-25:
No, with the new firmware and deconz version I'm able to receive data from my already attached Xiaomi sensors. But I'm not able to attach new Xiaomi sensors, Osram plugs (resetted) or Hue bulbs (resetted).

Superwuzy, 2020-05-26:
> Hello, have you a GUI, to see if you can see the device in deconz ?

@Smanar 

I just installed deCONZ 02.05.75 on a Windows 10 Laptop and plugged my Conbee II (newest firmware: 26580700) into the laptop: All the Osram plugs were detected as lights within a few seconds and can be controlled without any problems.

On my Raspi 4 with deConz latest (2.05.77) I'm not able to learn in any new sensors or plugs - two weeks ago a lern in of my Xiamoi sensors with deConz 2.05.75 worked flawlessly. So I downgraded yesterday to beta 2.05.75 - unfortunately with the same problem. A very frustrating experience.

Smanar, 2020-05-26:
Ok so the conbee on a win10 machine all is working
But the same conbee on the raspi, inclusion don't work ?

Check first if the problem is not hardware, using USB extension cable, no USB 3.0, unplug other USB device, ect ...

If you are sure the problem is not hardware, check if you have a firmware number in Phoscon/Gateway (If I m right already done)

Then in last solution you can take a look here, in the hidden configuration https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Network-lost-issues. But if I m right, theses setting are on the device, so it s the same configuration on the windows machine than on the raspberry. So if it's work on WIN, it can't be the problem. For me the problem is hardware.

Superwuzy, 2020-05-26:
@Smanar You're absolutely right. Normally I'm using an extension cable for the Conbee because the Rapsi "lives" in a network cabinet. During the last days I've moved the raspi to my desk because of other issues and attached the Conbee stick directly to the Raspi. The Conbee stick did work so and received traffic from sensors etc. But learning in of new sensors did not work. Now it works using the newest firmware and deConz release when I plug the extension cable to the Raspi. I guess that RF interference from the Raspi has caused reception problems with the Conbee stick attached to the Raspi directly. Thats why the sick worked on my Windows 10 Laptop because there is a better RF isolation. Thanks for your support!

closed 2020-05-26 by Superwuzy

Smanar, 2020-05-26:
NP, but I will end up believing that I am the only one who does not use the USB extension ^^

ebaauw, 2020-05-26:
I run a test deCONZ with a ConBee II on a RPi 4B on my desk without an extension cable.  It's connected to USB-2 (of course), nothing in USB-3 and Wifi and Bluetooth disabled.  Zigbee channel 15.

Interference is very erratic in nature.  This is really an area where you mileage _will_ vary.