dresden-elektronik/deconz-rest-plugin #2816: Tradfri Dimmer (on/off switch) pairing problems

mastervol, 2020-05-25:
ConBee II on Raspberry 2
Version: 2.05.77/8.3.2020
Firmware: 26580700
OS Openhabian
OS Release Raspbian GNU/Linux 8 (jessie)

I cannot add a Ikea Tradfri On/Off Dimmer Switch, not as IKEA and not as OTHER.
It is not recognized in the phoscon app.
Every other light bulb I had was recognized perfectly.

Thanks for comments.

meremortals70, 2020-05-26:
I have just placed an issue for a Philips Hue Dimmer with the same problem

jvyoralek, 2020-05-31:
I have ConBee II as well with slightly newer version 2.05.77/22.5.2020 and maybe older firmware 264A700 and was able to pair it via Ikea TRÅDFRI option and see it in the list of connected switches:

[image]

Have you seen red blinking led after pressing 4 times?

mastervol, 2020-06-01:
> Have you seen red blinking led after pressing 4 times?
Yes it blinks a few times, but it is not added.

I will try to install an older deconz version, hopefully all other items will stay.

mastervol, 2020-06-09:
Ok it turns out I actually cannot downgrade.
I tried now installing http://deconz.dresden-elektronik.de/raspbian/alpha/deconz_2.05.77-raspbian-jessie-alpha_armhf.deb and the stable one from apt-get.
Still I cannot add the switch.

mastervol, 2020-06-15:
Thanks it is fixed with version http://deconz.dresden-elektronik.de/raspbian/alpha/deconz_2.05.78-raspbian-jessie-alpha_armhf.deb.
;-)

closed 2020-06-15 by mastervol

ohjohnsen, 2020-06-20:
Struggling with the same problem, on Windows platform.  How long until 2.05.78 for Windows is released?

LeducH, 2021-02-18:
Still not working on version 

```
version 2.09.03 / 23/12/2020
Firmware 26660700
```

Mimiix, 2021-02-24:
@LeducH BEst would be to open a seperate issue :)

TuomasPakkanen, 2021-03-06:
> Still not working on version
> 
> ```
> version 2.09.03 / 23/12/2020
> Firmware 26660700
> ```

Pairing worked for me by using the "Other switch" option. Did not work with the Ikea Trådfri dimmer option.

vbaros, 2021-03-17:
Same for me. Tried multiple times to pair using Ikea option and it didn't work. It worked immediately using Other option. Version 2.09.03 / 12/23/2020. Firmware 26660700.

juniorz, 2021-03-30:
I have just paired two brand new switches. Both switches at version 2.2.010 and were bought from an IKEA store in Germany (if the region matters).

```
2.10.04 / 14/03/2021
26660700
```

Add New Switch > IKEA > TRÅDFRI remote control

Press the reset button 4x. Wait for the red(?) light on the switch to turn on. Wait for the light on the switch to turn off. Close the window with the instructions.

istepaniuk, 2023-08-05:
All my TRAFI (E1743) pair normally with deconz, except the one I updated the firmware to (using zigbee2mqtt OTA feature). This last firmware from this year (TRADFI firmware 20230308) does not pair with deconz. 

It does "something" if I choose "Other" instead of IKEA. Though the device seems to finish pairing and the phoscon app says "Compound prepared" (whatever that means), no new device is added to deconz.